springfall2008/batpred #486: Solis charge does not work because of 0A current

marcoaddario, 2023-12-17:
**Describe the bug**
When set the charge window sets the charging current to 0 (avoiding the inverter to charge the batteries)

**Expected behavior**
charge current should be set at charging current. I have tried setting the charge current to a different value, but it has been overwritten on the next predbat run.

**Predbat version**

7.14.13

**Environment details**

- Solis inverter
- Standard HAOS

**Screenshots**
[image]

[image]

**Log file**
```
2023-12-16 22:30:01.683727 INFO pred_bat:  METRIC: [ 127.96,  178.71,  322.12,  371.04,  429.95,  445.43,  448.35,  451.05,  453.87,  456.85,  460.08,  460.08,  460.08,  460.08,  460.08,  460.0│
│2023-12-16 22:30:01.938002 INFO pred_bat: Evaluate trigger {'name': 'large', 'minutes': 60, 'energy': 1.0} results off total_energy 0.0                                                          │
│2023-12-16 22:30:01.951701 INFO pred_bat: Evaluate trigger {'name': 'small', 'minutes': 15, 'energy': 0.25} results off total_energy 0.0                                                         │
│2023-12-16 22:30:01.959501 INFO pred_bat: Best charging limit socs [7.1, 7.1] export [] gives import battery 45.62 house 6.3 export 1.2 metric 767.22 metric10 811.9                             │
│2023-12-16 22:30:02.048688 INFO pred_bat: Export threshold used for optimisation was 11.0p                                                                                                       │
│2023-12-16 22:30:02.101645 INFO pred_bat: High export rate windows [0: 12-16 22:00:00 - 12-16 23:00:00 @ 15.0, 1: 12-16 23:00:00 - 12-17 00:00:00 @ 15.0, 2: 12-17 00:00:00 - 12-17 01:00:00 @ 15│
│2023-12-16 22:30:02.109019 INFO pred_bat: Import threshold used for optimisation was 11.0p                                                                                                       │
│2023-12-16 22:30:02.160531 INFO pred_bat: Low import rate windows [0: 12-16 23:00:00 - 12-17 08:30:00 @ 7.5, 1: 12-17 23:30:00 - 12-18 05:30:00 @ 7.5]                                           │
│2023-12-16 22:30:02.168232 INFO pred_bat: Charge window will be: 2023-12-16 23:00:00+00:00 - 2023-12-17 08:30:00+00:00 - current soc 17 target 100                                               │
│2023-12-16 22:30:02.169473 INFO pred_bat: Configuring charge window now (now 12-16 22:30:00 target set_window_minutes 30 charge start time 12-16 23:00:00                                        │
│2023-12-16 22:30:04.182432 INFO pred_bat: Inverter 0 Wrote 23:00:00 to charge_start_time, successfully now 23:00:00                                                                              │
│2023-12-16 22:30:06.192274 INFO pred_bat: Inverter 0 Wrote 23 to charge_start_hour, successfully now 23                                                                                          │
│2023-12-16 22:30:06.199048 INFO pred_bat: Inverter 0 No write needed for charge_start_minute: 0 == 0                                                                                             │
│2023-12-16 22:30:08.748327 INFO pred_bat: Inverter 0 Wrote 08:30:00 to charge_end_time, successfully now 08:30:00                                                                                │
│2023-12-16 22:30:10.756733 INFO pred_bat: Inverter 0 Wrote 8 to charge_end_hour, successfully now 8                                                                                              │
│2023-12-16 22:30:10.763142 INFO pred_bat: Inverter 0 No write needed for charge_end_minute: 30 == 30                                                                                             │
│2023-12-16 22:30:12.771900 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-16 22:30:12.796491 INFO pred_bat: Inverter 0 Updated start and end charge window to 23:00:00 - 08:30:00 (old 00:00:00 - 05:30:00)                                                        │
│2023-12-16 22:30:12.801115 INFO pred_bat: Setting ECO mode as no discharge window planned                                                                                                        │
│2023-12-16 22:30:12.808860 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00                                               │
│2023-12-16 22:30:12.813298 INFO pred_bat: Car charging from battery is off, next slot for car 0 is 12-16 23:00:00 - 12-17 00:30:00                                                               │
│2023-12-16 22:30:12.820585 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target                                                                                                 │
│2023-12-16 22:30:12.827524 INFO pred_bat: Setting Solis Energy Control Switch to 35 from 3 to enable grid  charging                                                                              │
│2023-12-16 22:30:14.836339 INFO pred_bat: Inverter 0 Wrote select.solis_modbus_energy_storage_control_switch to Timed Charge/Discharge successfully                                              │
│2023-12-16 22:30:16.845906 INFO pred_bat: Inverter 0 Wrote 0.0 to timed_charge_current, successfully now 0.0                                                                                     │
│2023-12-16 22:30:16.852808 INFO pred_bat: Current SOC 17% is less than Target SOC 100. Grid charging enabled with charge current set to 0.00                                                     │
│2023-12-16 22:30:16.883265 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes                                                                           │
│2023-12-16 22:30:16.886573 INFO pred_bat: Completed run status Idle
```

springfall2008, 2023-12-17:
I think my change to use battery_rate_max from the config broke it, can you check if it's set correctly and maybe show the log from the start of a 5 minute run?

springfall2008, 2023-12-17:
Actually I think I found the bug, will release a fix shortly

springfall2008, 2023-12-17:
https://github.com/springfall2008/batpred/releases/tag/v7.14.14

marcoaddario, 2023-12-18:
It perfectly worked.
Thanks

closed 2023-12-18 by marcoaddario