springfall2008/batpred #488: Terrible plan with v7.14.14

slopemad, 2023-12-17:
Just upgraded from 7.14.11 to 7.14.14 and the new plan is... DREADFUL!

[image]

appdaemon.log is here https://pastebin.com/2CuqWapS

Rolled back to 7.14.11 and got a more sensible plan.

[image]

springfall2008, 2023-12-17:
I'm struggling to reproduce what you were seeing on my system configured for Agile, but I think either putting best_soc_keep up a little higher or turning off 'calculate_fast_plan' would fix it.

slopemad, 2023-12-18:
It's very similar to when I set best_soc_keep to zero - the entire plan disappears.  I wonder if it's related?

springfall2008, 2023-12-18:
I think best_soc_keep should be 0.5 or above on the latest version, did you try higher numbers?

slopemad, 2023-12-18:
I only had it set to 0.5 when on 7.4.14, but the effect of the huge drop in low threshold when I upgraded to 7.4.14 was similar to the effect of the drop in low threshold when best_soc_keep is set to zero on 7.4.11.

springfall2008, 2023-12-18:
Yes I think so, a higher value would probably resolve the issue - it's most as I made the keep threshold less aggressive.

slopemad, 2023-12-19:
So I had another look at this tonight, and upgraded from 7.14.11 to 7.14.15, and it went back to a substantially worse plan than the one it originally had - the low threshold went from 10.4p to 18p and the best metric (cost) went from about 100p to nearly 200p.  Similar results when I went up to 7.14.16 also, and then to 7.14.13.  And fortunately it reverted back to the lower low threshold of 10.4p and a best metric (cost) of closer to 100p when I went back to 7.14.12.

It feels like one of the changes made between 7.14.12 and 7.14.13 had a big impact on the low threshold calculation and the plan created by Predbat?

I did play around with the best soc keep, but didn't see any notable change between 0.5 (which I have it set to normally) and 1.

slopemad, 2023-12-19:
This is the current plan with 7.14.12

[image]
[image]

And this was the plan with 7.14.15

[image]
[image]

springfall2008, 2024-02-10:
As things have moved on a lot I will close this ticket, feel free to open a new one if you still have issues.

closed 2024-02-10 by springfall2008