springfall2008/batpred #491: predbat instructing hold  and freeze charging when battery already empty

gcoan, 2023-12-17:
**Describe the bug**
The battery became exhausted at 18:17:
[image]

As the rates are staying flat/bumping up and down, no further charging is planned until much later on
[image]

But Predbat keeps on instructing the inverter to hold charge, freeze charge, all of which come through as extra mobile notifications
[image]

**Expected behavior**
Once the battery is exhausted, no further freeze charge or hold charge or change to idle instructions are made until the soc is greater than min soc.

**Predbat version**
7.14.14

**Environment details**
Gen 1 5kW hybrid, 5.2 battery
HAOS

**Log file**
[appdaemon (5).log](https://github.com/springfall2008/batpred/files/13697498/appdaemon.5.log)

gcoan, 2023-12-17:
More of the same for the next couple of hours, swapping between Idle, Hold Charging

[image]
[image]

I noticed it swaps to Idle on the :00 or :30 minutes and Hold Charging on :06 and :36 minutes so may be related to re-evaluating the plan at the start of every slot and then not a full reval every 5 minutes
[Uploading appdaemon (6).log…]()

springfall2008, 2024-09-01:
Should be fixed now

closed 2024-09-01 by springfall2008