springfall2008/batpred #492: Predbat plans to discharge the battery when its already exhausted

gcoan, 2023-12-17:
**Describe the bug**
At 13:20. I can see in the plan that after the evening peak the battery is exhausted on 4%.  The evening slots 19:00-21:30 are planned to leave the house being supplied by the grid as there's no benefit to charging and discharging after losses taken into account 
[image]

But then from 22:00 predbat's plan is to discharge through to 02:00.  The battery is of course empty so none of these discharges can be executed.
[appdaemon (4).log](https://github.com/springfall2008/batpred/files/13697507/appdaemon.4.log)

Update at 19:15.
The long period of impossible discharge has shortened.  Still trying to discharge for the first hour (22:00 and 22:30) slots, but then charge in the next two before further discharging through to 02:00
[image]

[appdaemon (5).log](https://github.com/springfall2008/batpred/files/13697508/appdaemon.5.log)

**Expected behavior**
When soc is 4% (or min soc), predbat doesn't plan to discharge the battery as there's no soc to discharge.  Instead remain on idle with the house being grid supplied (as is planned from 19:30 to 21:30 slots)

**Predbat version**
7.14.14

**Environment details**
Gen 1 Hybrid, 5.2kW battery, HAOS

gcoan, 2024-02-11:
Predbat still doing this unfortunately. Here's an example from today where it's trying to discharge from 20:30-22:00 when the battery is clearly empty and no solar charging.

[image]

springfall2008, 2024-09-01:
Fixed now, please re-open if there is a new issue

closed 2024-09-01 by springfall2008