springfall2008/batpred #493: Restart Appdaemon on Predbat install

iainfogg, 2023-12-17:
Getting the error about template YAML file / restarting Appdaemon feels like a faff.

It feels like there should be a way to update without Appdaemon messing up the config file, but in the meantime, how about just firing the "restart add-on" service on upgrade (assuming you can detect a version change) and getting Appdaemon to restart itself, rather than users doing it?

I expect you could detect a new version with the predbat_version entity, where the entity has a different version to the constant in the code. That could be the trigger for updating the entity to the new version, then fitting an Appdaemon restart.

gcoan, 2023-12-17:
I have wondered something similar

I don't think the predbat version entity is necessarily the trigger though as it's sometimes taken a while for that to update in my system.

I was thinking whether an automation triggered by a change to the predbat.py file would work?

Or is there a way to not have to restart app daemon at all when the predbat python file is updated?

I also think that the error message is a bit ambiguous pointing to either template =true or a need to restart app daemon.  Could these be more specific statii?

gcoan, 2023-12-18:
And the other thing, the update entity only changes when home assignment recognises that an upgrade is available.  Quite often this doesn't happen for quite a while and I have to do "update information" in hacs or "redownload" to force an update l

slopemad, 2023-12-18:
I find that I don't have to restart Appdaemon after a version upgrade, if I just open and then save the apps.yaml file in a terminal (without making any changes) it will trigger predbat itself to reload without needing a restart of Appdaemon.

springfall2008, 2023-12-18:
Yes that works too.

The issue is HACS installs the template apps.yaml and then moves your
original one back into place afterwards but AppDaemon seems not to see the
second step.

springfall2008, 2023-12-24:
I've address this with a new way to update Predbat from itself, you can select update from the dropdown menu or enable auto-update. No need to restart appdaemon either.

gcoan, 2023-12-24:
Very useful new feature, thanks

iainfogg, 2023-12-24:
Nice one @springfall2008 thanks!

closed 2023-12-24 by iainfogg