springfall2008/batpred #494: Unable to charge AIO to 100%

blastehh, 2023-12-18:
When charging to 100%, predbat will attempt to set reserve to 100% and fail, it will do this continuously meanwhile the battery is charging and discharging constantly without reaching 100%
[image]
[image]

I've since changed the `inverter_reserve_max` setting to 99 which stops the error, but the AIO will charge to 99% and stop.
[image]

It should be enough to set the target SoC to 100% on the AIO and it will charge up to that without needing to set the reserve. The reserve only matters outside of the charging time.
The workaround might be to disable setting the reserve, but then if there's a grid failure during a charge hold, then it won't provide emergency backup?

**Predbat version**

v7.14.15

**Environment details**

- AIO
- Standard HAOS

**Log file**

Log file with errors:
[appdaemon.log.2.txt](https://github.com/springfall2008/batpred/files/13701884/appdaemon.log.2.txt)

Log file with reserve set to 99
[appdaemon.log.1.txt](https://github.com/springfall2008/batpred/files/13701886/appdaemon.log.1.txt)

DJBenson, 2023-12-21:
Just gone back over my logs and spotted the same however I think this is a known issue with some inverters and you can override the maximum SoC in the config file (inverter_reserve_max) to work around this - AFAIK this isn't a Predbat issue.

[image]

blastehh, 2023-12-21:
Yes, I have set it to 99% but that stops it charging to 100%

predbat shouldn't need to touch the reserve setting while charging.

blastehh, 2023-12-22:
https://github.com/springfall2008/batpred/blame/dcbc89f1f058f3362b9e77f5d52560516f7fddb8/apps/predbat/predbat.py#L8467
should this be `inverter.soc_percent` instead of `inverter.soc_percent + 1` ?

but also, removing any reserve fiddling while charging with AIO would be best.

springfall2008, 2023-12-22:
There is a switch called 'switch.set_reserve_enable' which if you turn off will stop Predbat changing the reseve

>should this be inverter.soc_percent instead of inverter.soc_percent + 1 ?

No the + 1 is a workaround to try to prevent discharge when the battery level is being held on the reserve

blastehh, 2023-12-22:
I want it to set reserve to hold the battery level while outside of
charging slot, but this setting will turn it off.

My point is that reserve doesn't need to be adjusted while charging because
the battery doesn't discharge until you leave the charging slot even if you
reach the target soc

dochtie, 2023-12-26:
Agreed. I am seeing the same with my AIO. Always had max reserve set to 99% but predbat could and would charge to 100%. Now it won’t charge above 99%.

PianSom, 2024-01-01:
+1 - my AIO is remarkably unenthusiastic about charging to 100%

blastehh, 2024-01-02:
I'm currently modifying my copy of predbat by deleting:
```python

            # If we should set reserve?
            if self.set_soc_enable and self.set_reserve_enable and not setReserve:
                # In the window then set it, otherwise put it back
                if self.charge_limit_best and (self.minutes_now < inverter.charge_end_time_minutes) and (self.minutes_now >= inverter.charge_start_time_minutes):
                    if self.set_charge_freeze and (self.charge_limit_best[0] == self.reserve):
                        self.log("Adjust reserve to hold current soc {} % (set_reserve_enable is true)".format(inverter.soc_percent))
                        inverter.adjust_reserve(min(inverter.soc_percent + 1, 100))
                    elif inverter.soc_percent >= self.charge_limit_percent_best[0]:
                        self.log("Adjust reserve to hold target charge {} % (set_reserve_enable is true)".format(self.charge_limit_percent_best[0]))
                        inverter.adjust_reserve(min(self.charge_limit_percent_best[0] + 1, 100))
                    else:
                        self.log("Adjust reserve to target charge {} % (set_reserve_enable is true)".format(self.charge_limit_percent_best[0]))
                        inverter.adjust_reserve(self.charge_limit_percent_best[0])
                    resetReserve = False
                else:
                    self.log("Adjust reserve to default (as set_reserve_enable is true)")
                    inverter.adjust_reserve(0)
                    resetReserve = False
```
found at https://github.com/springfall2008/batpred/blob/v7.14.19/apps/predbat/predbat.py#L8659

and also modify the following code from:
```python
                        else:
                            if self.set_soc_enable and self.set_reserve_enable and self.set_reserve_hold and ((inverter.soc_percent + 1) >= self.charge_limit_percent_best[0]):
                                status = "Hold charging"
                                inverter.disable_charge_window()
                                disabled_charge_window = True
                            else:
                                status = "Charging"
                            status_extra = " target {}%".format(self.charge_limit_percent_best[0])
```
to
```python
                        else:
                            if self.set_soc_enable and self.set_reserve_enable and self.set_reserve_hold and ((inverter.soc_percent) >= self.charge_limit_percent_best[0]):
                                status = "Hold charging"
                            else:
                                status = "Charging"
                            status_extra = " target {}%".format(self.charge_limit_percent_best[0])
```
found at https://github.com/springfall2008/batpred/blob/v7.14.19/apps/predbat/predbat.py#L8466

this makes predbat not mess with soc/reserve and it will charge to 100% again.

springfall2008, 2024-01-02:
Surely just toggle off set_reserve_enable?

blastehh, 2024-01-02:
> Surely just toggle off set_reserve_enable?

Set reserve is used for holding the charge level outside of charging. I don't want to disable that feature.
I just want it to stop messing with reserve while charging.

gcoan, 2024-03-27:
@blastehh sorry this issue has been languishing

Its subsequently been identified that there’s an issue with AIO’s and some Gen 2’s and 3’s that don’t like the reserve being set to 100, they reject the setting.  I think this is what’s happening for you.

A new setting in apps.yaml has been introduced to define the max reserve setting to use, default is now 98 to circumvent this inverter issue https://springfall2008.github.io/batpred/apps-yaml/#inverter-reserve-maximum

I know you think its not necessary to set the reserve and have modified your predbat, but could you try the latest predbat and try setting this, see if it works for you?  You’d also then be on a much newer version even if you wanted to re-introduce your changes

springfall2008, 2024-09-01:
Fixed now

closed 2024-09-01 by springfall2008