springfall2008/batpred #496: Predbat not fully charging overnight

iainfogg, 2023-12-19:
For the last two nights, Predbat has not fully charged the batteries, getting to around 90% then going between discharging and charging during the remainder of the IO window.

This is contrary to what the plan was late evening, which was to charge through the whole window.

Normal behaviour is to stay charging until the IO window ends, leaving the battery full for the day.

I'm on 7.14.15.

Givenergy 2*9.5 batteries and hybrid inverter, plus EV (plugged in last night, but not the night before, and both nights didn't charge fully).

[image]

[image]

[image]

[image]

[image]

[image]

Note, calculate plan faster was on, I just turned it off now before the screenshot.

[image]

Logs to follow.

iainfogg, 2023-12-19:
Logs:

```txt
2023-12-19 04:10:00.005149 INFO pred_bat: --------------- PredBat - update at 2023-12-19 04:10:00.003971+00:00 with clock skew 0 minutes, minutes now 250
2023-12-19 04:10:00.006359 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 10.0 minutes
2023-12-19 04:10:00.007459 INFO pred_bat: Predbat version v7.14.15 currently running, latest version is v7.14.15
2023-12-19 04:10:00.017946 INFO pred_bat: Inverter type GE forecast_hours 48 max_windows 48 num_cars 1 debug enable is False calculate_plan_every 10
2023-12-19 04:10:00.019801 INFO pred_bat: Cars 1 charging from battery False planned [False], charging_now [False] smart [False], plan_time ['07:00:00'], battery size [100.0], limit [100.0], rate [7.4]
2023-12-19 04:10:00.021290 INFO pred_bat: Predbat mode is set to Control charge & discharge
2023-12-19 04:10:00.318932 INFO pred_bat: Found 11520 load_today datapoints going back 7 days
2023-12-19 04:10:00.605018 INFO pred_bat: Current data so far today: load 3.56 kWh import 15.5 kWh export 0.0 kWh pv 0.0 kWh
2023-12-19 04:10:00.606388 INFO pred_bat: Fetching futurerate data from None
2023-12-19 04:10:00.617882 INFO pred_bat: Car 0 using Octopus Intelligent, charging limit 80.0, ready time 07:30:00 - battery size 100.0
2023-12-19 04:10:00.619896 INFO pred_bat: Current Car SOC kWh: [35.0]
2023-12-19 04:10:00.630161 INFO pred_bat: Joined Octopus saving session: 2023-11-16T17:30:00+00:00 - 2023-11-16T18:30:00+00:00 at rate 225.0 state False
2023-12-19 04:10:00.631835 INFO pred_bat: Joined Octopus saving session: 2023-11-29T17:00:00+00:00 - 2023-11-29T18:30:00+00:00 at rate 400.0 state False
2023-12-19 04:10:00.633517 INFO pred_bat: Joined Octopus saving session: 2023-12-01T16:30:00+00:00 - 2023-12-01T18:00:00+00:00 at rate 400.0 state False
2023-12-19 04:10:00.635249 INFO pred_bat: Joined Octopus saving session: 2023-12-12T17:00:00+00:00 - 2023-12-12T18:00:00+00:00 at rate 225.0 state False
2023-12-19 04:10:00.636906 INFO pred_bat: Joined Octopus saving session: 2023-12-14T17:00:00+00:00 - 2023-12-14T18:00:00+00:00 at rate 225.0 state False
2023-12-19 04:10:00.638552 INFO pred_bat: Joined Octopus saving session: 2023-12-15T17:00:00+00:00 - 2023-12-15T18:00:00+00:00 at rate 225.0 state False
2023-12-19 04:10:00.640246 INFO pred_bat: Standing charge is set to 0.0 p
2023-12-19 04:10:00.648621 INFO pred_bat: Octopus Intelligent slot at 12-19 01:00:00-12-19 05:00:00 assumed price 7.5
2023-12-19 04:10:01.052238 INFO pred_bat: Rate min forward looking: now 7.499899999999999 at end of forecast 7.5
2023-12-19 04:10:01.054191 INFO pred_bat: Import rates min 7.5 max 29.9 average 24.3
2023-12-19 04:10:01.062082 INFO pred_bat: Export rates min 15.0 max 15.0 average 15.0
2023-12-19 04:10:01.064376 INFO pred_bat: Rate thresholds (for charge/discharge) are import 19.44p (0.8) export 15.0p (1.0)
2023-12-19 04:10:01.069865 INFO pred_bat: High export rate found rates in range 15.0 to 15.0
2023-12-19 04:10:01.073123 INFO pred_bat: Low Import rate found rates in range 7.5 to 7.5
2023-12-19 04:10:01.074982 INFO pred_bat: Car 0 is using Octopus intelligent schedule
2023-12-19 04:10:01.076884 INFO pred_bat: Car 0 charging plan is: [{'start': 60, 'end': 300, 'kwh': 11.56, 'average': 7.5, 'cost': 86.7}]
2023-12-19 04:10:01.116049 INFO pred_bat: Todays energy import 15.5 kWh export 0.0 kWh cost 116.25 p import 116.25 p export 0.0 p
2023-12-19 04:10:01.119854 INFO pred_bat: PV Data for pv_forecast_today total 0.66 kWh
2023-12-19 04:10:01.123659 INFO pred_bat: PV Data for pv_forecast_tomorrow total 0.65 kWh
2023-12-19 04:10:01.127416 INFO pred_bat: PV Data for pv_forecast_d3 total 0.77 kWh
2023-12-19 04:10:01.131188 INFO pred_bat: PV Data for pv_forecast_d4 total 0.65 kWh
2023-12-19 04:10:01.148993 INFO pred_bat: Today's load divergence 100.0 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 0.0 kWh with 8.26 kWh car excluded and 3.31 kWh import ignored and 0.0 forecast extra, Actual so far 0.15 kWh with 0.06 kWh car excluded and 3.5 kWh import ignored
2023-12-19 04:10:01.186123 INFO pred_bat: Historical data totals for days [7] are [13.12] - min 13.12
2023-12-19 04:10:01.188397 INFO pred_bat: WARN: Historical day 7 has 180 minutes of gap in the data, filled from 13.12 kWh to make new average 14.99 kWh (percent 88%)
2023-12-19 04:10:01.192522 INFO pred_bat: Historical days now [7] weight [1]
2023-12-19 04:10:01.193934 INFO pred_bat: Inverter 0 using Rest API http://192.168.68.58:6345
2023-12-19 04:10:02.427105 INFO pred_bat: Invertor time 2023-12-19 04:09:26+00:00 AppDaemon time 2023-12-19 04:10:00.003971+00:00 difference -0.58 minutes
2023-12-19 04:10:02.428738 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2023-12-19 04:10:02.431421 INFO pred_bat: New Inverter 0 with soc_max 19.05 kWh nominal_capacity 19.05 kWh battery rate raw 3600.0 w charge rate 3.6 kW discharge rate 3.6 kW battery_rate_min 200.0 w ac limit 5.0 kW export limit 5.0 kW reserve 4.0 % current_reserve 99.0 %
2023-12-19 04:10:03.664430 INFO pred_bat: Inverter 0 SOC: 17.14 kW 90 % Current charge rate 3600.0 w Current discharge rate 200.0 w Current power -3492.0 w Current voltage 52.0
2023-12-19 04:10:03.666060 INFO pred_bat: Inverter 0 scheduled charge enable is True
2023-12-19 04:10:03.667652 INFO pred_bat: Inverter 0 charge windows currently [{'start': 0, 'end': 330, 'average': 0}, {'start': 1440, 'end': 1770, 'average': 0}, {'start': 2880, 'end': 3210, 'average': 0}]
[... cut ...]
2023-12-19 04:10:57.273718 INFO pred_bat: Inverter 0 charge target 4 via REST successful on retry 0
2023-12-19 04:10:57.316375 INFO pred_bat: Adjust reserve to hold target charge 3 % (set_reserve_enable is true)
2023-12-19 04:10:57.318216 INFO pred_bat: Inverter 0 Current Reserve is 99.0 % and new target is 4 %
2023-12-19 04:11:05.456704 INFO pred_bat: Set inverter 0 reserve 4 via REST successful on retry 0
2023-12-19 04:11:05.493154 INFO pred_bat: Completed run status Freeze discharging, Hold for car

```

iainfogg, 2023-12-20:
Last night it again started the night with a plan to fully charge, but once the battery level got to around 16kwh it stopped charging.

Last night calculate_plan_faster was turned off, everything else was the same.

springfall2008, 2023-12-20:
Can you try turning off the second pass optimisation?

iainfogg, 2023-12-20:
Done, I'll see what that changes tonight, cheers

iainfogg, 2023-12-21:
It was better last night, but it still didn't fully charge, and alternated between discharging and charging after a certain point, just that it was around 18 instead of 16 kWh.

Shout if you want logs again @springfall2008 

[image]

[image]

iainfogg, 2023-12-22:
It was fine last night. The only thing that was different was the car wasn't charging.

At the moment I'm having issues with the car integration, so I've got no car SOC sensor, but I do have a sensor for the supply to the car.

Also, the car was charging on its own timer, rather than with an IO plan (Saladin for to issues with the integration).

Any idea why the car charging in that scenario would seem to be connected to the battery not fully charging?

springfall2008, 2023-12-22:
If the car is charging but Predbat doesn't know about it then the battery will get drained unexpectedly. Can you use car_charging_now sensor instead?

iainfogg, 2023-12-22:
I've currently got no sensors from the car, all I've got is the sensor from the circuit that supplies the socket that is used to charge the car. Predbat can see the amount of load that's gone to the car to work out actual house load, but I can't see how to give it any more than that right now.

The issue isn't that the car is charging and running down the battery though, it's that Predbat is stopping charging the house battery part way through the Intelligent overnight window. For most of the window, the car and house battery are both charging, then the house battery gets to 80-90% and Predbat then cycles between discharging and charging for the rest of the Intelligent overnight window (even though the plan the evening before was to charge to 100%).

Historically the battery would be at 100% by 05:30, but not these two nights where the car was charging. I'm charging the car again tonight, and I'm on the new version. Will see what happens tonight.

iainfogg, 2023-12-23:
Last night I set it to Control Charge, and also updated to .19. It charged normally, and the car charged too (the car is charging on its own timer, not controlled by Predbat).

I'm only guessing it was related to the car, as the two nights it didn't work correctly were nights that the car also charged.

iainfogg, 2024-02-26:
Closing as this is consistently working fine now

closed 2024-02-26 by iainfogg