springfall2008/batpred #497: Batpred plan not optimised for charge when cheap

onetoomany, 2023-12-19:
[image]

There are a few areas of my Batpred plan that confuse me.
1. There is a powerup today at 3pm. However, the plan has the battery charging right up until 2.30pm. I'd have expected that it would either stop charging around 11am or even tactically charge, such that it glides into the powerup nearly empty.
2. At 10.30pm, prices today are 8p. Again, I'm charging from 8pm at 16p such that my battery is already nearly full for this cheap rate.
3. Even ignoring the powerup, prices this morning vary between 15p and 19p. I'm still charging at 19p, thereby sitting on a full battery at 15p - even with battery losses, it would make sense to discharge at 19p rather than charge.

gcoan, 2023-12-19:
@onetoomany I too have a power up event today, and by comparison my predbat has done what I'd expect it to do.  The battery filled up overnight and has discharged this morning. There's some spurious hold charges #491 but other than that, its not charging the battery until the power up period, at 8:30pm to get over a 9pm spike, and then later tonight in the cheap rate
[image]

My load (ASHP) is much higher than yours, but must be something different in the way our predbat's are configured.
Here's my settings if you can see any key differences:
[image]
[image]
[image]

gcoan, 2023-12-19:
@onetoomany I assume you don't have an export tariff or could export and refill for free

springfall2008, 2023-12-19:
@onetoomany can you post your log?

onetoomany, 2023-12-19:
> @onetoomany I assume you don't have an export tariff or could export and refill for free

Correct - no export tariff as mine is a DIY install so I don't have MCS.
My load is probably  slightly lower than I'd expect - I usually use 15kw in a day, so I'd expect an average of 0.25kwh in each period and 1kwh in the three periods at dinner time when I have the oven on.

springfall2008, 2023-12-19:
Curious, I think turning off combine_charge might help

onetoomany, 2023-12-19:
we've another powerup tomorrow, so I'll turn it off and see what happens

onetoomany, 2023-12-19:
The plan is definitely looking better for tomorrow now that I've turned off combine charge.

Only remaining niggle with the plan is a target SOC% (as low as 6%) drops below the "reserve" value set in apps.yaml - in my case 20%; at least, that is what I assume the "reserve" value is for. My inverter's current firmware, for some reason, doesn't allow my battery to go flatter than 20% at the moment.

[image]

Its also interesting how the powerups skew the predicted load based on my behaviour.

onetoomany, 2023-12-20:
[image]

I've still got Predbat doing some confusing behaviour - charging up now at 16p / 17p only to discharge when its 2p cheaper in an hour. The battery is nearly full, so I don't need a topup.

springfall2008, 2023-12-20:
How did you set the reserve value, it should be taken into account but maybe a bug in non-GE inverter setup?

onetoomany, 2023-12-20:
From apps.yaml
```
  battery_power:
    - sensor.sofar_inverter_battery_power
  pv_power:
    - sensor.sofar_solar_power
  load_power:
    - sensor.sofar_house_consumption_power
  soc_kw:
    - sensor.sofar_battery_energy
  soc_max:
    - input_number.sofar_battery_max_charge
  charge_limit:
   - 100
  reserve:
    - 20
  scheduled_charge_enable:
   - off
  scheduled_discharge_enable:
   - off
```

onetoomany, 2023-12-20:
Perfect! That works.

closed 2023-12-20 by onetoomany