springfall2008/batpred #498: Changed to Error - metric_octopus_import not set correctly or no energy rates can be read

timtheGoat, 2023-12-19:
**Describe the bug**
Changed to Error - metric_octopus_import not set correctly or no energy rates can be read

**Expected behavior**
I expected Predbat to run without an error.

**Predbat version**

v7.14.15

**Environment details**

- Inverter and battery - Givenergy 5kW gen 2 Hybrid 2 9.5 kWh batteries
- Standard HAOS installer

gcoan, 2023-12-19:
Is the Octopus integration working, have you overridden/ manually setup rates in apps.yaml?

Could you please share the log file and apps.yaml config of your rates

timtheGoat, 2023-12-19:
Hi,
   The octopus integration seems to be working.
I think that the following could be the cause:
   my octopus integration sensor is:
sensor.electricity_20j0041185_1415491290001_current_rate
  and in the configuration it has the prefix *octopus_energy*:
metric_octopus_import:
're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'

*Best Regards*

*Tim Smith *
- Technical Director

*Unit 46 Queens Court Trading Estate*
*Greets Green Road*
*West Bromwich*
*West Midlands*
*B70 9EQ*
*T*: 0121 557 3837 *F*: 0121 520 6226 *M*: 07747025415 *W*:
www.gotronic.co.uk

For all accounts inquiries, please contact  the accounts team at
***@***.***

Gotronic Ltd is a company registered in England & Wales with company number
3776064

Registered Office: As Above Vat No. 729 9424 92

IMPORTANT: The contents of this email and any attachments are confidential.
They are intended for the named recipient(s) only.

If you have received this email by mistake, please notify the sender
immediately and do not disclose the contents to anyone or make copies
thereof.

gcoan, 2023-12-19:
My regular expression is the same but my sensors are all named octopus_energy_electricity...

I think the regular expression should cope with this but how about you delete the (octopus_energy_|) bit from the expression and the same in the export line and see if it works.

In the app daemon log it should say if it matches ok

Cheers Geoffrey

timtheGoat, 2023-12-19:
Hi,
    I have tried deleting that bit, it's still the same. I've also tried
putting the actual sensor name between the '', still the same.

springfall2008, 2023-12-19:
Can you snapshot these two, the event and the sensor?

[image]
[image]

springfall2008, 2023-12-19:
And then post your log?

timtheGoat, 2023-12-19:
Hi,
    Where do I find the log that you need?

timtheGoat, 2023-12-19:
Hi,
    I hope this is what you are looking for.
I do not have a sensor for the second item with the octopus_energy prefix.

*Best Regards*

*Tim Smith *
- Technical Director

*Unit 46 Queens Court Trading Estate*
*Greets Green Road*
*West Bromwich*
*West Midlands*
*B70 9EQ*
*T*: 0121 557 3837 *F*: 0121 520 6226 *M*: 07747025415 *W*:
www.gotronic.co.uk

For all accounts inquiries, please contact  the accounts team at
***@***.***

Gotronic Ltd is a company registered in England & Wales with company number
3776064

Registered Office: As Above Vat No. 729 9424 92

IMPORTANT: The contents of this email and any attachments are confidential.
They are intended for the named recipient(s) only.

If you have received this email by mistake, please notify the sender
immediately and do not disclose the contents to anyone or make copies
thereof.

state_class: total
mpan: "1415491290001"
serial_number: 20J0041185
is_export: false
is_smart_meter: true
tariff: E-1R-INTELLI-VAR-22-10-14-E
start: "2023-12-19T05:30:00+00:00"
end: "2023-12-19T23:30:00+00:00"
is_capped: false
is_intelligent_adjusted: false
current_day_min_rate: 0.074999
current_day_max_rate: 0.299037
current_day_average_rate: 0.243027
data_last_retrieved: "2023-12-19T14:00:56.822665+00:00"
last_evaluated: "2023-12-19T14:23:56.823638+00:00"
unit_of_measurement: GBP/kWh
device_class: monetary
icon: mdi:currency-gbp
friendly_name: Electricity 20J0041185 1415491290001 Current Rate

event_types:
  - octopus_energy_electricity_current_day_rates
event_type: octopus_energy_electricity_current_day_rates
rates:
  - start: "2023-12-19T00:00:00+00:00"
    end: "2023-12-19T00:30:00+00:00"
    value_inc_vat: 0.074999
    is_capped: false
  - start: "2023-12-19T00:30:00+00:00"
    end: "2023-12-19T01:00:00+00:00"
    value_inc_vat: 0.074999
    is_capped: false
  - start: "2023-12-19T01:00:00+00:00"
    end: "2023-12-19T01:30:00+00:00"
    value_inc_vat: 0.074999
    is_capped: false
  - start: "2023-12-19T01:30:00+00:00"
    end: "2023-12-19T02:00:00+00:00"
[... 157 lines cut ...]
    is_capped: false
  - start: "2023-12-19T21:30:00+00:00"
    end: "2023-12-19T22:00:00+00:00"
    value_inc_vat: 0.299037
    is_capped: false
  - start: "2023-12-19T22:00:00+00:00"
    end: "2023-12-19T22:30:00+00:00"
    value_inc_vat: 0.299037
    is_capped: false
  - start: "2023-12-19T22:30:00+00:00"
    end: "2023-12-19T23:00:00+00:00"
    value_inc_vat: 0.299037
    is_capped: false
  - start: "2023-12-19T23:00:00+00:00"
    end: "2023-12-19T23:30:00+00:00"
    value_inc_vat: 0.299037
    is_capped: false
  - start: "2023-12-19T23:30:00+00:00"
    end: "2023-12-20T00:00:00+00:00"
    value_inc_vat: 0.074999
    is_capped: false
mpan: "1415491290001"
serial_number: 20J0041185
tariff_code: E-1R-INTELLI-VAR-22-10-14-E
friendly_name: Electricity 20J0041185 1415491290001 Current Day Rates

springfall2008, 2023-12-19:
>I do not have a sensor for the second item with the octopus_energy prefix.

This would explain the issue, the sensor name is used to figure out the event name. 

Can I suggest you remove the Octopus add-in, reboot home assistant, delete the old sensor and then install it again and hopefully the names will be fixed?

timtheGoat, 2023-12-19:
Hi,
     Thanks, that's sorted it out.

gcoan, 2023-12-19:
I'll add something to the FAQ's to cover this in case anyone else gets it happen to them

closed 2023-12-22 by gcoan