springfall2008/batpred #507: Exception raised invalid literal for int() with base 10: '20.0'

onetoomany, 2023-12-20:
Logs attached 
[appdaemon.log](https://github.com/springfall2008/batpred/files/13731303/appdaemon.log)

```
ValueError: invalid literal for int() with base 10: '20.0'

2023-12-20 18:55:01.514683 WARNING pred_bat: ------------------------------------------------------------
2023-12-20 19:00:05.573750 WARNING pred_bat: ------------------------------------------------------------
2023-12-20 19:00:05.573888 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-12-20 19:00:05.573988 WARNING pred_bat: Worker Ags: {'id': '42ab963081c9483a8f52fa55d20fdcab', 'name': 'pred_bat', 'objectid': '80b0777080b04484b936a45f1b7f54c3', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fa2460be390>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2023-12-20 19:00:05.574058 WARNING pred_bat: ------------------------------------------------------------
2023-12-20 19:00:05.574874 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10166, in run_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10162, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9473, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8618, in execute_plan
    inverter.adjust_battery_target(self.charge_limit_percent_best[0])
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 1338, in adjust_battery_target
    current_soc = int(self.base.get_arg("charge_limit", index=self.id))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```
Also at 2am, I am discharging when its -ve 1.4p, having just charged up in the slots before. Not sure if its connected
[image]

springfall2008, 2023-12-20:
What is charge limit set to in apps.yaml?

onetoomany, 2023-12-20:
```
  charge_limit:
   - 100
```

springfall2008, 2023-12-20:
Should be fixed here: https://github.com/springfall2008/batpred/releases/tag/v7.14.19

onetoomany, 2023-12-20:
That's probably a record bug fix! Testing now

onetoomany, 2023-12-20:
The error looks like its gone. Many thanks.

The plan, though, is still sub optimal with the negative pricing tonight
[image]
I'd have expected low level charging between 2230 and 0130, and then fill the battery at negative pricing.

springfall2008, 2023-12-20:
Can you share a log? What do you have best soc keep set to?

onetoomany, 2023-12-20:
Best SOC Keep is 2.
Logfiles attached

[appdaemon.log](https://github.com/springfall2008/batpred/files/13733123/appdaemon.log)

slopemad, 2023-12-21:
To me, that looks like it's discharging a *full* battery at 01:30-02:30 in order to make room for a bit of charge when prices are -3.93, and thus maximise the amount of -3.93p electricity you can buy.  Of course, what it really should be doing is discharging so that it is at 4% SOC by 01:30 (or 02:00) so you can buy lots of negative price power during the plunge.  I guess it wont do that if your export rates are zero?

onetoomany, 2023-12-21:
> To me, that looks like it's discharging a _full_ battery at 01:30-02:30 in order to make room for a bit of charge when prices are -3.93, and thus maximise the amount of -3.93p electricity you can [...]

Yes, come 01:30, it would be logical for the battery dump if its still has charge then; however, I wouldn't expect the battery to **plan** to charge up just beforehand at cost, only to mildly dump it the period after. The discharge is also only by 1%, 2% and then 3%.

springfall2008, 2023-12-24:
I don't think export will ever trigger at 0 rate, you would have to set it to 0.1 to allow it.

closed 2023-12-28 by springfall2008