springfall2008/batpred #509: Optimal Charge Slot Determination

JonathanLew1s, 2023-12-20:
**Describe the bug**

There are 6 negative slots overnight, predbat is chosing to discharge in 3 of them despite battery SOC not being at maximum and to compound there is a choice to recharge later at rate, that is 9p higher than the lowest slot.

**Expected behavior**
In this example, 23:00 could be a discharge slot reducing SOC by 8.5% and freeing enough headroom to use all the negative rate slots to import. Allowing a later discharge to recover costs and bring SOC down to planned demand for tomorrow. That 8.5% charge at 23:00 is simply not needed to cover the demand, maintain SOC or optimise cost.
21:30 I could also argue shouldn't be such a deep charge, to 16%, when only .4kWh/2% is required to bridge the 22:00 to 22:30 slot. SOC keep is 0.1 so only maintaining a 4% reserve.

**Predbat version**

7.14.18

**Environment details**

- 5kWh Hybrid Gen 2, 2 x 9.5kWh batteries
- Docker

**Screenshots**

[image]
[image]
[image]
[image]

**Log file**
[_appdaemon_logs-7.txt](https://github.com/springfall2008/batpred/files/13732534/_appdaemon_logs-7.txt)

springfall2008, 2023-12-20:
I just pushed a version to main to address this exact issue, would you mind trying it if you get time?

JonathanLew1s, 2023-12-20:
Looks much better, for the negative slots.

Still challenge that the 21:30 to 18% is way too high and also not possible. 30 minute max charge would be around 8% on a 3.6kWh charge rate. It's aiming for 14%

[image]

springfall2008, 2023-12-20:
I wouldn’t read too much into the target % as it’s clipped with some margin, it says it goes from 5% to 14% which seems okay doesn’t it?

JonathanLew1s, 2023-12-20:
Agree.

I think 22:30 to 00:00 is still a little janky, 22:30 could be a full charge 23:00 a full discharge. 23:30 is close to 1p cheaper than 00:00.

It's all fine margins, and small gains. I'm happy now that the negative slots are fully leveraged. Plan comes out 4p worse off but with an SOC 14% higher. 

Thanks for the quick reply.

gcoan, 2023-12-20:
@springfall2008 whats changed in the main release?

TBH my plan for tonight is quite reasonable already.  Yes it's trying to discharge an already empty battery until 2am when it could be charging and discharging, but for the pence gained I'm not bothered and it is more battery cycles.

Charges in 5 of the 6 slots, lets the soc drop in the 5th, then a bit of charging and discharging to generate income.

Fairly ok

[image]

springfall2008, 2023-12-20:
The change on main (now https://github.com/springfall2008/batpred/releases/tag/v7.14.19) when selecting export slots all at the same export rate will select from those with the most expensive import rates first. Not sure if it will change your plan to be honest.

gcoan, 2023-12-20:
OK, I've installed 7.14.19 now.  Will just have to re-apply my pending changes to the code.

The new plan is the same in the negative period, but has introduced a bit more charge/discharge this side of midnight for a few pence extra income
[image]

closed 2023-12-20 by JonathanLew1s