springfall2008/batpred #511: PV shown in plan (presumably calculation too) is wrong

Philhews, 2023-12-21:
**Describe the bug**
The expected PV value is wrong for the time period causing an incorrect expectation. Screenshot show 8.30 this morning with PV forecast even on the 90 value to be less than 0.1 but the plan is showing 0.18. Not a big difference here but proportionally it is. 

**Expected behavior**
PV value in the plan should be what solcast is predicting (or maybe modified based on the 10% weight I don't know exactly what it's supposed to be) 

**Predbat version**

7.14.15

**Environment details**

- Gen 2 3.6 hybrid. 9.5kwh battery
- standard HAOS

**Screenshots**
[image]
[image]

springfall2008, 2023-12-21:
Do you have the log for that period, or if not the latest log?

Philhews, 2023-12-21:
Yes, though I was struggling to work out how to lift the file from my phone but at worst I'll copy the section out when I get a second.

Philhews, 2023-12-21:
[appdaemon.log](https://github.com/springfall2008/batpred/files/13743912/appdaemon.log)

Latest log file, this mornings ones are no longer available

Philhews, 2023-12-21:
Will include tomorrow morning though which the plan shows 0.02 when all of the estimates for that half hour are essentially 0.00
[image]
[image]

springfall2008, 2023-12-22:
Can you double check your input_number.pv_scaling setting in case it's set wrongly?

It's worth noting the values you get from solcast are actually not for the half hour but the power (in watts) so actually they are more like hourly figures - this makes the difference even worse. I did check my system and I don't see this issue.

Are all figures too big or only these low ones?

Philhews, 2023-12-22:
Pv scaling is set to 1.

I can't work out how predbat is getting to its numbers, they seem to not follow the same formula from the pv10, 50 and 90 values for each slot at all. Plan and solcast for daylight tomorrow attached along with l a log file. 

[image]
[image]
[image]
[image]
[image]
[appdaemon.log](https://github.com/springfall2008/batpred/files/13752889/appdaemon.log)

Philhews, 2023-12-22:
Metric 10 weight is 0.2 if that matters.

Philhews, 2023-12-23:
I wonder if there's a time shift going on somehow, I have 0.02 in the plan for the 7:30 slot this morning where all values in the solcast slot are 0

springfall2008, 2023-12-23:
Hopefully fixed in the latest release

Philhews, 2023-12-24:
Yep, looks better, thanks

springfall2008, 2023-12-26:
Hopefully fixed this in release now: https://github.com/springfall2008/batpred/releases/tag/v7.14.25

closed 2023-12-26 by springfall2008