springfall2008/batpred #512: Unnecessary battery charging before Savings Event

Bighton, 2023-12-21:
**Describe the bug**
I use Predbat in the Control Charge mode and I only wish to charge from grid during cheap rate periods. I do not have an export tariff so have no need to export.
When an Octopus Saving event is recognised Predbat initiates a SOC hold and plans import during peak rate periods, I assume to maximise the potential export during the Savings event.

**Expected behavior**
With a simple set up of PV, battery and Intelligent Octopus with no export tariff, the easiest solution for me is to just let the SOC diminish in tandem with the house load as per normal since any Octopoint gains are financially minimal.
I realise that I could just leave the Octopus scheme, but is there a way to disable the Predbat auto joining feature?

**Predbat version**
7.14.6

**Environment details**
- Inverter GIV-AC3.0 Battery 8.19kWh
- Standard HAOS

**Screenshots**
[Screenshot 2023-12-19 at 15.40.30.pdf](https://github.com/springfall2008/batpred/files/13741473/Screenshot.2023-12-19.at.15.40.30.pdf)

gcoan, 2023-12-21:
It looks to me that predbat is doing what you wanted from the expected behaviour.

There's no importing going on on this plan until some time in the 19:30 period when the soc reaches the 27% limit predbat has set.  That limit may be set by your min soc keep, or it may actually drop the limit when it gets nearer to the execution time.

During the actual saving event, you can see the rates go up, but predbat plan is just a downwards arrow which is Eco mode.  It's not doing a forced discharge.

Do you have calculate plan faster on or calculate full second pass on?  What's min soc keep?

If you wanted to disable automatically joining the event you might be also to achieve this by disabling the octopus_energy.join_octoplus_saving_session_event service, but this might cause predbat to fail when it tries to join it.  Since you might earn something from using less grid draw in the period (eg putting the kettle on that exceeds your inverter discharge rate), it's no harm to leave it joining as long as predbat doesn't do anything untoward. 

Geoffrey

springfall2008, 2023-12-22:
>If you wanted to disable automatically joining the event you might be also to achieve this by disabling the octopus_energy.join_octoplus_saving_session_event service, but this might cause predbat to fail when it tries to join it.

You can just comment out the saving session sensor in apps.yaml and Predbat will ignore saving sessions totally, you can still join manually of course.

Bighton, 2023-12-22:
Thank you for your help. Your suggestion is the perfect solution for my needs.

closed 2024-01-02 by Bighton