springfall2008/batpred #513: Changing mode does not stop/interrupt the current mode

DJBenson, 2023-12-21:
**Describe the bug**
When the mode is changed from say "Control charge & discharge" to "Control charge" and there is an active discharge action, that action is not interrupted.

**Expected behavior**
I would expect Predbat to respond to such a change by stopping the export immediately. The only way I could get the system to stop discharging was by modifying GivTCP settings directly (turned off timed export).

**Predbat version**
7.14.19

**Environment details**

- GivEnergy AIO
- HAOS (up to date)

springfall2008, 2023-12-22:
I can see the confusion, I'm afraid this is kind of by design in that if Predbat is not set to control discharge then it won't control them. 

I suspect I could do something to cancel discharge when the mode is changed for safety reasons so will keep this ticket open while I think on it. The same issue applies to read-only mode where the current charge/discharge will be left as-is.

DJBenson, 2023-12-22:
Thanks Trefor. For the less advanced user I could see this causing panic as it's not immediately obvious how to stop the export.

springfall2008, 2023-12-23:
Fixed in the current release

closed 2023-12-26 by springfall2008