springfall2008/batpred #514: iBoost Immersion boost control based on when agile rates are cheaper than Gas

AGLAISTER, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
I have an iboost on which I have stuck a fingerbot which i can control in a basic way based only on the current active agile rate slot. 
I found material online that said electric immersions are 100% efficient at energy exchange vs gas boiler heating water at roughly 85%
I've created sensors to help work out a threshold at which the electric heating of the water is cheaper than gas - these are below 
sensor:
  - platform: template
    sensors:
      octopus_gas_rate:
        friendly_name: "Octopus Gas Rate"
        value_template: "{{ states('sensor.octopus_energy_gas_xxxxxxxxxxxxxx_xxxxxxxxxxxxxxx_current_rate') }}"
        unit_of_measurement: "pence per kWh"

      gas_rate_multiplier:
        friendly_name: "Gas Rate Multiplier"
        icon_template: mdi:percent
        unit_of_measurement: "%"
        value_template: "{{ states('input_number.immersion_thresold') | float(default=0) / 100 }}"

      calculated_gas_rate:
        friendly_name: "Calculated Gas Rate"
        unit_of_measurement: "pence per kWh"
        value_template: "{{ (states('sensor.octopus_gas_rate') | float(default=0) * states('sensor.gas_rate_multiplier') | float(default=0)) | round(2) }}"

**Describe the solution you'd like**
To be able to find the periods in the day where it would be cheaper to use electricity and turn on the immersion boots feature on my iboost using the fingerbot
I'd like to be able to do boost that happen at the time of the cheapest slots on my agile rates with the following priority logic order
1) lowest available negative slots
2) zero (free) slots
3) slots with rates lower than 'calculated_gas_rate'

it would be great if there could be a status that would sit alongside the predbat plan that has something like
immersion_boost_status (On/Off)
immersion_boost_time values 15(default), 30(only selected if there are 2 consecutive slots meeting the criteria 1),2) or 3)

if this isn't possible then just some guidance on how i can use the agile rates code to be able construct my own python of how to do the above

springfall2008, 2023-12-23:
Now in this release:

https://github.com/springfall2008/batpred/releases/tag/v7.14.21

You need to set your gas rates in apps.yaml

```
  # Gas rates for comparison
  metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'
```

Then Enable and configure the iboost model, set your max rate and max kwh.

You can enable iboost on gas (for lower than gas rates) and tuning the gas scaling to adjust for losses
If you enable iboost on charging then iboost will be assumed to happen only when the battery charges

Finally use binary_sensor.iboost_active to trigger your iboost hardware

AGLAISTER, 2023-12-23:
Hi Trefor thanks for this is amazing you going to this effort.

Sorry for this but I have quite a few questions to help me work how to configure it.

1) iboost_solar - does this have to be enableed for the iBoost_gas to be considered?
2) iboost_min_soc - when it says enabled what does this mean is it enabling the calculation or mean turning the iboost on?
3) iboost_gas - is this enabling the calculation or turning the iboost on/off via the binary sensor?
4) iboost_gas_scale - is this the same as my % sensor? i.e. if I want electric rates to be considered when 70% of the gas rates do I set it to 0.7 or to 1.3?
5) iboost_charging - is this the same as my setting to only do the manual boost when the battery is charging?
6) iboost_max_energy - I assume this is the same as before and used for the modelling of what the iboost is likely to consume if PV is good
7) iboost_min_soc - if I set this to 100% will it assume that the iBoost takes no power from the PV until  the battery is at 100%
8) How does the algorithm work does it prioritize negative rates over 0 rates over being lower than gas rates or is is simply when the rates are lower than gas rates?

springfall2008, 2023-12-23:
> iboost_solar - does this have to be enableed for the iBoost_gas to be considered?

No they are independent, the solar option models using solar power for iboost only.

> iboost_min_soc - when it says enabled what does this mean is it enabling the calculation or mean turning the iboost on?

This just controls when the solar iboost is modelled to be enabled

> iboost_gas - is this enabling the calculation or turning the iboost on/off via the binary sensor?

No the binary sensor includes any type of iboost (solar, gas or charging) and reflects when the model says iboost should be enabled and hence you can use it to enable your controller.

> iboost_gas_scale - is this the same as my % sensor? i.e. if I want electric rates to be considered when 70% of the gas rates do I set it to 0.7 or to 1.3?

This is multiplied by the gas price before comparing to the electric price, so I guess 0.7 in your example. However given a gas boiler has losses which are higher than electric this might not be the right value.

> iboost_charging - is this the same as my setting to only do the manual boost when the battery is charging?

Yes

> iboost_max_energy - I assume this is the same as before and used for the modelling of what the iboost is likely to consume if PV is good

It's just the maximum per day, covers all three iboost types - predbat tracks the amount in a day.

> iboost_min_soc - if I set this to 100% will it assume that the iBoost takes no power from the PV until the battery is at 100%
Yes

> How does the algorithm work does it prioritize negative rates over 0 rates over being lower than gas rates or is is simply when the rates are lower than gas rates?

It's simply when the rates are lower than the gas rates, or when it's lower than the gas rates and the battery is charging (if you enable the charging toggle). Given predbat will charge on the cheapest rates possible anyhow surely that covers both scenarios?

AGLAISTER, 2023-12-23:
Hi Trefor - brilliant once more.

I've not gotten my head round things completely (just looking at the code now) my only thought was if there were slots where there are slots where the price is between 0 and the gas rate prior to negative pricing but due to there being not many negative slots there are charging slots on the predbat plan

However I guess I can control that with setting the gas scaling rate to something like 0.6 or 0.5 to control for the losses

springfall2008, 2023-12-23:
> Hi Trefor - brilliant once more.
>
> I've not gotten my head round things completely (just looking at the code now) my only thought was if there were slots where there are slots where the price is [...]

Yes true, the difficulty with just trying to pick the cheapest slots in the day is that those slots may never arrive.

Surely a gas boiler is ony 80% efficient, not sure how efficient an emersion heater is?

AGLAISTER, 2023-12-23:
Very true on the slots

From what I've seen from my searches is that 1kwh of immersion energy puts in 100% of that heat where as a boiler at 80% efficiency only puts in 0.8kw of heat
I used this video to try to understand https://www.youtube.com/watch?v=m6cZLgitb0s and that is what I based my automation off.

Once thing - I'd love to be able to see the pence/kw calculation for the gas rate used in the comparison ( gas_rate = self.rate_gas.get(minute_absolute, 99) * self.iboost_gas_scale) the dashboard - is there a way I can expose that outside predbat?

springfall2008, 2023-12-23:
>Once thing - I'd love to be able to see the pence/kw calculation for the gas rate used in the comparison ( gas_rate = self.rate_gas.get(minute_absolute, 99) * self.iboost_gas_scale) the dashboard - is there a way I can expose that outside predbat?

I can probably add this to the next release, but it's very easy to calculate as most people's gas rates are fixed and the scale factor is already a HA entity.

AGLAISTER, 2023-12-23:
Thanks Trefor.

I tried to quickly set it up but was having issues -  I'll look again when I get chance.

Good news the iboost active has changed which is fab.
Trouble is the automations want to see either on or off but the sensor itself seems to be showing true or false so my automation didn't trigger as I'd put it to detect moving from off - on

I went to put it as a condition but that is also looking for on/off/unavailable or unknown

Am I missing something?

AGLAISTER, 2023-12-28:
Hi Trefor,

I'm finding the binary_sensor.iboost_active is not working in the way I was hoping

My automation triggers a 15 minute boost on the immersion.

I've found in trialing the code you kindly added it was tiggering on the first few slots and then not triggering later because of the max threshold in the code which I have set to the max 5kw

The result of this is like today where i have cheaper slots than the next morning the iboost_active won't trigger

[image]

To overcome this i've had to switch back to my rudimentary automation.

Would commenting out these 2 lines enable the iboost active independently of the max value?
                if iboost_today_kwh < self.iboost_max_energy:
                if iboost_today_kwh < self.iboost_max_energy:

springfall2008, 2023-12-28:
I guess I could find the cheapest slots that add up to max in the 24 hour period.

but setting max to something crazy big eg 99 in HA would be the same as commenting out these lines

AGLAISTER, 2023-12-28:
Hi trefor,

The input maxes out at 5 so I can't change it any higher

springfall2008, 2023-12-28:
> Hi trefor, The input maxes out at 5 so I can't change it any higher

If you just type in the number and hit enter it will change regardless, just ignore the red line

AGLAISTER, 2023-12-28:
Thanks Trefor,

I've tried that (after updating to the latest version) 
If I'm interpreting the plan correctly it still doesn't seem to be setting a trigger for the immersion

[image]

springfall2008, 2023-12-28:
Did you change the maximum or the current value? Either way at 1:30 the iboost is running, the number (0.0) is the value at the start of the time window and the up arrow means it's boosting. The total in the 1:30 slot was 1.2 as reported at 2am

AGLAISTER, 2023-12-28:
DOH!!! :)
I Changed the amount for today by mistake. Now just changed the max to 99 but its still looks different to what i'd expect

[image]

Don't understand why the 23:30 doesn't seem to have a trigger

springfall2008, 2023-12-28:
Oddly 23;30 probably does have a trigger but the html plan doesn’t show it as the total resets to 0 at midnight

closed 2024-02-03 by springfall2008