springfall2008/batpred #516: Charge/Discharge plan

slopemad, 2023-12-23:
I'm not entirely convinced by this plan to discharge all my battery now, and recharge during the afternoon peak.  (the plan has been absolutely brilliant, up until just now. I'm on 7.14.19 and I don't have the second pass or tweak calculations enabled.  I've noticed more recently that the 'far ahead' plan is great, but they go to pot when you get closer to the time.

[image]

springfall2008, 2023-12-23:
Wtf!?!

can you share a log?

onetoomany, 2023-12-23:
My plan is even worse - free electricity this afternoon with a powerup and not bothering to charge for the evening peak
[appdaemon.log](https://github.com/springfall2008/batpred/files/13759165/appdaemon.log)

[image]

springfall2008, 2023-12-23:
The issue here is you have zero load predicted, something is wrong with the historical data

springfall2008, 2023-12-23:
Maybe try changing days_previous to:

days_previous:
  - 5
  - 6
  - 7

onetoomany, 2023-12-23:
> Maybe try changing days_previous to:
> 
> ```
> days_previous:
>   - 5
>   - 6
>   - 7
> ```

Ah - my Octopus mini stopped communicating a few days ago!

slopemad, 2023-12-23:
Alas, I didn't catch a log and it has long since rotated out.  i just operated it in read only mode with something more sane going on instead.  Now just trying to work out what to do with tomorrow's plungeathon.

slopemad, 2023-12-23:
Incidentally, I changed my days_previous to not include the past couple of days as I was away and so my usage wasn't reflective of what was likely to happen today, but it did occur to me - it might be worth filtering out usage during savings sessions and infilling from days where there was no savings session, because usage (during peak period) is being deliberately kept low, and when we've had so many of them lately, it means Predbat has a distorted view of reality in the 4pm-7pm slot.

springfall2008, 2023-12-23:
> Incidentally, I changed my days_previous to not include the past couple of days as I was away and so my usage wasn't reflective of what was likely to happen today, but it did occur to me - it might [...]

Usually saving sessions don't have much impact on baseline usage for battery users, so averaging a few days tends to resolve this.

dandwhelan, 2023-12-27:
I've a similar situation. I've no idea why it's discharging at peak pricing. 

[image]
[image]
[image]

dandwhelan, 2023-12-27:
It's as if it's learnt export saver session discharge and wants to keep it up. Id rather power the house from the battery during this time period

springfall2008, 2023-12-27:
I don't specifically see the issue isn't the 12p export giving you a profit as you charge later at a cheaper rate?

dandwhelan, 2023-12-27:
> I don't specifically see the issue isn't the 12p export giving you a profit as you charge later at a cheaper rate?

It's 29p to import at the moment. 12p to export.

springfall2008, 2023-12-27:
> > I don't specifically see the issue isn't the 12p export giving you a profit as you charge later at a cheaper rate?
> 
> It's 29p to import at the moment. 12p to export.

But the charging isn't happing now you still have battery left don't you?

dandwhelan, 2023-12-27:
Yeah but why would it export during the peak times. Especially from
historical data it's my peak daytime power usage. With the oven and
electric hob on.

gcoan, 2023-12-27:
> Yeah but why would it export during the peak times. Especially from
> 
> historical data it's my peak daytime power usage. With the oven and
> 
> electric hob on.
> 
> 
> 
> 
> 
> 
> 
> > I don't specifically see the issue isn't the 12p export giving you a
> 
> > profit as you charge later at a cheaper rate?
> 
> >
> 
> > It's 29p to import at the moment. 12p to export.
> 
> >
> 
> > But the charging isn't happing now you still have battery left don't you?

It's hard to see on the screen shots what the state of soc gets down to before the next charge period, can you take some more showing the full width of the plan up to the next charging period.

Having said that, Predbat is looking to maximise your income/minimise your expenditure.

So if it calculates that your battery is sufficient to support your historical load it will discharge what it sees as "excess" soc and generate you more export income. That discharging will be done in peak period if that's when you get the best export rate.

dandwhelan, 2023-12-27:
Cheers for the replies. I'm not sure how to get the old data. I'll setup an automation to change mode from charge and discharge to charge only for the three hour period. That will do the job 👍

springfall2008, 2023-12-27:
There are numerous ways to prevent export if that's what you want to do, when do you actually want to force export and when don't you want to?

dandwhelan, 2023-12-27:
Id rather it didn't export between the hours of 4pm and 7pm

springfall2008, 2023-12-27:
> Id rather it didn't export between the hours of 4pm and 7pm
> […](#)

Why is that? I'm trying to understand your use case.

dandwhelan, 2023-12-27:
I discovered this cool project when looking for a better way to manage my battery after moving over from Octopus Go to Octopus Agile. As you know Octopus Agile price changes every 30 minutes. Between the hours of 4pm and 7pm is the peak price with Agile. Usually costing between 25p to 70p. 

I have a 8.2KW battery and while I was on Go the battery with no management would last until 5pm when it got flat. The software is doing a great got managing the SOC but seems to insist I discharge the 4 KW I have left over at 4pm so it's flat for tea time, when we use peak energy.

springfall2008, 2023-12-27:
Did you set metric keep to something non-zero e.g. 0.5 or 1?

dandwhelan, 2023-12-27:
If your talking about "Best SOC Keep" - I kept this at 0.5. 
In the .yaml after posting earlier I change this...  days_previous_weight: - 0.5
I think the software thinks I enjoyed the battery being discharged at 5pm

springfall2008, 2023-12-27:
Can you show the full 24 hours of the plan as your clip from earlier still showed battery left after the discharge?

dandwhelan, 2023-12-27:
Here is a picture of tomorrow with more of the same. 
[image]

springfall2008, 2023-12-27:
The plan looks okay to me, you don’t run out of battery even with the discharge. Of course you can raise the keep threshold or increase your load scaling if you want to be more pessimistic

closed 2024-02-10 by springfall2008