springfall2008/batpred #527: pred_bat: ERROR: Exception raised 'NoneType' object has no attribute 'get'

dandwhelan, 2023-12-25:
**Describe the bug**
I have a new install of predbat and everytime the appdeamon processes the givenergy / agile times I get this error...

ERROR: Exception raised 'NoneType' object has no attribute 'get'

**Expected behavior**
No error
**Predbat version**

7.14.23

**Environment details**

- Inverter and battery setup - Gen 2 H5 GivEnergy  - 8.2Kw Battery
- Standard HAOS installer or Docker - Standrad HAOS Pi
- Anything else?

**Screenshots**

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/13766543/appdaemon.log)

Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2023-12-25:
I probably need the more detailed error message that you will find in 'system, settings, logs, appdeamon' to find out exactly the problem, but it seems like you might not have set your import or export rates correctly. If you don't have export you should set it to 0.

dandwhelan, 2023-12-25:
Hey, this might be relevant... Does it help?

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in
basic_rates
    start_str = this_rate.get("start", "00:00:00")
                ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

2023-12-25 15:50:04.468790 WARNING pred_bat:
------------------------------------------------------------

dandwhelan, 2023-12-25:
File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10523, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9766, in update_pred self.fetch_sensor_data() File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9357, in fetch_sensor_data self.rate_export = self.basic_rates(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 16:35:03.783425 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:40:04.044928 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:40:04.047180 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2023-12-25 16:40:04.048795 WARNING pred_bat: Worker Ags: {'id': 'b21a4515aea74f6fb476a3817a8c3233', 'name': 'pred_bat', 'objectid': 'ae65d3fcd06144d488a222d829065e05', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fb8b97150>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}} 2023-12-25 16:40:04.050078 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:40:04.057479 WARNING pred_bat: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"])) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10527, in run_time_loop raise e File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10523, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9766, in update_pred self.fetch_sensor_data() File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9357, in fetch_sensor_data self.rate_export = self.basic_rates(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 16:40:04.057966 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:45:03.613576 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:45:03.614684 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2023-12-25 16:45:03.615374 WARNING pred_bat: Worker Ags: {'id': 'b21a4515aea74f6fb476a3817a8c3233', 'name': 'pred_bat', 'objectid': 'ae65d3fcd06144d488a222d829065e05', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fb8b97150>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}} 2023-12-25 16:45:03.616283 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:45:03.621972 WARNING pred_bat: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"])) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10527, in run_time_loop raise e File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10523, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9766, in update_pred self.fetch_sensor_data() File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9357, in fetch_sensor_data self.rate_export = self.basic_rates(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 16:45:03.623503 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:50:04.063567 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 16:50:04.064236 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2023-12-25 16:50:04.064787 WARNING pred_bat: Worker Ags: {'id': 'b21a4 [... cut ...] es(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 16:50:04.074073 WARNING pred_bat: ------------------------------------------------------------

springfall2008, 2023-12-25:
What is in your apps.yaml related to rates as it’s an errror in this file?

dandwhelan, 2023-12-25:
I'd be embarrassed if its something I've done. I wouldn't be suprised.

 I even got a fresh copy of the yaml file from github earlier on the day
just in case I'd ruined the previous.

# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat
  module predbat
  class PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix predbat
  # Timezone to work in
  timezone Europe/London

  # XXX This is a configuration template, delete this line once you edit your configuration

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial 'resensor.giv_(.+)_soc_kwh'
  geserial2 'resensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today
    - sensor.giv_{geserial}_load_energy_today_kwh
  import_today
    - sensor.giv_{geserial}_import_energy_today_kwh
  export_today
    - sensor.giv_{geserial}_export_energy_today_kwh
  pv_today
    - sensor.giv_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the loadimport and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  ge_cloud_data False
  ge_cloud_serial '{geserial}'
  ge_cloud_key 'xxxx'
  #
  # Controlsstatus - must by 1 per inverter
  #
  num_inverters 1
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds 60
  #
  # When set use the REST API rather than HA entity for control, should be more reliablefaster to control
  # Set one per inverter
[... 282 lines cut ...]
  # connect this to your automation to start whatever you want to trigger
  export_triggers
     - name 'large'
       minutes 60
       energy 1.0
     - name 'small'
       minutes 15
       energy 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today 'sensor.xxxxx'
  #iboost_energy_scaling 1.0
  # Gas rates for comparison
  #metric_octopus_gas 're(sensor.(octopus_energy_)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url 'httpswww.nordpoolgroup.comapimarketdatapage325currency=GBP'
  #futurerate_adjust_import False
  #futurerate_adjust_export True
  #futurerate_peak_start 160000
  #futurerate_peak_end 190000
  #futurerate_peak_premium_import 14
  #futurerate_peak_premium_export 6.5

springfall2008, 2023-12-25:
You probably need one space before the dash eg

rate_export:
 - rate: 15

dandwhelan, 2023-12-25:
Sorry still no joy.

Maybe I should delete everything batpred related and start from scratch.

# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat
  module predbat
  class PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix predbat
  # Timezone to work in
  timezone Europe/London

  # XXX This is a configuration template, delete this line once you edit your configuration

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial 'resensor.giv_(.+)_soc_kwh'
  geserial2 'resensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today
    - sensor.giv_{geserial}_load_energy_today_kwh
  import_today
    - sensor.giv_{geserial}_import_energy_today_kwh
  export_today
    - sensor.giv_{geserial}_export_energy_today_kwh
  pv_today
    - sensor.giv_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the loadimport and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  ge_cloud_data False
  ge_cloud_serial '{geserial}'
  ge_cloud_key 'xxxx'
  #
  # Controlsstatus - must by 1 per inverter
  #
  num_inverters 1
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds 60
  #
  # When set use the REST API rather than HA entity for control, should be more reliablefaster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
[... 282 lines cut ...]
  export_triggers
     - name 'large'
       minutes 60
       energy 1.0
     - name 'small'
       minutes 15
       energy 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today 'sensor.xxxxx'
  #iboost_energy_scaling 1.0
  # Gas rates for comparison
  #metric_octopus_gas 're(sensor.(octopus_energy_)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url 'httpswww.nordpoolgroup.comapimarketdatapage325currency=GBP'
  #futurerate_adjust_import False
  #futurerate_adjust_export True
  #futurerate_peak_start 160000
  #futurerate_peak_end 190000
  #futurerate_peak_premium_import 14
  #futurerate_peak_premium_export 6.5
File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10523, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9766, in update_pred self.fetch_sensor_data() File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9357, in fetch_sensor_data self.rate_export = self.basic_rates(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 18:15:03.489257 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 18:20:03.570429 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 18:20:03.571784 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2023-12-25 18:20:03.572234 WARNING pred_bat: Worker Ags: {'id': 'b21a4515aea74f6fb476a3817a8c3233', 'name': 'pred_bat', 'objectid': 'ae65d3fcd06144d488a222d829065e05', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fb8b97150>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}} 2023-12-25 18:20:03.572736 WARNING pred_bat: ------------------------------------------------------------ 2023-12-25 18:20:03.57 [... cut ...] es(self.get_arg("rates_export_override", [], indirect=False), "export", self.rate_export) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4816, in basic_rates start_str = this_rate.get("start", "00:00:00") ^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'get' 2023-12-25 18:30:03.514567 WARNING pred_bat: ------------------------------------------------------------

springfall2008, 2023-12-25:
Can you attach your apps.yaml as file rather than cut pasting it so I can check the format?

dandwhelan, 2023-12-25:
Here you go ☺️

dandwhelan, 2023-12-25:
[Desktop.zip](https://github.com/springfall2008/batpred/files/13767393/Desktop.zip)

dandwhelan, 2023-12-25:
Ok, I completely removed and reinstalled and everything looks fine. Thanks for your time!

closed 2023-12-25 by dandwhelan

springfall2008, 2023-12-26:
For reference the attached apps.yaml was corrupted, e.g. colons were missing