springfall2008/batpred #530: Solar sometimes shows correctly in plan, other times stays at fixed number

Boffinboy, 2023-12-26:
Happy Christmas! thank you for your work and making predbat available to us!

**Describe the bug**
In the plan, the PV kWh column sometimes shows a fixed value, after reaching a certain point - this continues from then onwards even through the night. I am not sure if this is being used for planning or just a display error. Usually appears to correct itself after I notice. 

**Expected behavior**
Should show forecast values 

**Predbat version**
v7.14.24

**Environment details**

- Givenergy AC3 and 9.5 battery
- HAOS running in VM on Synology NAS

**Screenshots**
“Stuck” at 0.53
[image]

**Log file**
Not been able to capture a log when I’ve noticed it yet - what period should I capture?

Boffinboy, 2023-12-26:
Looking at plan it does seem to be influencing it, see below where SoC looks to be increasing 

[image]

springfall2008, 2023-12-26:
Hopefully fixed now: https://github.com/springfall2008/batpred/releases/tag/v7.14.25

Boffinboy, 2023-12-26:
Thank you. I think it may have been fixed and I’d not updated to 14.24. Have updated to 25 and will let you know!

Boffinboy, 2023-12-26:
Looks like the issue isn’t occurring anymore! However, I am a bit puzzled by the current plan. It seems to only be doing a very small charge before the peak, despite many slots being nearly half the price of others. Even with round trip losses I would have thought it makes sense to charge? I have conservative settings for the solar because Solcast tends to significantly err on the positive for my array vs reality, and I have set a high bar for force discharge as I don’t want the extra cycles, but I am surprised there is no charging happening. 

[image]
[image]
[image]
[image]
[image]
[image]

springfall2008, 2023-12-26:
Try setting metric soc keep to 0.5 and maybe reduce metric battery cycle back to 2?

Boffinboy, 2023-12-26:
Setting keep to 0.4 (roughly equivalent to 4% reserve) totally transformed it back to something more like I’d expect. I’m still not entirely sure how the soc keep influences the plan - I would have thought allowing it to run to empty with would be optimal?

springfall2008, 2023-12-27:
Was the empty plan cheaper or more expensive for the next 24 hours?

Boffinboy, 2023-12-27:
I have to confess I didn’t check… very silly of me!

springfall2008, 2023-12-27:
>On a somewhat related note - does metric_min_improvement_discharge relate solely to forced discharge (including export)?

It relates to forced export, and the time period is one export slot which can be normally up to 1 hour maximum bit is sized in 5 minute chunks.

Boffinboy, 2023-12-28:
Thank you very much for the explanation. Have sent you a couple of beers. Really appreciate your work and responsiveness!

closed 2023-12-28 by springfall2008