springfall2008/batpred #534: Charging limited to 2500w

dandwhelan, 2023-12-26:
**Describe the bug**

On my givenergy my charge rate is set to 3600 as I'm on a 5k hybrid charger gen 2.
Yet for some reason it seems to only charge at 2500w. 
[image]

[image]
[image]

**Expected behavior**
I've set the charge limit to 3600w as a precaution as it still only charged at 2500w.

**Predbat version**

Latest 

**Environment details**

- Inverter and battery setup
H5 g2. 8.2kw battery 
- Standard HAOS installer or Docker
Haos standard
- Anything else?

springfall2008, 2023-12-26:
Can you post a log please, at any time would do as the inverter details are logged?

dandwhelan, 2023-12-26:
[appdaemon.zip](https://github.com/springfall2008/batpred/files/13773586/appdaemon.zip)
^^ latest log attached

dandwhelan, 2023-12-26:
I can see a few 

New Inverter 0 with soc_max 8.19 kWh nominal_capacity 8.19 kWh battery rate raw 3600.0 w charge rate 3.6 kW
[appdaemon.log.1.zip](https://github.com/springfall2008/batpred/files/13773596/appdaemon.log.1.zip)

In this log...

springfall2008, 2023-12-26:
It seems Predbat sees 3600w and your charge rate register is set to 3600 so I don’t think this is Predbat related.

dandwhelan, 2023-12-26:
Weird. I'll reset my settings on giv and set the charge mode to solar+grid.
I'll let you know how it goes.

dandwhelan, 2023-12-27:
Last night it seems to have sorted itself out... odd....

Did a few chargers at 2700w and then 3500w

[image]

closed 2023-12-27 by dandwhelan