springfall2008/batpred #535: Car charging time

dandwhelan, 2023-12-26:
**Describe the bug**

This is more than likely me but understanding he documentation but I've tried reading it many times and tweaks setting and I don't know what to do. 

I have a hypervolt charger, it is integrated into HA. It's no agile connectivity. On the hypervolt ha, One entity says ready all the time even when unplugged, I have no data saying its plugged in. A second entity I can use is charger state set, on and off. 

I want the car to charge up between 11pm and 7am with the cheapest possible price on agile. 

Please can somebody tell me what needs to be changed to achieve this? 

I made a few tweaks but it seems to think I want to charge all day on predbat.

springfall2008, 2023-12-26:
Can you control when the charger is charging via HA? If so you can use Predbat in smart car charging mode to pick the slots. If not then you can connect you car charging state to 'car_charging_now' in the apps.yaml and at least predbat will know when it's active.

dandwhelan, 2023-12-26:
Sorry, but the documentation isn't make much sense to me.

  # In some cases car planning is difficult (e.g. Ohme with Intelligent doesn't report slots)
  **# The car charging now can be set to a sensor to indicate the car is charging and to plan
  # for it to charge during this 30 minute slot**
  car_charging_now: 
      - 're:(sensor.hypervolt_charging_readiness)'

  # Positive responses for car_charging_now
  car_charging_now_response:
    - 'yes'
    - 'on'
    - 'charging'

Do you think the above looks correct? car_charging_now: said no before.. wasn't sure what data to input to look at the sensor.

springfall2008, 2023-12-26:
Have a looking at the history of your readiness sensor and see what it reports when charging

dandwhelan, 2023-12-26:
It reports 'charging'

springfall2008, 2023-12-27:
So what you have above should work, but obviously as it's just charging now it won't be planned for only dealt with at the time.

dandwhelan, 2023-12-27:
Maybe I've been reading all this wrong and I should be using "binary_sensor.predbat_car_charging_slot" I'll need create a manual switch to say I've plugged the car in and none of the entities for my charger present this info. 

Do you have any examples of how to call "binary_sensor.predbat_car_charging_slot"?

"This feature allows Predbat to create a plan for when you car will charge, but you will have to create an automation to trigger your car to charge using binary_sensor.predbat_car_charging_slot if you want it to match the plan."

dandwhelan, 2023-12-27:
A) 👍

dandwhelan, 2024-01-02:
I managed to find a workaround to my issue, I've submitted a feature request to help me out with a new issue :D

closed 2024-01-02 by dandwhelan