springfall2008/batpred #536: Unnecessary discharging then recharge at higher rate

gcoan, 2023-12-26:
**Describe the bug**
Predbat has been very reliable of late, coping and executing extremely well in the power up and negative agile periods we've had over Christmas.
But this morning the battery was nearly full but Predbat wanted to discharge it to empty at 16.93 export, only then to have to recharge the battery at 16.54-17.64 to then get me through the peak.
This can't have been cost effective.

I turned predbat into read only, stopped the discharge, and then a period of time later was able to turn predbat back on when the plan had evolved to now keep the charge.

**Expected behavior**
Battery held at same level, not discharged until the peak.

**Predbat version**
7.4.22

**Environment details**
Gen 1 hybrid, 5.2 battery.  HAOS.

**Screenshots**
[image]

[image]
[image]
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2023-12-26:
I think it's a little marginal with your loss settings of only 4% and no inverter loss. I'd suggest setting an inverter loss of say 4% or a non-zero metric battery cycle (1p or 2p)

gcoan, 2023-12-26:
> I think it's a little marginal with your loss settings of only 4% and no inverter loss. I'd suggest setting an inverter loss of say 4% or a non-zero metric battery cycle (1p or 2p)

My loss settings are 5% charge and 5% discharge, no inverter loss.

The charge period was planned from 12:30-16:00 (but would finish at around 15:00 when soc reaches 100%).  But in that period, only 12:30 and 13:30 (both at 16.54p) is the import rate lower than the export rate being achieved (if we totally ignore losses).  For all the other periods the import rate is higher than the achieved export rate.

So I don't think its just down to my loss settings.

springfall2008, 2023-12-27:
I think I see why this happened, at the point the export was picked the plan didn't have a charge afterwards, the charge was added later in the calculation and the discharge isn't removed. I'll have another look as it maybe tweakable

gcoan, 2023-12-27:
> I think I see why this happened, at the point the export was picked the plan didn't have a charge afterwards, the charge was added later in the calculation and the discharge isn't removed. I'll have another look as it maybe tweakable

Great, thanks.  It sounds quite complicated with these multiple passes, optimisations and things getting added at different stages.  It has pretty much otherwise been fine with Predbat, the plans look sensible enough, just this time it wasn't

springfall2008, 2023-12-27:
I push some tweaks to main that may or may not make it better, worth a try if you get time.

Yeh it's too complex, I wish I had a simpler solution. I think fboundy was convinced it could be done using a set algorithm of buy/sell prices but every time I go through that idea it falls down on the edge cases.

gcoan, 2024-08-16:
Ages old, code has moved on now

closed 2024-08-16 by gcoan