springfall2008/batpred #537: Cant get batpred to start

tonyvfield, 2023-12-26:
Hi,
A very new newbie, so help has to be explicate.
I have watched the video and all goes well till around 19mins in when I dont get the logs showing batpred has started in the settings>system>logs and then selected appdeamon.
I have attached the longs for this after a full power on cycle
[a0d7b954_appdaemon_2023-12-26T18-09-29.816Z.log](https://github.com/springfall2008/batpred/files/13773140/a0d7b954_appdaemon_2023-12-26T18-09-29.816Z.log)
The stuff loaded under hacs is shown here.
[image]

The support suggests I look at predbat.status but I have no idea where to look since this is not explained.

Also in the setup it suggests you can run batpred in monitoring mode only by setting select.predbat_mode. However this does not appear in the yaml file and the setup does not explain where to find this item. I want to test it to start with but obviously the first thing is to get it to run.

All simple help most appreciated
Tony

springfall2008, 2023-12-26:
Did you change appdaemon.yaml to set apps_dir? This bit is new and in the documentation but not on the video

tonyvfield, 2023-12-26:
Do you mean this file? /homeassistant/appdaemon/apps/batpred/config/apps.yaml
i cant find an appdaemon.yaml file

springfall2008, 2023-12-26:
Appdaemon install on this page https://springfall2008.github.io/batpred/install/#appdaemon-install

tonyvfield, 2023-12-26:
I think that is installed since I get a screen that says stop, restart, open webUI, uninstall.
[image]

tonyvfield, 2023-12-26:
Also I have this in the log
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (aarch64 / yellow)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[18:06:47] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2023-12-26 19:06:56.415938 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2023-12-26 19:06:56.418257 INFO AppDaemon: Python version is 3.11.6
2023-12-26 19:06:56.418829 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2023-12-26 19:06:56.419351 INFO AppDaemon: Added log: AppDaemon
2023-12-26 19:06:56.419965 INFO AppDaemon: Added log: Error
2023-12-26 19:06:56.420485 INFO AppDaemon: Added log: Access
2023-12-26 19:06:56.420994 INFO AppDaemon: Added log: Diag
2023-12-26 19:06:57.179641 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2023-12-26 19:06:57.925483 INFO HASS: HASS Plugin Initializing
2023-12-26 19:06:57.926052 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2023-12-26 19:06:57.926599 INFO HASS: HASS Plugin initialization complete
2023-12-26 19:06:57.927722 INFO AppDaemon: Initializing HTTP
2023-12-26 19:06:57.928783 INFO AppDaemon: Using 'ws' for event stream
2023-12-26 19:06:57.953030 INFO AppDaemon: Starting API
2023-12-26 19:06:57.960169 INFO AppDaemon: Starting Admin Interface
2023-12-26 19:06:57.961302 INFO AppDaemon: Starting Dashboards
2023-12-26 19:06:58.027081 INFO HASS: Connected to Home Assistant 2023.12.3
2023-12-26 19:06:58.086784 INFO AppDaemon: App 'hello_world' added
2023-12-26 19:06:58.089600 INFO AppDaemon: Found 1 active apps
2023-12-26 19:06:58.091390 INFO AppDaemon: Found 0 inactive apps
2023-12-26 19:06:58.092716 INFO AppDaemon: Found 0 global libraries
2023-12-26 19:06:58.094232 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2023-12-26 19:06:58.102008 INFO AppDaemon: Running on port 5050
2023-12-26 19:06:58.253962 INFO HASS: Evaluating startup conditions
2023-12-26 19:06:58.277893 INFO HASS: Startup condition met: hass state=RUNNING
2023-12-26 19:06:58.279382 INFO HASS: All startup conditions met
2023-12-26 19:06:58.465790 INFO AppDaemon: Got initial state from namespace default
2023-12-26 19:07:00.116274 INFO AppDaemon: Scheduler running in realtime
2023-12-26 19:07:00.123285 INFO AppDaemon: Adding /config/apps to module import path
2023-12-26 19:07:00.139253 INFO AppDaemon: Loading App Module: /config/apps/hello.py
2023-12-26 19:07:00.344785 INFO AppDaemon: Loading app hello_world using class HelloWorld from module hello
2023-12-26 19:07:00.352525 INFO AppDaemon: Calling initialize() for hello_world
2023-12-26 19:07:00.583722 INFO hello_world: Hello from AppDaemon
2023-12-26 19:07:00.587224 INFO hello_world: You are now ready to run Apps!
2023-12-26 19:07:00.590361 INFO AppDaemon: App initialization complete
2023-12-26 19:44:18.608820 INFO AppDaemon: New client Admin Client connected
2023-12-26 19:44:38.474527 INFO AppDaemon: Client disconnection from Admin Client

springfall2008, 2023-12-26:
Like I said you need to edit appdaemon.yaml file which is in the addons_config directory now. The instructions on that page shows how to find it

tonyvfield, 2023-12-26:
OK I do not have an addons_config directory so made a directory and a file
/homeassistant/conf/appdaemon.yaml

Copied his example, although the wrong timezone for me but hopefully that does not matter. Changed the address to what I use from my browser but what is a long life token? Do I make up a number or something?

# appdaemon.yaml
appdaemon:
  time_zone: CET
  latitude: 51.725
  longitude: 14.3434
  elevation: 0
  plugins:
    HASS:
      type: hass
      ha_url: <http://homeassistant.local:8123/>
      token: <some_long_lived_access_token>

springfall2008, 2023-12-26:
That won’t work.Did you change your file editor to allow access outside homeassistant directory?

tonyvfield, 2023-12-26:
I use the file editor in HA that some other app said I needed and I now have the following
[a0d7b954_appdaemon_2023-12-26T19-46-16.131Z.log](https://github.com/springfall2008/batpred/files/13773438/a0d7b954_appdaemon_2023-12-26T19-46-16.131Z.log)
Which seems to suggest it found the file and loaded everything.

So if this is true how do I see if batpred is running?

springfall2008, 2023-12-26:
Predbat isn’t running just appdaemon is running

Like I said you need to find the addons_config directory and edit appdaemon.yaml as per the documentation

springfall2008, 2023-12-26:
[image]

springfall2008, 2023-12-26:
[image]

tonyvfield, 2023-12-26:
I dont have or cant find that option. What I see from the gear icon is this 
[image]

tonyvfield, 2023-12-26:
Found it. Not in the settings in the app but the settings in HA for the app file editor.
Hold on, now I will try

tonyvfield, 2023-12-26:
OK what program etc do I use to move the entire directory?
"Move the entire 'apps' directory from addon_configs/a0d7b954_appdaemon (new location) to config/appdaemon (the old location)"

springfall2008, 2023-12-26:
You don’t need to move anything as it’s new install, just set app_dir in appdaemon.yaml

tonyvfield, 2023-12-26:
OK so got this now in the logs for appdaemon. No errors after a restart. But should batpred be in here?
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (aarch64 / yellow)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[20:49:55] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started

springfall2008, 2023-12-26:
Yes it should be, what does your appdaemon.yaml
Look like?

tonyvfield, 2023-12-26:
---
appdaemon:
  latitude: 52.379189
  longitude: 4.899431
  elevation: 2
  time_zone: Europe/London
  thread_duration_warning_threshold: 120
  plugins:
    HASS:
      type: hass
      app_dir: /homeassistant/appdaemon/apps
http:
  url: http://192.168.1.51:8123/
admin:
api:
hadashboard:
logs:
  main_log:
    filename: /homeassistant/appdaemon/appdaemon.log
    log_generations: 9
    log_size: 10000000

springfall2008, 2023-12-26:
Looks good, see if you can find appdaemon.log now which is inside the homeassistant directory

tonyvfield, 2023-12-26:
yes that is below but no mention of predbat

2023-12-26 20:50:02.369191 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2023-12-26 20:50:02.369970 INFO AppDaemon: Python version is 3.11.6
2023-12-26 20:50:02.370555 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2023-12-26 20:50:02.370971 INFO AppDaemon: Added log: AppDaemon
2023-12-26 20:50:02.371376 INFO AppDaemon: Added log: Error
2023-12-26 20:50:02.371719 INFO AppDaemon: Added log: Access
2023-12-26 20:50:02.372057 INFO AppDaemon: Added log: Diag
2023-12-26 20:50:02.805800 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2023-12-26 20:50:03.219304 INFO HASS: HASS Plugin Initializing
2023-12-26 20:50:03.219725 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2023-12-26 20:50:03.220110 INFO HASS: HASS Plugin initialization complete
2023-12-26 20:50:03.220851 INFO AppDaemon: Initializing HTTP
2023-12-26 20:50:03.221553 INFO AppDaemon: Using 'ws' for event stream
2023-12-26 20:50:03.227564 INFO AppDaemon: Starting API
2023-12-26 20:50:03.231566 INFO AppDaemon: Starting Admin Interface
2023-12-26 20:50:03.232220 INFO AppDaemon: Starting Dashboards
2023-12-26 20:50:03.263117 INFO HASS: Connected to Home Assistant 2023.12.3
2023-12-26 20:50:03.281843 INFO AppDaemon: App 'hello_world' added
2023-12-26 20:50:03.283666 INFO AppDaemon: Found 1 active apps
2023-12-26 20:50:03.284266 INFO AppDaemon: Found 0 inactive apps
2023-12-26 20:50:03.284795 INFO AppDaemon: Found 0 global libraries
2023-12-26 20:50:03.285448 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2023-12-26 20:50:03.287201 INFO AppDaemon: Running on port 8123
2023-12-26 20:50:03.371426 INFO HASS: Evaluating startup conditions
2023-12-26 20:50:03.389347 INFO HASS: Startup condition met: hass state=RUNNING
2023-12-26 20:50:03.390115 INFO HASS: All startup conditions met
2023-12-26 20:50:03.488731 INFO AppDaemon: Got initial state from namespace default
2023-12-26 20:50:05.300586 INFO AppDaemon: Scheduler running in realtime
2023-12-26 20:50:05.305376 INFO AppDaemon: Adding /config/apps to module import path
2023-12-26 20:50:05.308125 INFO AppDaemon: Loading App Module: /config/apps/hello.py
2023-12-26 20:50:05.385976 INFO AppDaemon: Loading app hello_world using class HelloWorld from module hello
2023-12-26 20:50:05.388501 INFO AppDaemon: Calling initialize() for hello_world
2023-12-26 20:50:05.502757 INFO hello_world: Hello from AppDaemon
2023-12-26 20:50:05.504616 INFO hello_world: You are now ready to run Apps!
2023-12-26 20:50:05.506682 INFO AppDaemon: App initialization complete

springfall2008, 2023-12-26:
Can you attach your appdaemon.yaml
As you might have indentation issues?

tonyvfield, 2023-12-26:
[image]

tonyvfield, 2023-12-26:
The items in hacs include kia, solecast and Octopus intelligent all of which show up in the integrations display. But nothing for predbat

springfall2008, 2023-12-26:
Your add_dir setting is wrong indent  should be like this 
[image]

tonyvfield, 2023-12-26:
Sorry had to go out.
But edited the file and things have changed.
1. I got a message on the phone about oredbat going idle
2. I can now remote access ha. Buggered. That will have to wait for
tomorrow.

Many thanks for your help and i hope i can still get in once at home

Tony

Regards
Tony Field

tonyvfield, 2023-12-27:
If you happen to still be up, backpred is running but one of my original issues was how to get it into test mode ie  select.predbat as I never found that.
I dont want it running for real until I am sure its doing the right thing as I empty my batteries most days and need it to charge to 100% over night

gcoan, 2023-12-27:
> If you happen to still be up, backpred is running but one of my original issues was how to get it into test mode ie  select.predbat as I never found that.
>
> I dont want it running for real until I am [...]

You need to add the predbat control entities to a dashboard page in home assistant.
There's a number of different controls and outputs described in the documents at https://springfall2008.github.io/batpred/output-data/

You'll need these controls to see what predbat is doing and to tune it to your preferences.

springfall2008, 2023-12-27:
If it's working the controls will be created, you can manually add them to a dashboard or use the auto-generated one or the collapsable cards.

If you go to developer tools, states you can search for predbat also to see what you have.

select.predbat_mode is the mode setting one

tonyvfield, 2023-12-27:
Thanks and found it.
Just now got to hope that the batteries charge properly tonight. They were
still discharging last night at 1am so they did not get fully charged. But
turned off HA totally and manually set them to charge so got to 80%.
Now that predbat is in monitoring mode i can watch what its doing for a few
days before i get it loose.

Many thanks for all your help

Regards
Tony Field

gcoan, 2023-12-27:
Glad you're getting sorted.

Follow the predbat configuration guide for the tariffs you are on. Best thing to understand what predbat is planning on doing is the html plan - configuration details in the output section of the documentation

gcoan, 2023-12-28:
@tonyvfield I've started on a refresh of the install instructions to hopefully make it easier to follow.
Done the install page so far https://github.com/gcoan/batpred/blob/main/docs/install.md 
Do feedback any comments and things you find not clear so I can add to the docs

tonyvfield, 2023-12-28:
The thing that stumped me for some time was the instruction to turn off the enforced base. This is not in what I would consider the usual place ie the gear icon of the tool itself but in the calling or controlling item that installed it. 
That is probably a poor description but I cant find that option again to explain how I would write it. But the path to that needs writing down.

gcoan, 2023-12-28:
> The thing that stumped me for some time was the instruction to turn off the enforced base. This is not in what I would consider the usual place ie the gear icon of the tool itself but in the calling [...]

Can you remember where that is in the instructions?  It sounds very unclear

springfall2008, 2023-12-28:
It’s here in install details 
[image]

tonyvfield, 2023-12-28:
> It’s here in install details
> IMG_3416.png (view on web)

No that is the picture once you find the right location for that option.
Its not in the gear section of the app which is where i looked.
I did try to find it again but could not

Anyway the last two nights the batteries have not charged properly.  Last
night i stayed up fighting the system since it was still not charging at
gine midnight. In the end i pulled the power on HA and then went and reset
things and manually drove the inverters.  They then charged.
So if it does it again tonight i will want to know how to cleanly uninstall
predbat.
It is supposed to be in monitor mode and all the options in the config file
are turned off so it ahould not be doing anything.
Regards
Tony Field

gcoan, 2023-12-28:
> No that is the picture once you find the right location for that option.
> 
> Its not in the gear section of the app which is where i looked.
> 
> I did try to find it again but could not
> 
> 
> 
> Anyway the last two nights the batteries have not charged properly.  Last
> 
> night i stayed up fighting the system since it was still not charging at
> 
> gine midnight. In the end i pulled the power on HA and then went and reset
> 
> things and manually drove the inverters.  They then charged.
> 
> So if it does it again tonight i will want to know how to cleanly uninstall
> 
> predbat.
> 
> It is supposed to be in monitor mode and all the options in the config file
> 
> are turned off so it ahould not be doing anything.
> 
> Regards
> 
> Tony Field
> 
> 
> 
> 
> 
> 
> 
> > It’s here in install details
> 
> > IMG_3416.png (view on web)

Thanks for pointing me to the problem with the install instructions, I'll have a look at making them clearer how to turn this setting on.

But to your problem with the battery not charging properly overnight, if you have set the entity select.predbat_mode to Monitor then predbat won't be making any changes to your inverter. It will be reading what your consumption is and planning what activities to charge your battery, but it won't be doing any changes.

It is possible to control your battery charging directly in home assistant via the givtcp add on, but it's not straight forward as there are several inverter settings you need to understand what they do and change correctly. Predbat takes away the need to understand how to do that.

If you leave predbat in monitor mode and control your charging overnight as you always have done (GivEnergy app or portal) you shouldn't be having problems.

We can talk you through how to understand the predbat plan and what it's doing (and I'm working on improving the instructions to help everyone).

gcoan, 2023-12-29:
> Thanks for pointing me to the problem with the install instructions, I'll have a look at making them clearer how to turn this setting on.

Updated the docs to explain how to change this setting in the File Editor add-in.
https://github.com/gcoan/batpred/blob/main/docs/install.md

tonyvfield, 2023-12-29:
Spot on now with the description of where to find the setting. I have just
found mine again.

Regards
Tony Field

tonyvfield, 2023-12-29:
Hi guys,

Sorry to say i need some help. I want to remove predbat. The HA is running
out of memory and has killed various processes because of this.
My system only has 2G of ram (google yellow) and that is the reason why we
can't now see the state of the batteries for another use.
I have tried to remove the app in hacs, and then rebooted HA but its still
there and still operating and i still cant see my batteries.
How do i totally uninstall predbat please

Regards
Tony Field

gcoan, 2023-12-29:
Uninstall the app daemon add on and this’ll remove predbat as well and release memory from the add on no longer running (addons run as separate docker containers)

Not good that the yellow is so tight on memory.  What other addons and integrations are running?

springfall2008, 2024-01-01:
@gcoan there are a few doc updates in this PR https://github.com/springfall2008/batpred/pull/566 which are for appdaemon-predbat combined install - maybe you can merge them into your updates rather than me merging them and creating conflicts for you?

gcoan, 2024-01-01:
> @gcoan there are a few doc updates in this PR https://github.com/springfall2008/batpred/pull/566 which are for appdaemon-predbat combined install - maybe you can merge them into your updates rather than me merging them and creating conflicts for you?

Yes certainly can do that.

I'll get onto it later today

gcoan, 2024-01-01:
I've been through the main install, energy rates and am most of the way through the config file doc.

Still to look at configuration settings, output data and apps.yaml template but I could finish the config file, merge in the stuff from your pr and we make this a version that at least gets some of it live?

springfall2008, 2024-01-01:
@gcoan sounds good

gcoan, 2024-01-01:
@springfall2008 I've finished work on the apps-yaml.md config file (but afterwards found I had missed out some iboost settings and load_forecast so will have to go back and add those in ...

I merged the updates to install.md in https://github.com/gcoan/batpred/blob/main/docs/install.md and started working my way through them.  Had some trouble picking up the new add-on in the add-on store, and at one point when I uninstalled the add-on, Home Assistant crashed completely.  Not sure why that was.

Anyway, have installed it now alongside my Appdaemon install of predbat (that I stopped).
Will have to make some changes in the docs as the apps.yaml and log file are now installed in different places (and the logfile is a different name) so the previous pathnames won't always work.

But looking at the logfile for the add-on, I noticed a couple of issues.

The logfile says it is picking up my appdaemon config from /config/appdaemon.yaml which wouldn't normally exist?

And looking at the /addon-configs/46f69597_appdaemon-predbat/appdaemon.yaml that's part of the new add-on, its pointing app_dir to /config/apps (which is my original predbat, not the one in /addon_configs/46....

Predbat doesn't like there being two installs!

[predbat.log](https://github.com/springfall2008/batpred/files/13806300/predbat.log)

springfall2008, 2024-01-09:
@gcoan just to confuse things /config in appdeamon points to /addon-configs/xxxx/ where as /config in HA points to /homeassistant

gcoan, 2024-01-09:
> @gcoan just to confuse things /config in appdeamon points to /addon-configs/xxxx/ where as /config in HA points to /homeassistant

oh that is awful.

I can see it now, the appdaemon.yaml is for appdaemon-predbat is pointing app_dir to /config/apps and the logfile to /config/predbat.log, but both are actually being picked up from /addon_configs/46fxxxx_appdaemon-predbat

whereas in the HA filesystem, /config and /homeassistant are symbolic linked together (i'm guessing from what I see in Samba that /config is the actual filesystem and /homeassistant is the linked dir).

I've noticed a while ago that nowhere does the doc refer to where the output log is and so I added it to to this iteration. I took the pathnames based on what had been set in the appdaemon.yaml files, but checking again, looks like I had managed to put the correct filesysystem path for the appdaemon-predbat logfile.  Phew

springfall2008, 2024-01-09:
I think /config is meant to the path to this applications configuration directory, and each add-on has it's own directory and home assistant has one too

gcoan, 2024-01-09:
True but its poor programming to muddle around with changing / (root path) filenames like that in my view.

I can see why it's done to separate out stuff for each app without them having to be chsnged, but it introduces confusion that /config refers to different places depending on context

Hey ho

closed 2024-02-03 by springfall2008