springfall2008/batpred #539: Issues installing & configuring Predbat

Keith-J, 2023-12-27:
I am having a nightmare trying to get Predbat up & running. I am new to HA but I am a programmer & I have worked with UNIX/Linux for over 30 years. So I have some idea what I'm doing, especially at the shell prompt. 
My issues currently lie with the apps.yaml file that comes with Predbat. 
There are 2 places where I feel the comments in this file could be improved. 
At about line 256 in the version I have, there is an "Energy Rates" section. The section leads with a  comment that starts "Please set one of these three......". In this line & the many lines lines that follow, there are no 3 obvious options to choose from? A little more explanation here would help.
 
Also, within the comment text a few lines below we see:
"NOTE: To get detailed energy rates you need to go in and manually enable the following events in HA".
That might mean something to seasoned HA users, but to newbies it means nothing. Googling "Enable events in Home Assistant" doesn't help either. 

I'm not asking for support. I will get the issues sorted. I just think that a few more words of help in the right place would prevent others from going through the 12 hour unsuccessful install that I have endured (so far).  

I know that you are doing this development in your own time & I am grateful for your efforts. 
My purpose here is to try to help by pointing out a couple of things that I feel are confusing, so that others will find configuration easier in future.

gcoan, 2023-12-27:
Thanks for the feedback, it's only through people saying what's clear and what needs more explanation can we make improvements!

If you are using Octopus as your supplier then using the Octopus integration is by far the easiest way to go.
The other options for manual config of the rates in apps.yaml is always an option, but probably mainly for use by people not supplied by Octopus.

The Octopus integration changed recently and made the setup for use with Predbat a bit more complex.

To enable the events...

Once the octopus integration is installed and configured with your octopus account api key, go to Integrations, Choose Octopus, then Entities
You should note get a list of entities (and events) supplied by the octopus integration 
Click the downwards arrow on the right hand side and make sure all the options are selected
Then in the search box, left hand side, type "current_day", click on the first event that comes up, check the name is the right one, click the cog wheel, then you should see the option to enable the event

Repeat for the other events

I'll look to add to the docs, but please feedback other experiences you have on setting up predbat

gcoan, 2023-12-27:
Hi @Keith-J I've started improving the installation instructions further, fresh pair of eyes and all that.

So far I've made changes to the main installation instructions, you can see my draft at https://github.com/gcoan/batpred/blob/main/docs/install.md

Let me know what you think, if there's anything key missing.  

I plan to add the details of the 'please set one of these three' to the Energy Rates and the apps.yaml file, also need to expand the output-data a bit more.
I'll look at those things tomorrow.

Cheers Geoffrey

Keith-J, 2023-12-28:
Hi Geoffrey

I have had a quick read through & the documentation looks excellent. There are just a couple of things I noticed. 
1. I remember seeing "apps_dir" instead of "app_dir" in the previous version. Whilst this has been corrected in the important place, there are a couple of other places where "apps_dir" is mentioned. A simple search & replace will sort this.

2. I come from a UNIX background. So, to me, it's important to use a leading "/" when specifying an absolute path. There are a number of places where the document shows paths to files or folders in the "/congfig" or "/addon_configs" folders where the leading "/" is missing. However, whenever the "/homeassistant" folder is shown, it has the leading "/". 
I think all absolute paths should have the leading slash for consistency. Also, people copy & pasting the paths will hit problems unless they are in the root folder to start with.

Thank you for letting me read through this. I hope my minor comments help. 

FYI, I now have batpred up & running in monitor mode🙂. However, my Solar panels are not currently in the configuration. See below.

I see that support for SolarEdge is in development. I have a SolarEdge PV system and a GivEnergy battery system. So I have 2 inverters, one for PV & one for the battery. This was not due to any particular requirements on my part. This is just what our supplier installed back in 2021.
I guess the supplier must have sold this configuration to other customers, so it's probably not unique. 
I would love to help with the batpred SolarEdge integration, if my set up is not too niche?

Best regards 

Keith

⁣Get BlueMail for Android ​

gcoan, 2023-12-28:
Hi Keith, thanks for the feedback.   I too have a Unix background (going right back to Bell Labs Unix in fact!).  I've updated the install.md page to correct the unix paths and double check the spelling of app_dir (and apps.yaml which also had apps.yml in a few places).  I had corrected a couple of entries already as you'd seen, hopefully all on this page now done.

Will get onto further doc pages later on.

Your solar edge/givenergy inverter setup.  To clarify, do you have a solar edge DC to AC inverter for PV generation and then an AC-coupled GivEnergy inverter with batteries?
Or do you have a hybrid GivEnergy inverter with its own panels (and batteries) as well as the Solar Edge inverter?

There'll be a way of getting the two to work with predbat for sure.

Keith-J, 2023-12-28:
Hi Geoffrey

 

Based on a diagram I lifted from the internet, this is what I currently have.
I’ll have a play around with the config files when I get a chance to see if I can make it work.

 

 

Regards

 

Keith

gcoan, 2023-12-28:
Hi Keith, the diagram didn't come through, sorry.
I've noticed this before that pictures attached to emails don't work into GitHub, you have to upload them directly in github (either website or the github app).
If you can try again.  Do you have a sensor giving the pv generation?
Geoffrey

Keith-J, 2023-12-28:
Hi Geoffrey 

Sorry about the image. 
Basically I have a SolarEdge 5kW inverter for PV. 
The GivEnergy inverter is 3kW & is just for the battery. For Solar to charge the battery, it has to pass through 2 inverters.
Effectively, 2 standalone systems.

I do have both my own sensor and the SolarEdge sensors for PV generation monitoring. All are available in HA, so I should be able to experiment with the config.

Regards

Keith 

⁣Get BlueMail for Android ​

springfall2008, 2023-12-28:
So I think this will work just fine, set your inverter hybrid to off, and your inverter limit to 5kw.

gcoan, 2023-12-28:
As Trefor says, this should work, you have an AC coupled inverter.  The GivEnergy template apps.yaml should give most of what you need, but I suggest you change pv_today to point to the sensor that's giving your SolarEdge PV output today in kW.

That way predbat will be able to adjust the in-day plan based on solar actuals not just the Solcast daily prediction.  Useful in the summer !

I have just finished updating the Energy Rates page of the documentation to better explain the different options and ways to configure the Energy Rates into predbat.  I've tried to structure it to be more logical and to guide new users through more clearly.
https://github.com/gcoan/batpred/blob/main/docs/energy-rates.md

I will (later on) update apps.yaml to align and explain the "3 options" - which are the Octopus Energy integration, Octopus rates URL or Rate Bands.  They are all covered in this doc page now

Cheers

Keith-J, 2023-12-30:
Hi Geoffrey

 

I’ve had a skim through “energy-rates.md”.

It looks a lot clearer to me now, although I have to say, I was looking more at the comments in the apps.yaml file  when I was having issues with the energy rates configuration.

 

I have a couple of comments which I have put in the attached document. I hope this attachment works this time.

 

As a complete beginner in HA, just saying “ Go to Integrations”, didn’t immediately make sense to me. I didn’t know if this was in HACS (which has an integrations page), AppDaemon, general settings or something new. Just adding a clue for us newbies I think helps. I have added some text & highlighted it.

There’s a section about clicking a down arrow, which doesn’t make sense to me. I’m not sure what you’re trying to do here. I have added a highlighted comment & a screenshot. All of my comments are in the attached doc.

 

FYI, by following the document, I can see that all of my required Octopus entities are enabled, so I’m all good.

 

If any of my comments are silly, please ignore them. I’m still trying to grope my way around HA😊.

 

Regards

 

Keith

 

https://github.com/gcoan/batpred/blob/main/docs/energy-rates.md

gcoan, 2023-12-30:
Hi Keith 
Unfortunately the attached file got stripped out.  I think this happens if you email responses to github. If you can login to github and upload the file as part of a comment on the issue, I will be able to see it. 

Thanks for the feedback.  Trying to get the balance right in terms of level of detail, don't want to end up with a home assistant tutorial!

I've started work on the apps.yaml configuration document, done about half of it so far - same directory.

Cheers Geoffrey

Keith-J, 2023-12-31:
Hi Geoffrey 
Hopefully you can see the attachment now?
[batpred-comments1.odt](https://github.com/springfall2008/batpred/files/13801081/batpred-comments1.odt)
[HomePVSetup.odt](https://github.com/springfall2008/batpred/files/13801110/HomePVSetup.odt)

Best regards

Keith

closed 2023-12-31 by Keith-J

gcoan, 2023-12-31:
Hi Keith, I've updated energy rates doc in line with the comments you've made above https://github.com/gcoan/batpred/blob/main/docs/energy-rates.md
- when I get to see your attached doc I can see those comments as well.

The main install doc has also been updated https://github.com/gcoan/batpred/blob/main/docs/install.md and I'm part way through the guide to editing apps.yaml

- just saw your message about the attached doc, will look at it now !

gcoan, 2023-12-31:
Yep, I've made changes in line with your comments in the attached doc.

From the schematic I can see that you have an AC-coupled inverter.  If you add the appropriate sensor from the PV generation to apps.yaml, this'll tell Predbat what you have generated today - I have already added this into the apps.yaml doc as it was originally written implying a GivEnergy hybrid inverter

gcoan, 2024-01-01:
@Keith-J if you get the chance, I've finished work on the https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md (previously config-yml-settings.md)

Keith-J, 2024-01-09:
Hi Geoffrey 

Apologies, this got buried in my New Year emails & I forgot about it. I have had a skim through & my only comment is that the descriptions for days_previous & days_previous_weight are confusing & could do with more explanation. 
I have a single value for each of these, but the documentation seems to suggest a comma separated list? Surely days previous is just a single value? 7 in my case. If you can specify more, values, then I think some explanation is needed.

Also the doc says "One entry per line". This is confusing given the coma separated values shown in the doc. 
Might be better to show a couple of examples for this setting, one simple & one more complex, with explanations for each? 

The comments In the apps.yaml file  (at least the version I have), seem to suggest a single value for each of these settings. 

Best regards

Keith

⁣Get BlueMail for Android ​

gcoan, 2024-01-09:
> Hi Geoffrey Apologies, this got buried in my New Year emails & I forgot about it. I have had a skim through & my only comment is that the descriptions for days_previous & days_previous_weight are [...]

Hi Keith,

No problem, you can have multiple entries for days_previous and days_previous_weight if you want, but if you do then you have to put them as a list which in Home Assistant is one entry per line.
e.g.
days_previous:
- 2
- 3
- 4
- 5
- 6
- 7
- 8

To average over all days of the last week (this is what I have)

Or

days_previous:
- 7

If you just want same day last week's consumption

or

days_previous:
- 7
- 14

If you want average of the same day for the last 2 weeks

Good idea to put some actual examples into the documentation.  I'll do that when Trefor releases my last documentation update.

I've been working my way through the documentation trying to update it to make it clearer.   Done most of it now but not all yet, so there will be further changes made.  One of the things I will be doing is to update the template apps.yaml file, resequencing it to align to the documentation, adding sub-headings, and stripping some of the explanatory text out as this really should be in the documentation not duplicated in the file.

Thanks, Geoffrey

Keith-J, 2024-01-09:
Hi Geoffrey 

The explanation in your email is great😊
Something very close to that in the documentation would be a great help to people in my opionion. 
I thought my value of 7, averaged the last 7 days. I now know better. 

Regards

Keith

⁣Get BlueMail for Android ​

gcoan, 2024-01-18:
Have included these changes in PR #625