springfall2008/batpred #540: Issues with Predbat taking the battery level too low

iainfogg, 2023-12-27:
**Describe the bug**

Predbat typically comes up with a plan to discharge my batteries down to 2.0kWh (c. 10-11%) immediately before the IO window starts. The plan is always fine during the day.

However, as the evening progresses (possibly after discharge starts, I'm not certain), the target discharge level in the plan alters and gets lower and lower, down to 4-5%. This then ends up with the battery stopping discharging as it's reached its limit, and around 23:00 or 23:15 I start importing from the grid at normal day rate, losing any benefit from having made something on the export earlier.

This is a relatively new behaviour, as previously it would stick within the limits I'd set.

**Expected behavior**

I normally have `best_soc_keep` at 2.0, and `best_soc_min` at 0.5. My understanding is that Predbat would then come up with a plan to not take the battery level below 2.0kWh. That's what has happened until very recently.

However, I should also add that I'm not as clear as I'd like to be on what  `best_soc_keep` and `best_soc_min` actually do. To me, the descriptions don't fully make sense in the documentation, I'd be keen on improving them if we can (I can do a separate ticket if you prefer). In the [docs](https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options) it says:

> best_soc_keep is minimum battery level to try to keep above during the whole period of the simulation time, soft constraint only (use min for hard constraint). It's usually good to have this above 0 to allow some margin in case you use more energy than planned between charge slots.

It's not clear what "soft limit" means or does, and how that relates to the plan, and to stopping battery discharge (or not). Also, it doesn't state the unit of measurement.

> best_soc_min (expert mode) sets the minimum charge level (in kWh) for charging during each slot and the minimum discharge level also (set to 0 if you want to skip some slots). If you set this non-zero you will need to use the low rate threshold to control which slots you charge from or you may charge all the time.

This one doesn't feel clear at all to me - is charge level the same as battery level, or the rate at which the battery charges or discharges? And I don't understand the bit about using the low rate threshold to control which slots you charge from.

**Predbat version**

7.14.21

**Environment details**

Givenergy Hybrid 5.0, 2x9.5 batteries

**Log file**

I've not got one as I've been turning Predbat discharge control off the last few nights and relying on GivTCP to do a timed discharge that won't run for too long for me. I can try and grab one if you like. Also, I changed `best_soc_min` to 2.0 last night (to be the same as `best_soc_keep`) and this seemed to sort it, but I'm not clear of the repercussions of doing that.

gcoan, 2023-12-27:
@iainfogg Trefor has made a change to best_soc_keep so that it isn't a "hard must not go below" soc level, and is instead something that can be breached if the rates or consumption cause it to do so. 
Previously best_soc_keep was the minimum level to always keep in the battery, so if your consumption caused the soc to go below this, Predbat would then charge the battery to get above the best_soc_keep level - even if this meant charging the battery in peak period rates.   The recent change stops this from happening.

Think of best_soc_min as being minimum battery level that will always be maintained and best_soc_keep as being 'preferred minimum battery level, keeping a bit in reserve in case you need it'

Both are in kWh.

That's my understanding.  I don't have any advice on the low rate threshold, sorry

iainfogg, 2023-12-27:
Historically, I think if I was using more electricity during the evening, the discharge would pause for a bit, then restart, so it still got the keep level at 23:30, but now it seems to keep going 

Do you think it's because it thinks it can make money from it anyway? And if so, what's the benefit of that setting now?

Thanks @gcoan

iainfogg, 2023-12-27:
Sorry - also, what's the best way to set up my system now? I want it to aim for keeping 1.5kWh in, but it can drop to, say, 1.0 without running out and running the house from the grid, so if there's extra evening load it's got that margin in the battery.

But I don't want it to charge from the grid to maintain either of those, I just don't want it to choose to discharge below them.

iainfogg, 2024-01-03:
@springfall2008 when you get a chance, please can you clarify how these battery limit parameters are used now, and maybe also updated the docs to make it clearer for people?

My setup is working now, having increased the min parameter to 1.5, but I'm keen to understand their usage.

springfall2008, 2024-09-01:
I think this is resolved now, but let me know if you need more info

closed 2024-09-01 by springfall2008