springfall2008/batpred #541: Getting rates from Octopus plugin

naltsta, 2023-12-27:
**Describe the bug**
2023-12-27 23:20:03.900204 INFO pred_bat: Error: metric_octopus_import is not set correctly or no energy rates can be read
2023-12-27 23:20:03.954666 INFO pred_bat: ERROR: Exception raised 

**Expected behavior**
Gets data from octopus integration

**Predbat version**

17.17.26

**Environment details**

Standard HACS
I have been running Predate for a couple of weeks by commenting out metric_octopus_import:  and using the fixed rates version. Trying to get the Octopus integration working as considering a switch to agile but can't get it to pull the data through.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
2023-12-27 23:20:03.864969 INFO pred_bat: Fetching futurerate data from None
2023-12-27 23:20:03.877253 INFO pred_bat: WARN: No Octopus data in sensor sensor.electricity_20l3376041_2200016464540_previous_rate attribute 'all_rates'
2023-12-27 23:20:03.888752 INFO pred_bat: WARN: No Octopus data in sensor event.electricity_20l3376041_2200016464540_current_day_rates attribute 'all_rates'
2023-12-27 23:20:03.900204 INFO pred_bat: Error: metric_octopus_import is not set correctly or no energy rates can be read
2023-12-27 23:20:03.954666 INFO pred_bat: ERROR: Exception raised

gcoan, 2023-12-27:
It looks from the errors as if the Octopus integration isn’t working correctly.

Can you go into developer tools and look at whats in the octopus entities
- eg any of the octopus entities named below?

Did you configure the Octopus integration with your octopus api and enable the event entities predbat needs - see https://springfall2008.github.io/batpred/energy-rates/

Thanks Geoffrey

naltsta, 2023-12-27:
[image]
I've been running the Octopus integration successfully long before Predbat and enabled the required sensors. 

Looks like I have a selection of different prefixes for the events and sensors in the integration though
[image]

naltsta, 2023-12-28:
I've deleted and reinstalled the octopus integration, my entities all have the same prefix and I'm up and running... Thanks

naltsta, 2023-12-28:
Did have to remove the brackets so app.yaml now looks like the

  metric_octopus_import: 're:(sensor.octopus_energy_electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
  metric_octopus_export: 're:(sensor.octopus_energy_electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'

closed 2023-12-28 by naltsta

gcoan, 2023-12-28:
> I've been running the Octopus integration successfully long before Predbat and enabled the required sensors. 
> 
> Looks like I have a selection of different prefixes for the events and sensors in the integration though
> 

Think thats probably the cause of the issue. The Octopus sensor names have changed over different versions and the predbat regular expression to find the entities should cope with that, but the introduction of the events are with the newest naming convention whereas older sensors are with the original names.
We had this happen with someone else a week or two ago and deleting and setting up the octopus integration fixes it; as you did.
I had added it to the FAQ's as well.

Glad you're sorted

Geoffrey