springfall2008/batpred #542: Car charge timing issue?

slopemad, 2023-12-28:
Predbat used to be faultless when it came to picking car charging slots.  I currently have car ready time set to 07:00:00, and it has chosen to do some charging in the 02:00-02:30 slot at 1.54p rather than the negative slots at 03:00 and 05:00 (or indeed many of the other slots which are below the 1.54p).

[image]

And from the car charging slot attributes... (You can see it thinks 02:00-02:10 is in the -0.44p slot when it isn't actually in that slot).

- start: 12-28 01:30:00
  end: 12-28 02:10:00
  kwh: 3.48
  average: -0.44
  cost: -1.53
- start: 12-28 03:30:00
  end: 12-28 04:00:00
  kwh: 3.7
  average: -1.63
  cost: -6.03
- start: 12-28 04:30:00
  end: 12-28 05:00:00
  kwh: 3.7
  average: -1.63
  cost: -6.03

springfall2008, 2023-12-28:
Looks like a bug, the 40 minute slot should be 30 minutes, I'll take a look

springfall2008, 2023-12-28:
Can you try this Beta release and see if it fixes it please? https://github.com/springfall2008/batpred/releases/tag/v7.14.27

closed 2024-02-10 by springfall2008