springfall2008/batpred #545: Scheduled charge is switched off in Monitoring Mode

mpartington, 2023-12-29:
**Describe the bug**
When selecting Monitoring mode, or if I am already in monitoring mode and I either: change apps.yaml, restart App Daemon or restart HA I find that Predbat switches scheduled charge off.  This means the batteries would not charge in the scheduled timeslot, unless this is manually switched back on.  

**Expected behaviour**
When selecting Monitoring mode, or if I am already in monitoring mode, I would expect Predbat to maintain the current settings and not make any changes.  

**Predbat version**

v7.14.26

**Environment details**

- Standard HAOS installer

**Log file**
I don't believe this is required as very repeatable - just select Monitoring mode

mpartington, 2023-12-29:
Trouble with the current implementation is that it doesn't just cease any charge or discharge activity on selecting monitoring mode, it also overrides any user changes made subsequently whilst in the mode if you restart HA, AppDaemon or change the config. I don't believe this was the intention of the previous changes, as it effectively renders the mode useless unless your just charging from solar.

springfall2008, 2023-12-29:
I think I can tweak this so that the mode works a bit like the read only toggle, where when you change mode you get a reset but then not again if in monitor mode until you move out of it

mpartington, 2023-12-29:
Sounds good, and I must admit after trialling both, I actually prefer using the read only toggle and leaving it in charge / discharge mode. Good to fix though as it's likely to cause issues with missed charges sooner or later

springfall2008, 2023-12-29:
https://github.com/springfall2008/batpred/releases/tag/v7.14.28

mpartington, 2023-12-29:
Thanks Trefor. Can confirm it works as per your intent i.e. it.gets switched off initially on selecting Monitoring mode. However if I switch the charge schedule back on, it survives both HA and AppDaemon restarts.

Personally I'd rather it maintained the current settings on selection of monitoring mode, but I see why you've done it this way. I think it does need a 'warning' in the docs to cover the fact it will initially be reset though and xx, yy, zz will all be turned off, requiring to be set manually.

closed 2023-12-31 by mpartington

DJBenson, 2024-01-11:
As this has been referenced in the Facebook group, just copying and pasting my response there...

"...yeah that feature request was mine. I agree there are multiple views about what changing modes should do but the issue I was having was if you switched to monitor mode the inverter was left in the previous state indefinitely with no way to override it without manually amending the relevant GivTCP entities, the same was true of "read only" I think until Trefor put some logic in to stop that.
My view is still that if you disable active management (by changing to either "Read Only" or "Monitor") then Predbat should stop what it's currently doing, return the inverter to its default state and stop controlling it - the argument then would be "what's the default state" 😃 For that I would say "whatever ECO mode is in the app" because that's the default mode in the app IIRC."