springfall2008/batpred #546: Selector switch for Savings Sessions

mpartington, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
Predbat currently auto enrols, which is brilliant.  However what about the people who use Predbat for battery management and are with Octopus for energy, but are enrolled in DFS sessions with other providers?  I imagine that auto enrolling could invalidate their session with the other provider.  

A second issue is I imagine that you need to be in charge / discharge mode to allow predbat to manage the session.  If like me, you are on a simple GO tariff, with a fixed window and decision to 100% charge every night, I only really need to use Predbat in Monitoring mode, however I assume that this would allow Predbat to manage the saving session (although this is not explicitly covered in the documents)

**Describe the solution you'd like**
A toggle to enable saver session management. 
-  If off, it does nothing. 
- If on, then it will 
  (i) auto enrol and 
  (ii) also ideally manage the session regardless of the mode that you are in.  I envisage that it would automatically change the mode to charge & discharge. Now the complication could be that you would also need to make a number of tweaks to settings, such as SoC min, SoC keep, combine charge slots (false), combine discharge slots (false) and then revert back at the end of the session.  This would also include the charge slots.  Maybe this would be too complicated.

**Describe alternatives you've considered**
All of (ii) can be done using automations in HA, as I am currently doing.  Maybe an update to the DOCs to state you must be in charge / discharge mode to auto manage saver sessions is sufficient to close (ii)

**Additional context**
Add any other context or screenshots about the feature request here.

gcoan, 2023-12-29:
If the saving session config in apps.yaml is commented out/ deleted then predbat won't join or manage any saving session activity.

I can add this to the documentation 

Geoffrey

mpartington, 2023-12-29:
That's a good suggestion, that would probably be the simplest way around it. As much as it like to see the savers session started in any mode, I think this could get terribly complicated. It is fairly easy to automate the rest using the savings session trigger (as per bottlecapdaves example)

closed 2023-12-29 by mpartington

gcoan, 2023-12-29:
@mpartington I've added this now to the latest draft of the Energy Rates documentation that I'm currently working on https://github.com/gcoan/batpred/blob/main/docs/energy-rates.md