springfall2008/batpred #547: Ability to limit the maximum charge rate

7Van737, 2023-12-29:
**Describe the bug**
Following this discussion https://community.givenergy.cloud/d/3696-first-night-live-on-predbat/67 I attampted to limit the maximum battery charge rate  by making the following addition to the apps.yaml 

  inverter_charge_limit:
   - 3000

This had no impact - my AIO still charged at max rate.

By setting  the "Set Charge Low Power Mode"  on, a reduced rate of charging was achieved.   I am not sure if the maths is correct however as the full charge was achieved in just over 4 hours of the available 7 hour window.

**Expected behavior**
Expected for the actual battery charge rate to be set at 3000W (as specified in the yaml).   This value did not appear to be successfully transmitted to the battery as the 'Battery Charge Power' register still read the previous default 6000W

**Predbat version**

7.14.26

**Environment details**

- All in One
- Standard HAOS installer
- 

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Don't think this is necessary.

springfall2008, 2023-12-29:
This is probably as it's 'inverter_limit_charge' as per the documentation:

inverter_limit_charge and inverter_limit_discharge - One per inverter (optional), when set overrides the maximum charge/discharge rate register settings used when controlling the inverter. This can be used for workarounds if you need to cap your inverter battery rate as Predbat overwrites the maximum rate registers when it enables a timed charge or discharge.

springfall2008, 2023-12-29:
> I am not sure if the maths is correct however as the full charge was achieved in just over 4 hours of the available 7 hour window.

I'd probably need a full logfile to review this one, if you can capture one from an overnight charging session I can look.

7Van737, 2023-12-29:
Thanks.   Will attach a log file over next few days when I’ve needed a decent charge.   Tonight will just need a small top up after some good solar generation today and not being in this evening.

7Van737, 2024-01-02:
I've continued to charge my AIO every night with "Set Charge Low Power Mode On" activated.   Every night the charge has completed early in the charge window.   Giving it some thought, I suspect this is due to the battery having 'lost' its calibration (SOC drop issue) and hence achieving the target SOC level with a lower energy demand than calculated - hence the earlier completion.    Suggest this issue is given no attention until BM8 is proven to be effective.

One other issue I have noticed is that on one occasion I charged the AIO with a set 'inverter_limit_charge' and the Low Power mode switched off - the battery successfully charged at the specified rate - but near to the end of the charge the rate did not decrease as I had expected as I have the 'battery_charge_power_curve' (your default settings) enabled in the apps.yaml.   I allowed the charge to continue till the SOC was at 95% before manually interrupting the charge.

springfall2008, 2024-01-14:
battery_power_curve models the maximum charge rate of the battery at certain high SOC % levels, it doesn't mean that Predbat will scale back the charge speed.

7Van737, 2024-01-16:
Thanks - please take any of my raised 'issues' (two so far) as observations/comments as I'm not sure what is actually a feature or is in fact a bug.

springfall2008, 2024-02-03:
I think I've done some work to fix this now, please look at the charge curve calibration feature and re-try

closed 2024-02-10 by springfall2008