springfall2008/batpred #548: user_config_enable?

gcoan, 2023-12-29:
**Describe the bug**
I'm working on tidying up the documentation, incorporating feedback from some of the new users and trying to make it easier to understand.

in config-yml-settings, the apps.yaml config item appears: 

**user_config_enable** - When True the user configuration is exposed in Home Assistant as input_number and switch, the config file becomes just the defaults to use

Its not clear to me what this config does, and looking at the predbat code I can't see that its actually read at all so it may just need deleting from the docs.

Or if it is needed, can you explain further how it works so I can update the docs.

**Expected behavior**
A clear and concise set of user instructions.

**Predbat version**
7.14.27

springfall2008, 2023-12-29:
user_config_enable is long gone, it used to enable the controls inside HA

gcoan, 2023-12-29:
Thanks @springfall2008 , I'll delete it out of the docs.
Will keep this issue open for any other queries I have.

gcoan, 2023-12-31:
@springfall2008 As I've been updating the instructions for defining predbat config https://github.com/gcoan/batpred/blob/main/docs/config-yml-settings.md, I've been thinking about apps.yaml, and updates to respond to recent install feedback and to make it easier for new users to edit.

I've been thinking that apps.yaml ought to be re-sequenced so that follows the same order as the documentation https://github.com/gcoan/batpred/blob/main/docs/config-yml-settings.md
Over time things have got added and some of them are grouped but others are a bit randomly located.

The other thing I was thinking about was to combine the apps.yaml for all the different inverter types into a single file.
Obviously this will make it longer and it'll contain stuff commented out for different inverters that doesn't apply for everyone, but it means that everyone gets the same apps.yaml of which a good 70% or so is common across inverters and it means you don't have to keep editing 3 different apps.yaml (and maybe more in future with other inverters). 

What do you think?

springfall2008, 2023-12-31:
Seems like a good plan to me, I’d be happy for you to reorder apps.yaml and maybe put some section dividers in place as well, perhaps even with links back to the docs?

gcoan, 2024-01-03:
Have incorporated the documentation updates from PR #566 and I've created my PR #570 for the work-to-date.

Looks like there are a couple of merge conflicts in customisation.md and video-guides.md.

Customisation.md I have dumped some stuff in there that was removed from config-yml-settings.md (related to predbat config options not apps.yaml settings).
e.g. Balance Inverters and Holiday Mode  
They are not really formatted, just included in relevant places as I planned to come back to that in a subsequent iteration.
Video-guides.md I just changed one reference to apps.yml to apps.yaml.

gcoan, 2024-01-09:
@springfall2008 can you add some documentation to apps-yaml.md (or send something to me so I can do it) about how load_forecast works?

The only doc is in the actual apps.yaml file and its very vague

I had a go at explaining how to use it in https://github.com/springfall2008/batpred/issues/574#issuecomment-1877863613 but this was from me reading the code, so is definitely incomplete

ta

closed 2024-01-20 by springfall2008