springfall2008/batpred #554: Exposing Import Optimization threshold as an entity/helper

AGLAISTER, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
When there is plunge pricing I pretty much always set my load scaling to 1.5 to maximize import, then as the rates increase I reduce the load scaling to avoid charging too much

**Describe the solution you'd like**
I would like to be able to automatically set load scaling based on the optimization threshold
So if the import optimisation threshold could be exposed externally then I could set up my own automation to decide when to increase and decrease the load scaling input

springfall2008, 2023-12-31:
I think you want 'predbat.rates' there is an attribute called 'threshold'  and also 'min', 'max' and 'average' which give data on the upcoming rates.

However I'm not sure this is a good thing to do, it maybe should you set load_scaling10 higher and adjust the pv10% threshold instead

AGLAISTER, 2023-12-31:
Hi Trefor,

Thanks for the information.
Load Scaling PV10% is currently set to 1.1 and pv scaling to 1.05

When I've had issues with the plan not reacting to input rates before the advice has always been to increase load scaling so this was why I asked how I could automate this

closed 2024-01-20 by springfall2008