springfall2008/batpred #560: select.predbat_update not populated or disappears

gcoan, 2023-12-31:
**Describe the bug**
Another user of predbat on the givenergy forum reported that they had done an auto-update and the select.predbat_update entity has disappeared, i.e. was on the dashboard but now didn't exist
https://community.givenergy.cloud/d/3696-first-night-live-on-predbat/103

I updated predbat to 7.14.29 and the entity is still there, but the dropdown was empty

5 minutes later, the drop-down is populated correctly

**Expected behavior**
Immediately after upgrade select.predbat_update continues to be populated with a list of predbat versions and never disappears or is blank

**Predbat version**
7.14.29 and 7.14.28

**Environment details**
HAOS

**Screenshots**
[image]

**Log file**
[appdaemon (7).log](https://github.com/springfall2008/batpred/files/13801747/appdaemon.7.log)

springfall2008, 2023-12-31:
I've put in a fix for this to stop the blank update, but be warned the selectors don't refresh their new values on the GUI for some time until you reload your browser window - not sure how I can fix that.

gcoan, 2023-12-31:
Thanks.  I'm not seeing the .30 update yet in the drop-down, despite refreshing the browser window, navigating on/off the page.
I'm guessing that its because there is caching on how often predbat populates the drop-down that's not expired yet.

gcoan, 2023-12-31:
And now the .30 release has appeared and I didn't refresh the browser window so it must have been stuck in a predbat cache of how often it checks github.

springfall2008, 2024-01-01:
Predbat only checks github every few hours

gcoan, 2024-01-01:
Thought so.

But the good news is that when it found the new release I didn't have to refresh the browser, the new version appeared in the drop down