springfall2008/batpred #561: select.predbat_update entity disappeared from dashboard

Dave-Parry, 2023-12-31:
I added the dashboard.yaml file to a card on my dashboard and all was well, but now select.predbat_update is reported as not available.
On searching it's not listed as an entity and is not now in the yaml file. It must have been there before. Could this have been because predbat did an auto update to v7.14.29?

Dave-Parry, 2023-12-31:
Also noticed I now don't have the Auto update toggle either.
[image]

Dave-Parry, 2023-12-31:
Closed by mistake

springfall2008, 2023-12-31:
What version are you running, it's saying .29 is the latest but maybe you are on an older version that doesn't have this feature?

Dave-Parry, 2023-12-31:
It said I was on 29 but I think I saw v21 listed somewhere (I think in appdaemon logs maybe?) just seen a update to 30 which is loading and I seem to have the update sensor now listed.

Dave-Parry, 2023-12-31:
That seems to have worked. Have both the entities that were missing. Thank you.

closed 2023-12-31 by Dave-Parry