springfall2008/batpred #568: Car Charging SOC - Manual Update

dandwhelan, 2024-01-02:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

I want to tell Predbat not to check my state of charge after it's updated, as I have a user input SOC. 

I've setup HA/Predbat to charge the car using the cheap slots. The future cheap slots were ignored once the charging started because I assumed it was looking for a SOC change. So the car just keeps charging until it's full. 

**Describe alternatives you've considered**

I've noticed on Predbat that it does calculate the estimated Kws it will be putting into the car. Maybe when the manual SOC is set it uses the estimated KW's? 

**Additional context**

I own an MG 4, which doesn't have feasible integration into Homeassistant. I have created a few helpers in HA which I manually enter the cars SOC. I've also created one for the charge limit and made a toggle when the car is plugged in. In Predbat, I've pointed the configuration on predbat to a manually updated helper. I have no other way to get the SOC from the MG unless.

springfall2008, 2024-01-02:
Please can you test the version on 'main' which contains this feature, if you turn on switch.predbat_car_charging_manual_soc then Predbat will create input_number.car_charging_manual_soc_kwh which will be your cars current charge level. You need to set this in kWh manually before a charge but Predbat _should_ increment it during a charge session. I've not tested it myself fully so let me know.

dandwhelan, 2024-01-03:
Thanks for looking at this so quickly. I'll try it out tonight. Should I hash out the old soc input on the app.yaml file now this had been enabled. 

[image]
[image]

springfall2008, 2024-01-03:
The old soc in apps.yaml will be ignored but you can remove it if you like

dandwhelan, 2024-01-03:
I have the car soc set to 72 and target is 80. The number of kw planned overnight doesn't look right. I'm not sure why it says disabled for the soc... 
[image]
[apps (1).txt](https://github.com/springfall2008/batpred/files/13823781/apps.1.txt)
[appdaemon.log](https://github.com/springfall2008/batpred/files/13823782/appdaemon.log)

springfall2008, 2024-01-03:
The manual soc number is in kWh and the battery size is kWh but the target is in %

dandwhelan, 2024-01-03:
[image]
Ahhh yes my mistake. I got my calculator out and the car kws charging looks better. I'll report back on the morning how it went.

dandwhelan, 2024-01-04:
That worked a treat. Thanks a lot.
[image]
 I'm happy you added this feature.

closed 2024-01-04 by dandwhelan