springfall2008/batpred #571: Storm mode

phiebs, 2024-01-03:
A feature, similar to what is built into Powerwalls, to override the schedule/cost part of the system temporarily in preparation for a storm or other event where there is a high liklihood of grid disconnection to provide as much "island mode" energy as possible.

I've managed to get this working by having an automation to set predbat to Idle and then manually trigger a charge directly with the battery/inverter, but it would be neat to have this built into predbat (if it already is and I've missed it in the documentation then apologies!)

The storms across the UK yesterday resulted in our grid power being off for around 6 hours, and while the battery (AIO) did last throughout, it would have been great to pay the additional grid cost to keep the battery at 100% and disregard the plan entirely until the storm had passed. Perhaps an on/off switch that you can set?

phiebs, 2024-01-03:
I think I may have answered my own question, but can I just simply set the "Reserve Min" to 100% prior to an expected storm/grid disconnect and then set it back to my usual setting after the event?

gcoan, 2024-01-03:
Yes that'd work.

I can add a bit to the documentation to highlight this scenario.

Geoffrey

closed 2024-01-04 by phiebs