springfall2008/batpred #572: Ohme HA integration

mrb090909, 2024-01-03:
Request to integrate Ohme API into Predbat as there is now a Ohme integration

Ability to see that the car is charging due to intelligent octopus and trigger the batter to charge also / hold the battery / stop the car being charged by the battery.

**Describe alternatives you've considered**
Currently using Car charging true switch by editing the Apps.yaml file

**Additional context**
Sensors attached
sensor.ohme_accumulative_energy
Ohme
—
Approve Charge
button.ohme_approve_charge
Ohme
—
Car Charging
binary_sensor.ohme_car_charging
Ohme
—
Car Connected
binary_sensor.ohme_car_connected
Ohme
—
Charge Slot Active
binary_sensor.ohme_slot_active
Ohme
—
CT Reading
sensor.ohme_ct_reading
Ohme
—
Current Draw
sensor.ohme_current_draw
Ohme
—
Next Charge Slot End
sensor.ohme_next_slot_end
Ohme
—
Next Charge Slot Start
sensor.ohme_next_slot
Ohme
—
Pending Approval
binary_sensor.ohme_pending_approval
Ohme
—
Power Draw
sensor.ohme_power_draw
Ohme
—
Target Percentage
number.ohme_target_percent
Ohme
—
Target Time
time.ohme_target_time
Ohme
—
Voltage
sensor.ohme_voltage

[image]

springfall2008, 2024-01-03:
My first guess on how to configure this would be:

  car_charging_energy: sensor.ohme_accumulative_energy

  num_cars: 1

  car_charging_planned: binary_sensor.ohme_car_connected

  car_charging_now: binary_sensor.ohme_car_charging

  car_charging_battery_size:
    - YOUR_CAR_BATTERY_SIZE_KWH
  car_charging_limit:
    - number.ohme_target_percent

  car_charging_soc:
    - ????

^^ Do you have a SOC sensor for your car available, if not maybe try using the new Manual SOC feature I just released?

mrb090909, 2024-01-04:
So as Ohme isn't talking properly to intelligent at the moment when I plug the car in it will either start to charge straight away because it picks up a slot or waits until there is a slot and starts charging. there isnt anything telling predbad to take advantage of the cheap rate as its not seeing it so predate continues as normal and the house battery dumps into the car. if predate can see "Ohme is charging" and trigger a battery charge and when home stops charging it stops the charge and returns to normal operation then that would work well.

Slot stuff when I plug in that doesn't seem to show anything at the moment.

Just to throw a spanner in the works for the car battery size, we have a second EV coming next week which is 88.5kw and the one we have now is 48kw so I will be playing the cheap rate game with both cars.

No soc sensor for the car as it is dumb no api or link at all (had the first MG ZS)

mrb090909, 2024-01-04:
Ive added your guess to the apps.yaml so lets see next time I plug the car in what happens

jeldwendean, 2024-02-04:
I just wanted to add to this as I’d also like to see this working. Ohme on Intelligent Octopus like you say doesn’t work through the Octopus Energy integration and that’s because the control has been done in the Ohme app and not the Octopus app. That’s due to change in the foreseeable but for now it’d be good to have something working.

I thought about mapping octopus_intelligent_slot, octopus_ready_time and octopus_charge_limit in the apps.yaml file to the Ohme sensors but it seems that octopus_intelligent_slot is looking for the planned_dispatches attribute which the Ohme integration doesn’t have (it’s formatted differently and is on its own sensor).

For now I’ve just mapped in the apps.yaml file to let Predbat know when the Ohme is planned to charge (but this just sets the whole off-peak period to charge which could be wrong) and when the Ohme is charging to stop the car from charging from the battery.

Ideally Predbat would know when Ohme is planning to charge using the slots. I’m not sure there is an easy way to do this?

gcoan, 2024-02-04:
Can you share what's produced in the Ohme sensor that you have mapped at the moment to the octopus_intelligent_slot and the other Ohme sensor that contains the planned despatches info (albeit formatted differently)

It seems to me that should be possible to create a template sensor that maps the Ohme sensors into an sensor that ca. be used in place of octopus_intelligent_slot, with the data in the right places/structures

jeldwendean, 2024-02-04:
Yeah sure:

octopus_intelligent_slot > binary_sensor.ohme_slot_active

This basically goes on/off if the slot is active so if we’re in a charging slot (e.g. let’s say I was given a slot of 8:30-9:30pm then this would be on during that hour, works same as the Octopus one except it doesn’t have any attributes)

octopus_ready_time > time.ohme_target_time

This is when you need your car charged by 

octopus_charge_limit > number.ohme_target_percent

This is how much charge you want to add 

Finally we have Charge Slots which is basically the planned_dispatches attribute of octopus_intelligent_slot on its own sensor and formatted differently 

[image]
[image]
[image]
[image]

springfall2008, 2024-02-05:
For 'charge_slots' can you snapshot some valid data using the developer tools, states menu in HA so I can see the exact format of the sensor?

jeldwendean, 2024-02-06:
Dan who set up the Ohme HA integration has kindly changed the Charge Slot Active sensor to mimic the Octopus Energy sensor which is in beta and I’m testing right now so we might have a solution!

springfall2008, 2024-02-06:
That would be even better, let me know if it works

jeldwendean, 2024-02-07:
I come with good news, Dan, the developer of the Ohme integration in HA has updated one of the sensors to mimic the Octopus Energy sensor. I tested it last night and it works a treat. I must say, I was quite nervous as I woke up at 6:45 this morning to find my battery discharging as I was given a slot at 7am and the car did in fact start charging at 7 so the home battery got topped back up. Phew!

Here's how I've mapped them:

octopus_intelligent_slot: 'binary_sensor.ohme_slot_active'
octopus_ready_time: 'time.ohme_target_time'
octopus_charge_limit: 'number.ohme_target_percent'

You'll need to be on the latest Ohme integration release (v0.6.1) and this should work nicely.

I do have some questions though just to make sure I'm 100% on this:

I've got car_charging_now set in my apps.yaml file just so Predbat knows the car is actually charging. Is this good practice?

e.g. car_charging_now:
    - 'binary_sensor.ohme_car_charging'

I've also set car_charging_energy to a sensor that shows the amount of kWh added for that charge session (e.g. if I'm charging from 70% to 100% then I might add 20kWh in a session). Is this also expected or should it be an accumulative sensor since day 1?

e.g. car_charging_energy: 'sensor.ohme_session_energy'

Thanks!

jeldwendean, 2024-02-07:
What happens if the car is due to charge during a slot but doesn’t actually charge? I would’ve thought that Predbat will plan based on the slots provided but knowing if the car is actually charging or not will prevent it from charging up on a peak rate slot - so is car_charging_now not being used as a fail safe?

springfall2008, 2024-02-07:
All slots reported by the IO plug-in were assumed to be cheap rate. Normally with IO if the car doesn't charge then they take the slot away and things adjust. I don't know with Ohme if this also applies or not. Car charging now doesn't currently cancel slots it just triggers them.

jeldwendean, 2024-02-07:
Hmm I’m not sure on that

You could have a SOC of 70% and put that need to add 100% charge so you end up getting early evening slots until say 7am but your car would only charge from say 9pm and then stop at 2am (because you only needed 30% not 100%) - you would still have the slots until 7am but they wouldn’t be classed as off-peak as the car isn’t charging

I wasn’t aware IO removes the slots if the car doesn’t charge

springfall2008, 2024-02-08:
> Hmm I’m not sure on that
>
> You could have a SOC of 70% and put that need to add 100% charge so you end up getting early evening slots until say 7am but your car would only charge from say 9pm and [...]

For me the car always charges in the slots given so I have never seen the issue. If you have the problem do let me know and I could add a ignore option to discard slots if the car doesn't actually charge but its not trivial as you need a certain time period also.

springfall2008, 2024-02-10:
Closing this one as I think the integration is done and documented.

Feel free to raise a new ticket if you have issues with IO

closed 2024-02-10 by springfall2008

HnabrU, 2024-02-25:
I think this needs to be re-open.
Issue: when car disconnected from Ohme while in between IO charging slots, then Predbat seems to lose all historical data regarding previous charging slots and assumes previous electricity import was at peak rate.

Ohme notification with slots:
[image]

Initial plan card while the car was plugged in:
[image]

Predbat plan card when car disconnected at around 18:22:
[image]

History:
[image]

springfall2008, 2024-02-25:
Is the data still available somewhere, can you find the history?

HnabrU, 2024-02-25:
Yes, can find them here:

[image]

However costs somehow returned to the correct ones at 19:45:

[image]