springfall2008/batpred #574: Not utilising battery

naltsta, 2024-01-03:
**Describe the bug**
Not choosing cheapest slots to charge

**Expected behavior**
Cheapest slots selected

**Predbat version**

7.14.31

**Environment details**

- Hybrid and 2 batteries
- Standard HAOS installer 
- Anything else?

**Screenshots**
[image]

**Log file**
2024-01-03 20:56:21.304329 INFO pred_bat: Optimise price set 25.0 start_at_low True best_price 29.0
2024-01-03 20:56:22.107520 INFO pred_bat: Try optimising charge window(s)    29: 01-04 15:00:00 - 01-04 15:30:00 price 20.53 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 392.3, 0.21: 388.12}
2024-01-03 20:56:22.117346 INFO pred_bat: Optimise price set 25.8 start_at_low True best_price 29.0
2024-01-03 20:56:22.954790 INFO pred_bat: Try optimising charge window(s)    22: 01-04 10:00:00 - 01-04 10:30:00 price 21.25 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 390.86, 0.21: 388.12}
2024-01-03 20:56:22.964790 INFO pred_bat: Optimise price set 25.9 start_at_low True best_price 29.0
2024-01-03 20:56:23.767574 INFO pred_bat: Try optimising charge window(s)    26: 01-04 13:00:00 - 01-04 13:30:00 price 21.35 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 388.9, 0.21: 388.12}
2024-01-03 20:56:23.777286 INFO pred_bat: Optimise price set 26.6 start_at_low True best_price 29.0
2024-01-03 20:56:25.113291 INFO pred_bat: Try optimising charge window(s)    20: 01-04 08:30:00 - 01-04 09:30:00 price 21.98 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.97: 388.02, 4.72: 388.02, 4.47: 388.02, 0.0: 391.95, 0.21: 388.12}
2024-01-03 20:56:25.900991 INFO pred_bat: Try optimising charge window(s)    23: 01-04 10:30:00 - 01-04 11:00:00 price 21.96 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 388.17, 0.21: 388.12}
2024-01-03 20:56:25.910802 INFO pred_bat: Optimise price set 27.0 start_at_low True best_price 29.0
2024-01-03 20:56:27.486642 INFO pred_bat: Try optimising charge window(s)    28: 01-04 14:00:00 - 01-04 15:00:00 price 22.39 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.97: 388.02, 4.72: 388.02, 4.47: 388.02, 4.22: 388.02, 3.97: 388.02, 3.72: 388.02, 0.0: 395.15, 0.21: 388.12}
2024-01-03 20:56:27.496590 INFO pred_bat: Optimise price set 27.4 start_at_low True best_price 29.0
2024-01-03 20:56:28.226453 INFO pred_bat: Try optimising charge window(s)    21: 01-04 09:30:00 - 01-04 10:00:00 price 22.7 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 388.82, 0.21: 388.12}
2024-01-03 20:56:28.236605 INFO pred_bat: Optimise price set 27.9 start_at_low True best_price 29.0
2024-01-03 20:56:28.976921 INFO pred_bat: Try optimising charge window(s)    38: 01-04 20:00:00 - 01-04 20:30:00 price 23.18 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 389.89, 0.21: 388.12}
2024-01-03 20:56:28.987656 INFO pred_bat: Optimise price set 29.0 start_at_low True best_price 29.0
2024-01-03 20:56:29.823996 INFO pred_bat: Try optimising charge window(s)    30: 01-04 15:30:00 - 01-04 16:00:00 price 24.2 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 394.57, 0.21: 388.12}
2024-01-03 20:56:29.834785 INFO pred_bat: Optimise price set 30.5 start_at_low True best_price 29.0
2024-01-03 20:56:30.673144 INFO pred_bat: Try optimising charge window(s)    36: 01-04 19:00:00 - 01-04 19:30:00 price 25.6 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 388.89, 0.21: 388.12}
2024-01-03 20:56:30.687615 INFO pred_bat: Optimise price set 40.3 start_at_low True best_price 29.0
2024-01-03 20:56:31.510182 INFO pred_bat: Try optimising charge window(s)    35: 01-04 18:30:00 - 01-04 19:00:00 price 34.73 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 388.45, 0.21: 388.12}
2024-01-03 20:56:31.519927 INFO pred_bat: Optimise price set 41.8 start_at_low True best_price 29.0
2024-01-03 20:56:32.306213 INFO pred_bat: Try optimising charge window(s)    31: 01-04 16:00:00 - 01-04 16:30:00 price 36.18 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 392.35, 0.21: 388.12}
2024-01-03 20:56:32.315888 INFO pred_bat: Optimise price set 42.9 start_at_low True best_price 29.0
2024-01-03 20:56:33.092342 INFO pred_bat: Try optimising charge window(s)    34: 01-04 18:00:00 - 01-04 18:30:00 price 37.13 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 390.5, 0.21: 388.12}
2024-01-03 20:56:33.102602 INFO pred_bat: Optimise price set 43.6 start_at_low True best_price 29.0
2024-01-03 20:56:34.030714 INFO pred_bat: Try optimising charge window(s)    32: 01-04 17:00:00 - 01-04 17:30:00 price 37.82 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.28, 4.22: 388.28, 0.0: 391.74, 0.21: 388.38}
2024-01-03 20:56:34.039763 INFO pred_bat: Optimise price set 44.4 start_at_low True best_price 29.0
2024-01-03 20:56:34.808609 INFO pred_bat: Try optimising charge window(s)    33: 01-04 17:30:00 - 01-04 18:00:00 price 38.58 cost 461.23 metric 388.01 keep 0.0 selected 5.22 was 5.22 results {5.22: 388.01, 4.72: 388.02, 0.0: 391.19, 0.21: 388.12}
2024-01-03 20:56:34.828106 INFO pred_bat: Best charge windows in price group 44.4 best_metric 388.01 best_cost 461.23 metric_keep 0.0 windows [ 03-01 20:00:00 - 03-01 21:00:00 @ 18.2p 0.21%, 03-01 21:00:00 - 03-01 22:00:00 @ 18.41p 5.22%, 03-01 22:00:00 - 03-01 22:30:00 @ 18.38p 5.22%, 03-01 22:30:00 - 03-01 23:00:00 @ 16.4p 5.22%, 03-01 23:00:00 - 04-01 00:00:00 @ 16.78p 5.22%, 04-01 00:00:00 - 04-01 00:30:00 @ 16.27p 5.22%, 04-01 00:30:00 - 04-01 01:00:00 @ 18.17p 5.22%, 04-01 01:00:00 - 04-01 01:30:00 @ 18.6p 5.22%, 04-01 01:30:00 - 04-01 02:00:00 @ 16.33p 5.22%, 04-01 02:00:00 - 04-01 02:30:00 @ 16.74p 5.22%, 04-01 02:30:00 - 04-01 03:30:00 @ 16.12p 5.22%, 04-01 03:30:00 - 04-01 04:00:00 @ 15.16p 5.22%, 04-01 04:00:00 - 04-01 04:30:00 @ 16.27p 5.22%, 04-01 04:30:00 - 04-01 05:00:00 @ 15.12p 5.22%, 04-01 05:00:00 - 04-01 05:30:00 @ 16.27p 5.22%, 04-01 05:30:00 - 04-01 06:00:00 @ 17.07p 5.22%, 04-01 06:00:00 - 04-01 06:30:00 @ 18.96p 5.22%, 04-01 06:30:00 - 04-01 07:00:00 @ 17.78p 5.22%, 04-01 07:00:00 - 04-01 08:00:00 @ 18.96p 5.22%, 04-01 08:00:00 - 04-01 08:30:00 @ 19.19p 5.22%, 04-01 08:30:00 - 04-01 09:30:00 @ 21.98p 5.22%, 04-01 09:30:00 - 04-01 10:00:00 @ 22.7p 5.22%, 04-01 10:00:00 - 04-01 10:30:00 @ 21.25p 5.22%, 04-01 10:30:00 - 04-01 11:00:00 @ 21.96p 5.22%, 04-01 11:00:00 - 04-01 12:00:00 @ 19.04p 5.22%, 04-01 12:00:00 - 04-01 13:00:00 @ 19.28p 5.22%, 04-01 13:00:00 - 04-01 13:30:00 @ 21.35p 5.22%, 04-01 13:30:00 - 04-01 14:00:00 @ 19.8p 5.22%, 04-01 14:00:00 - 04-01 15:00:00 @ 22.39p 5.22%, 04-01 15:00:00 - 04-01 15:30:00 @ 20.53p 5.22%, 04-01 15:30:00 - 04-01 16:00:00 @ 24.2p 5.22%, 04-01 16:00:00 - 04-01 16:30:00 @ 36.18p 5.22%, 04-01 17:00:00 - 04-01 17:30:00 @ 37.82p 5.22%, 04-01 17:30:00 - 04-01 18:00:00 @ 38.58p 5.22%, 04-01 18:00:00 - 04-01 18:30:00 @ 37.13p 5.22%, 04-01 18:30:00 - 04-01 19:00:00 @ 34.73p 5.22%, 04-01 19:00:00 - 04-01 19:30:00 @ 25.6p 5.22%, 04-01 19:30:00 - 04-01 20:00:00 @ 19.56p 5.22%, 04-01 20:00:00 - 04-01 20:30:00 @ 23.18p 5.22%, 04-01 20:30:00 - 04-01 21:00:00 @ 19.71p 5.22%, 04-01 21:00:00 - 04-01 21:30:00 @ 21.86p 5.22%, 04-01 21:30:00 - 04-01 22:00:00 @ 19.27p 5.22%, 04-01 22:00:00 - 04-01 22:30:00 @ 17.78p 5.22%, 04-01 22:30:00 - 04-01 23:00:00 @ 15.46p 5.22%, 04-01 23:00:00 - 05-01 00:00:00 @ 16.78p 5.22%, 05-01 00:00:00 - 05-01 00:30:00 @ 16.27p 5.22%, 05-01 00:30:00 - 05-01 01:00:00 @ 18.17p 5.22%, 05-01 01:00:00 - 05-01 01:30:00 @ 18.6p 5.22%, 05-01 01:30:00 - 05-01 02:00:00 @ 16.33p 5.22%, 05-01 02:00:00 - 05-01 02:30:00 @ 16.74p 5.22%, 05-01 02:30:00 - 05-01 03:30:00 @ 16.12p 5.22%, 05-01 03:30:00 - 05-01 04:00:00 @ 15.16p 5.22%, 05-01 04:00:00 - 05-01 04:30:00 @ 16.27p 5.22%, 05-01 04:30:00 - 05-01 05:00:00 @ 15.12p 5.22%, 05-01 05:00:00 - 05-01 05:30:00 @ 16.27p 5.22%, 05-01 05:30:00 - 05-01 06:00:00 @ 17.07p 5.22%, 05-01 06:00:00 - 05-01 06:30:00 @ 18.96p 5.22%, 05-01 06:30:00 - 05-01 07:00:00 @ 17.78p 5.22%, 05-01 07:00:00 - 05-01 08:00:00 @ 18.96p 5.22%, 05-01 08:00:00 - 05-01 08:30:00 @ 19.19p 5.22%, 05- [... cut ...]
2024-01-03 20:56:47.154646 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 15:00:00 - 15:31:00 to None - None
2024-01-03 20:56:47.167967 INFO pred_bat: Within charge freeze setting target soc to current soc 100
2024-01-03 20:56:47.182901 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-01-03 20:56:47.196160 INFO pred_bat: Adjust reserve to hold current soc 100 % (set_reserve_enable is true)
2024-01-03 20:56:47.207847 INFO pred_bat: Inverter 0 Current reserve is 98.0 already at target
2024-01-03 20:56:47.271984 INFO pred_bat: Completed run status Freeze charging

springfall2008, 2024-01-03:
Can you attached a full cycle of the log (one 5 minute update), this one is incomplete?

naltsta, 2024-01-03:
2024-01-03 21:11:50.144092 INFO pred_bat: Updating HA config expert_mode to True
2024-01-03 21:11:50.180534 INFO pred_bat: Updating HA config pv_metric10_weight to 0.15
2024-01-03 21:11:50.221101 INFO pred_bat: Updating HA config pv_scaling to 1.0
2024-01-03 21:11:50.260663 INFO pred_bat: Updating HA config load_scaling to 1.0
2024-01-03 21:11:50.297590 INFO pred_bat: Updating HA config load_scaling10 to 1.1
2024-01-03 21:11:50.334170 INFO pred_bat: Updating HA config battery_rate_max_scaling to 1.0
2024-01-03 21:11:50.373333 INFO pred_bat: Updating HA config battery_loss to 0.03
2024-01-03 21:11:50.411501 INFO pred_bat: Updating HA config battery_loss_discharge to 0.03
2024-01-03 21:11:50.447594 INFO pred_bat: Updating HA config inverter_loss to 0.04
2024-01-03 21:11:50.486384 INFO pred_bat: Updating HA config inverter_hybrid to True
2024-01-03 21:11:50.527503 INFO pred_bat: Updating HA config inverter_soc_reset to False
2024-01-03 21:11:50.563834 INFO pred_bat: Updating HA config battery_capacity_nominal to False
2024-01-03 21:11:50.600979 INFO pred_bat: Updating HA config car_charging_energy_scale to 1.0
2024-01-03 21:11:50.640956 INFO pred_bat: Updating HA config car_charging_threshold to 6.0
2024-01-03 21:11:50.679033 INFO pred_bat: Updating HA config car_charging_rate to 7.4
2024-01-03 21:11:50.712572 INFO pred_bat: Updating HA config car_charging_loss to 0.08
2024-01-03 21:11:50.748358 INFO pred_bat: Updating HA config best_soc_min to 0.0
2024-01-03 21:11:50.782195 INFO pred_bat: Updating HA config best_soc_max to 0.0
2024-01-03 21:11:50.820417 INFO pred_bat: Updating HA config best_soc_keep to 1.0
2024-01-03 21:11:50.855322 INFO pred_bat: Updating HA config metric_min_improvement to 0.0
2024-01-03 21:11:50.889965 INFO pred_bat: Updating HA config metric_min_improvement_discharge to 0.3
2024-01-03 21:11:50.926464 INFO pred_bat: Updating HA config metric_battery_cycle to 3.0
2024-01-03 21:11:50.964925 INFO pred_bat: Updating HA config metric_battery_value_scaling to 1.0
2024-01-03 21:11:51.001305 INFO pred_bat: Updating HA config metric_future_rate_offset_import to 0.0
2024-01-03 21:11:51.035540 INFO pred_bat: Updating HA config metric_future_rate_offset_export to 0.0
2024-01-03 21:11:51.070959 INFO pred_bat: Updating HA config metric_inday_adjust_damping to 0.95
2024-01-03 21:11:51.107097 INFO pred_bat: Updating HA config metric_cloud_enable to True
2024-01-03 21:11:51.142193 INFO pred_bat: Set reset inverter true due to reset_inverter on item set_reserve_min
2024-01-03 21:11:51.148566 INFO pred_bat: Updating HA config set_reserve_min to 4.0
2024-01-03 21:11:51.182059 INFO pred_bat: Updating HA config rate_low_threshold to 0.0
2024-01-03 21:11:51.217809 INFO pred_bat: Updating HA config rate_high_threshold to 0.0
2024-01-03 21:11:51.252165 INFO pred_bat: Set reset inverter true due to reset_inverter on item car_charging_hold
2024-01-03 21:11:51.259540 INFO pred_bat: Updating HA config car_charging_hold to False
2024-01-03 21:11:51.302278 INFO pred_bat: Updating HA config car_charging_manual_soc to False
2024-01-03 21:11:51.336236 INFO pred_bat: Note: Disabled configuration item car_charging_manual_soc_kwh
2024-01-03 21:11:51.346016 INFO pred_bat: Updating HA config octopus_intelligent_charging to False
2024-01-03 21:11:51.383986 INFO pred_bat: Updating HA config octopus_intelligent_ignore_unplugged to False
2024-01-03 21:11:51.421906 INFO pred_bat: Updating HA config car_charging_plan_smart to False
2024-01-03 21:11:51.456496 INFO pred_bat: Set reset inverter true due to reset_inverter on item car_charging_from_battery
2024-01-03 21:11:51.464326 INFO pred_bat: Updating HA config car_charging_from_battery to False
2024-01-03 21:11:51.503744 INFO pred_bat: Updating HA config calculate_discharge_oncharge to True
2024-01-03 21:11:51.542976 INFO pred_bat: Updating HA config calculate_fast_plan to False
2024-01-03 21:11:51.581433 INFO pred_bat: Updating HA config calculate_second_pass to False
2024-01-03 21:11:51.622042 INFO pred_bat: Updating HA config calculate_tweak_plan to True
2024-01-03 21:11:51.657907 INFO pred_bat: Updating HA config calculate_inday_adjustment to True
2024-01-03 21:11:51.702007 INFO pred_bat: Updating HA config calculate_plan_every to 6.0
2024-01-03 21:11:51.734194 INFO pred_bat: Updating HA config combine_charge_slots to False
2024-01-03 21:11:51.771062 INFO pred_bat: Updating HA config combine_discharge_slots to False
2024-01-03 21:11:51.805580 INFO pred_bat: Updating HA config set_status_notify to False
2024-01-03 21:11:51.839979 INFO pred_bat: Updating HA config set_inverter_notify to False
2024-01-03 21:11:51.881004 INFO pred_bat: Set reset inverter true due to reset_inverter on item set_charge_freeze
2024-01-03 21:11:51.889772 INFO pred_bat: Updating HA config set_charge_freeze to True
2024-01-03 21:11:52.030072 INFO pred_bat: Set reset inverter true due to reset_inverter on item set_charge_low_power
2024-01-03 21:11:52.037194 INFO pred_bat: Updating HA config set_charge_low_power to False
2024-01-03 21:11:52.082564 INFO pred_bat: Set reset inverter true due to reset_inverter on item set_reserve_enable
2024-01-03 21:11:52.089689 INFO pred_bat: Updating HA config set_reserve_enable to True
2024-01-03 21:11:52.132087 INFO pred_bat: Set reset inverter true due to reset_inverter on item set_discharge_freeze_only
2024-01-03 21:11:52.138421 INFO pred_bat: Updating HA config set_discharge_freeze_only to False
[... cut ...]
2024-01-03 21:14:14.890714 INFO pred_bat: Inverter 0 current discharge rate is 0 and new target is 2600
2024-01-03 21:14:30.815606 INFO pred_bat: Inverter 0 set discharge rate 2600 via REST successful on retry 0
2024-01-03 21:14:30.891523 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 15:00:00 - 15:31:00 to None - None
2024-01-03 21:14:30.899930 INFO pred_bat: Charge window will be: 2024-01-03 21:00:00+00:00 - 2024-01-04 16:30:00+00:00 - current soc 100 target 4
2024-01-03 21:14:30.908593 INFO pred_bat: Inverter 0 current discharge rate is 2600 and new target is 0

naltsta, 2024-01-03:
Sorry had to enable the logging first - is this what you need?

springfall2008, 2024-01-03:
Sure looks good.

Some minor things:

> 2024-01-03 21:11:51.622042 INFO pred_bat: Updating HA config calculate_tweak_plan to True
> 2024-01-03 21:11:51.702007 INFO pred_bat: Updating HA config calculate_plan_every to 6.0

It might be worth setting tweak plan to False and plan_every to either 5 or 10, however I doubt either will change things much.

>2024-01-03 21:11:50.926464 INFO pred_bat: Updating HA config metric_battery_cycle to 3.0

You could reduce battery cycle down to 2.0 or 1.0 and see if it changes anything.

>2024-01-03 21:12:20.502991 INFO pred_bat: New Inverter 0 with soc_max 5.22 kWh nominal_capacity 5.22 kWh battery rate raw 2600.0 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 3.6 kW export limit 3.6 kW reserve 4.0 % current_reserve 98.0 %

Is this right, you said you have two batteries but it's reporting only 5.22 kWh of battery capacity in total??

dandwhelan, 2024-01-03:
Mine is looking very similar with 1 8.2kw battery
[apps (1).txt](https://github.com/springfall2008/batpred/files/13823762/apps.1.txt)
[appdaemon.log](https://github.com/springfall2008/batpred/files/13823763/appdaemon.log)

naltsta, 2024-01-03:
> It might be worth setting tweak plan to False and plan_every to either 5 or 10, however I doubt either will change things much.
Will give them a go - does Predbat only like multiples of 5 for some reason?

> You could reduce battery cycle down to 2.0 or 1.0 and see if it changes anything.

Will try

> Is this right, you said you have two batteries but it's reporting only 5.22 kWh of battery capacity in total??

Sadly yes, 2x2.6kWh

naltsta, 2024-01-03:
[image]

5 minutes, tweak off, battery cycle down to 1.0 and still pretty weird...

springfall2008, 2024-01-04:
Can you try best soc keep as 0?

naltsta, 2024-01-04:
[image]

The 5pm charge slot disappeared over night but id really like that 19.30 to go too! Best keep now 0

springfall2008, 2024-01-04:
How do the overnight rates compare can you show the full plan? I think if you have enough charge left that 7:30pm slot will vanish later in the day

dandwhelan, 2024-01-04:
[image]
I've been keeping at eye out on this issue as I feel the battery should be utilised more. I've tried setting best soc 0 and it doesn't change.

naltsta, 2024-01-04:
[image]
 The cheapest slot is at 22.30 tonight, my battery will be empty but it won't bother to charge...

But then it will charge tomorrow morning when it's much more expensive?

Seems to me that the relatively flat rates have predbat making some strange decisions.

naltsta, 2024-01-04:
> ![Screenshot_20240104_162158_Home Assistant](https://private-user-images.githubusercontent.com/79152287/294238021-4fec5edc-8dff-45d1-a24a-aa290ccdd2b8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzODkwOTMsIm5iZiI6MTcwNDM4ODc5MywicGF0aCI6Ii83OTE1MjI4Ny8yOTQyMzgwMjEtNGZlYzVlZGMtOGRmZi00NWQxLWEyNGEtYWEyOTBjY2RkMmI4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA0VDE3MTk1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkNzQ3MTFiZDk4ODljNWZlOTYzMmMyYTExYTExOGI4OTY3OWE5ZjY5MzQwODE5Njc2MmQ1N2I5NmMwMjFlOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0DQgb0drxh-mSSdxpuJkut9HfPVp2fHmIK24e1v30Ak) I've been keeping at eye out on this issue as I feel the battery should be utilised more. I've tried setting best soc 0 and it doesn't change.

Absolutely - Freeze at 75 until 22.30 would seem like a better plan

gcoan, 2024-01-04:
Tuesday I think Predbat seemed to struggle with the flat rates (and my export rate had gone up slightly on 1/1), so it wanted to discharge for no benefit to recharge later on.
I set it to read-only mode and stopped the discharge, played around a bit with setting mode to Charging only (which improved things), but after that its been quite OK with charge and discharge planning.

But tomorrow morning we have a power-up event from 7-9am (first time we've had one so early in the morning) and predbat v7.14.31 is really struggling to manage the battery for it. A bit of charging is fine to get through the 21:00 slot, 00:30 slot, and later on a bit of charge to get through the 06:00 slot, but otherwise just  keeping the soc low ready for the power up event would seem best, but instead the soc is being held at 86% all night so its having to do a force discharge in the power-up event to get the battery lower to take advantage of the free electricity.
[image]

I accidentally installed 7.14.29 - plan was the same.  Installed the new 7.14.32 and other than 20:30 now being a freeze charge rather than a charge, its the same issue, too much soc being retained through the night ahead of the zero rate period at 7am.
[image]

Changing mode to Charging only produces a much more sensible plan, overnight it only charges enough to get through the 06:00 more expensive slot
[image]

Its a cheaper plan as well, by 9am, its 18p cheaper than the Charging and Discharging plan

[appdaemon (12).log](https://github.com/springfall2008/batpred/files/13835493/appdaemon.12.log)

@naltsta @dandwhelan have you got Mode set to Charging only or Charging and Discharging?

dandwhelan, 2024-01-04:
I'm in charge only mode but set to monitor for the evenings.

I've a mixergy hot water tank which can charge on electric on cheap agile
rates and gas when it's expensive. I'm thinking maybe some of my data is
off because of this. Maybe I should put a request in to treat this hot
water tank as a car charger.

naltsta, 2024-01-04:
Switching between charging only and charging and discharging does not seem to change my plan (did check it had updated!) normally running on charge and discharge.

No amount of fiddling seems to persuade it to charge at 22.30 tonight which is the cheapest slot for the foreseeable...
[image]

Ps @gcoan who is giving you those export rates?!

gcoan, 2024-01-04:
yes, is strange @naltsta why its not charging in that slot when mine is charging OK

here's my config if you can see anything relevant:
[image]
[image]
[image]

Export wise I'm on Octopus flux which is why you see the rates go up and down during the day/night

gcoan, 2024-01-04:
> I've a mixergy hot water tank which can charge on electric on cheap agile rates and gas when it's expensive. I'm thinking maybe some of my data is off because of this. Maybe I should put a request in to treat this hot water tank as a car charger.

Can you feed the mixergy electric charging data in as load_forecast in apps.yaml and it should automatically get excluded:
```
 # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
```

dandwhelan, 2024-01-04:
Ok, I'll check it out.

Will I enter something like this?

load_forecast:
    - predheat.heat_energy$sensor.namehere

naltsta, 2024-01-04:
> yes, is strange @naltsta why its not charging in that slot when mine is charging OK
> 
> 
> 
> here's my config if you can see anything relevant:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> 
> 
> Export wise I'm on Octopus flux which is why you see the rates go up and down during the day/night
> 
> 
> 
> 

Thanks - I've been tweaking and adjusting to see what I can get it to do but generally very similar settings.

I now have no charge until 09.30 on Saturday morning!

Aah so you're pretend on agile but actually on flux? 

Thanks for investigating

dandwhelan, 2024-01-04:
How did you get the Watt loss on the price tab?

naltsta, 2024-01-04:
Plan debug I think it's called

Managed to get a better plan but still doesn't seem really optimised[image]

Charging at midnight seems bizarre as it's really not that cheap!

gcoan, 2024-01-04:
> Ok, I'll check it out. Will I enter something like this? load_forecast: - predheat.heat_energy$sensor.namehere

Hi @dandwhelan 

Its unfortunately not at all well documented, the only description of how this works is in the template apps.yaml file (there's nothing [currently] in the documentation).

Looking at the python code that loads this extra load:
```
def fetch_extra_load_forecast(self, now_utc):
        """
        Fetch extra load forecast
        """
        load_forecast = {}
        if "load_forecast" in self.args:
            entity_ids = self.get_arg("load_forecast", indirect=False)
            if isinstance(entity_ids, str):
                entity_ids = [entity_ids]

            for entity_id in entity_ids:
                attribute = None
                if "$" in entity_id:
                    entity_id, attribute = entity_id.split("$")
                try:
                    data = self.get_state(entity_id=entity_id, attribute=attribute)
                except (ValueError, TypeError):
                    data = None

                load_forecast = self.minute_data(
                    data, self.forecast_days, self.midnight_utc, "energy", "last_updated", backwards=False, clean_increment=False, smoothing=True, divide_by=1.0, scale=1.0
                )

        return load_forecast
```

I take this to read

load_forecast is set in apps.yaml to the string "entityname$attribute name"

and you need a sensor called "entityname" which is of type energy, unit of measurement kWh

e.g. sensor.mixergy_consumption_today_kwh

and if the consumption is in an attribute called 'import' the apps.yaml entry would be:

```
load_forecast
  - sensor.mixergy_consumption_today_kwh$import
 ```

If there isn't a separate attribute to read and the data you want is just the main sensor value, I am assuming you put the string as:

sensor.mixergy_consumption_today_kwh$

But I am no expert on reading someone else's Python code.....

Answering your other Q,  yes, to see the effective rate for import and export after conversion losses, you set:
[image]

@naltsta and answering your Q about tariff.  I was on Octopus Flux in the summer, (import and export tariffs) then requested a move to Octopus Agile in mid October.   Octopus moved my import tariff over to Agile.
Hence why I am on what I am on.....

gcoan, 2024-01-04:
> Charging at midnight seems bizarre as it's really not that cheap!

@naltsta agreed, doesn't seem any real reason to charge at midnight at 18.57p, but  let the battery discharge at 03:00 when the rate is 17.63p.

Unless best_soc_keep or best_soc_min is causing predbat to want to retain your charge?  As you can see I have set mine to 0.1 and 0 to let predbat manage the battery fully and not keep anything in reserve. I only have a 5.2 battery so there's no point

JonathanLew1s, 2024-01-05:
I would agree in setting best keep to 0 and combine charge slots to off. I believe these two are causing predbat to see two continuous slots to perform the charge, ignoring cheaper slots to find one combined that satisfies. 

Again no expert but from my observations.

gcoan, 2024-01-05:
Agree about setting min to zero now otherwise predbat can be forced into charging the battery when it's expensive to do so.

Combine Charge slots, charge on discharge and charge only vs charge no discharge, for all of these I don't think it's clear cut.

Here's my plan today for combine charge slots=true
[image]

And if I change combine charge slots to false, predbat does more charging!
[image]

I think it's worth playing with these settings if you don't like the plan predbat produces

dandwhelan, 2024-01-10:
> Hi @dandwhelan
>
> Its unfortunately not at all well documented, the only description of how this works is in the template apps.yaml file (there's nothing [currently] in the documentation).
>
> Looking at [...]

Thanks for the detailed reply. I forgot to post but I tried that formatted and it caused a system error.
I ended up replacing it with... 

load_forecast:
    - predheat.heat_energy$sensor.mixergy_electric_heat_energy_kwh

I've no way to see if it's working but it never crashed.

gcoan, 2024-01-10:
> load_forecast:
> - predheat.heat_energy$sensor.mixergy_electric_heat_energy_kwh
> 
> 

If it's loading extra load info then you should see something in the logfile.

Can you attach a logfile please?

But I'm not convinced this is working as I think it'll be looking for an entity 'predheat.heat_energy' which it won't find.  But I may be wrong!

I have separately asked @springfall2008 for advice / documentation on how this is supposed to be configured so I can add it to the documentation

springfall2008, 2024-01-10:
The name should be the sensor name, the bit after the dollar is the attribute name. Clearly the above won’t work

gcoan, 2024-01-10:
That’s what I thought Trefor.

What if there’s no attribute, it’s just the default sensor value?

springfall2008, 2024-01-10:
You can’t put future data into a sensor value afaik. The data has to be an array of dates and values of a specific format.

springfall2008, 2024-01-10:
Like this 

[image]

gcoan, 2024-01-10:
Ah OK @springfall2008 it doesn't sound like this will do what @dandwhelan wanted which was:

> I've a mixergy hot water tank which can charge on electric on cheap agile rates and gas when it's expensive. I'm thinking maybe some of my data is off because of this. Maybe I should put a request in to treat this hot water tank as a car charger.

I had suggested:
> @dandwhelan Can you feed the mixergy electric charging data in as load_forecast in apps.yaml and it should automatically get excluded:
```
 # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
```

But now I think I better understand that load_forecast is to ADD an additional forecasted load into the predbat plan, NOT to remove load, like historical EV charging

Looks like @dandwhelan could get what he wants by configuring car_charging_energy:

> car_charging_energy - Set in apps.yaml to point to a Home Assistant entity which is the incrementing kWh data for the car charger. This has been pre-defined to a regular expression to auto-detect the appropriate Wallbox and Zappi car charger sensors, or edit as necessary in apps.yaml for your charger sensor.

if he set:
```
car_charging_energy:
  - sensor.mixergy_electric_heat_energy_kwh
```

then this would have the desired effect of excluding the mixergy load from the historic house load?

As for load_forecast, I can add some more to apps.yaml to explain how this works now I understand it better

springfall2008, 2024-01-10:
Yes I agree with your comments, car charging energy is for excluding while load forecast is for including extra load in the future, mostly to link to predheat

gcoan, 2024-01-10:
Been looking at [Predheat](https://github.com/springfall2008/predheat), I can see the family resemblance!

Have added to the apps.yaml documentation to explain how car_charging_energy can be used to exclude load data such as Mixergy electric heating,  and how load_forecast works now I (hopefully) understand it.

Draft updated doc here: https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md

closed 2024-02-03 by springfall2008