springfall2008/batpred #575: Update mechanism - allow ad-hoc updates via the HA update framework

DJBenson, 2024-01-03:
**Is your feature request related to a problem? Please describe.**

I've recently migrated to the AppDaemon+Predbat setup and all is working well. The update mechanism seems a little disjointed from the framework which Home Assistant provides in that I get a notification that there's a new update (update.predbat goes true) but there are no controls to perform the update.

**Describe the solution you'd like**
Allow the update.predbat entity to call the update service and get/install the version for which the notification was triggered (or simply the most recent version).

**Describe alternatives you've considered**
Updates work as per the Predbat docs but as per the intro seem out of kilter with the way other addons use the HA update framework.

**Additional context**
N/A

DJBenson, 2024-02-04:
Little bump on this. Not sure if it's possible but the update notifications are not functional currently.

springfall2008, 2024-02-04:
I did look at this a little but I couldn’t figure out if it was possible, I was trying to get an update event into Appdaemon but failed.

gcoan, 2024-02-04:
I have noticed a different issue with updates that if you're using the standalone app daemon and then predbat via hacs, when you update predbat itself with its own update mechanism (or auto update), hacs doesn't know about the fact that predbat is updated 

[image]

BTW @DJBenson if you turn auto update on in predbat it will update itself and you won't need to manually use the update mechanism

springfall2008, 2024-02-04:
Wow, I managed to make this work..... on a roll today..... another 5 bugs tomorow!

https://github.com/springfall2008/batpred/releases/tag/v7.15.15

DJBenson, 2024-02-04:
> I have noticed a different issue with updates that if you're using the standalone app daemon and then predbat via hacs, when you update predbat itself with its own update mechanism (or auto update), [...]

Yeah I could but I like to check out what's available before proceeding. It's a lazy request really (or if you like to look at it this way, it's making the Predbat updates work as intended when using the update framework 😎).

DJBenson, 2024-02-04:
> Wow, I managed to make this work..... on a roll today..... another 5 bugs tomorow!
> 
> 
> 
> https://github.com/springfall2008/batpred/releases/tag/v7.15.15

Legend!! Great work @springfall2008.

gcoan, 2024-02-04:
It's better to use the right ha mechanisms.

I've now got to the point of comfort where I've set predbat to auto update. Still read the release notes of course

DJBenson, 2024-02-05:
Not sure if it's related but Predbat has updated itself automatically twice despite auto updates being disabled when HA is restarted.

springfall2008, 2024-02-05:
Yeh it's a bug I've fixed in the latest release

closed 2024-02-07 by DJBenson