springfall2008/batpred #576: Has nord data future prices stopped working?

gcoan, 2024-01-04:
**Describe the bug**
I noticed yesterday that the future agile prices for Wednesday were exactly the same as the prices were on Tuesday.

Looking today, Thursdays prices (tomorrow) are the same as today's prices. 

In the logfile the nord data appears to be working ok.

**Expected behavior**
Tomorrow's prices different from today - not necessarily big differences, but not every half hour price the same 

**Predbat version**
7.14.31

**Environment details**
Haos

**Screenshots**
Tuesday plan:
[image]

Wednesday future plan (taken on Tuesday):
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-01-04:
It’s probably as you have export adjust on but it should be import for you?

gcoan, 2024-01-04:
Doh !

That's better 
[image]

I had just copied what was in the template apps.yaml across which had import=False, export=True and hadn't really paid it any attention.  
The explanation of the norddata config in energy-rates has True, True, but doesn't explain what the fields do.
I'll change apps.yaml to import=True, export=False as that will be what most people have and add a fuller explanation of the config to energy-rates.md.
Will include in the next PR I do.

Thanks

closed 2024-01-04 by gcoan