springfall2008/batpred #577: Pulling from Grid unnessarily

b1nuzz, 2024-01-04:
**Describe the bug**
Pulling from the grid when everything points to battery usage would be OK. 

PredBat up to date

[appdaemon.log](https://github.com/springfall2008/batpred/files/13830518/appdaemon.log)
[image]
[image]
[image]
[image]

springfall2008, 2024-01-04:
I think there are two things here:

1. Your timeouts in the log are probably as you have the run loop time for GivTCP set too fast, can try changing it to maybe 20 seconds in your GivTCP settings?

2. The plan above looks okay but probably not idea due to round trip losses, if you try to turn off 'combine charge' then I think it will get better.

springfall2008, 2024-01-04:
Can you also try this release: 

https://github.com/springfall2008/batpred/releases/tag/v7.14.32

And make sure set_reserve_enable is true

b1nuzz, 2024-01-04:
Thanks. 
I have actioned all of the above. 
Immediately the battery prediction looks more like it used to, with overnight charges and daily useage. 

Will run like this for 24/48 hours and report back.

springfall2008, 2024-01-09:
Please reopen if you have a new issue

closed 2024-01-09 by springfall2008