springfall2008/batpred #583: Documentation Query

Bcrutt, 2024-01-05:
Hi
I noticed this section titled "None" in the documentation, screenshot below

[image]

[image]

gcoan, 2024-01-05:
That was my fault.

As part of the recent documentation changes in #570 7.14.31, I renamed config-yaml-settings.md to apps-yaml.md to reflect the correct file name.

Changed links from the install guide etc, but I missed editing the menu structure that referred to it.

Saw the problem myself this morning and included a fix in PR #581 (along with a few more documentation improvements).
It's waiting for @springfall2008 to review and approve 

Cheers Geoffrey

gcoan, 2024-01-06:
Fixed in 7.14.33 
[image]

closed 2024-01-06 by gcoan