springfall2008/batpred #586: Unusual Charge Hold/Charging when PV is generating

AGLAISTER, 2024-01-06:
**Describe the bug**
1) Charge hold @ battery 100% causing grid import when load was called - Around 1pm in log
2) Charging from grid while PV generating, battery was at 97/98% with still 2 hours of PV generation available. When load was turned off the plan went to charging and immediately started charging from PV and grid (occurs around 1:30pm in log)

**Expected behavior**
1) No charge hold when PV is being generated, providing more than household load - battery should be idle ready to provide to house if required
2) No grid charge when there is PV generation present and still more PV available 

**Predbat version**
1) 7.14.31
2) 7.14.32

[appdaemon.log](https://github.com/springfall2008/batpred/files/13851311/appdaemon.log)

springfall2008, 2024-01-06:
Thanks for the info, have you tried .33 yet with a potential fix for this?

AGLAISTER, 2024-01-06:
Hi Trefor,

My mistake in the original ticket - the 2nd issue was on 7.14.33 not .32

springfall2008, 2024-01-07:
Can you try this release please?

https://github.com/springfall2008/batpred/releases/tag/v7.14.34

springfall2008, 2024-08-06:
Old ticket, closing for now. Feel free to open a new ticket on the latest version if still an issue

closed 2024-08-06 by springfall2008