springfall2008/batpred #590: Beta test for calculate regions feature

springfall2008, 2024-01-07:
I've added a new feature on 'main' called Calculate Regions optimisation (which is off by default).

For me it seems to improve the Agile optimisation quite a bit, please can someone else test it?

naltsta, 2024-01-07:
Happy to give it a go! Do I need autoupdate off if I'm installing this manually?

What do regions do?..

springfall2008, 2024-01-07:
Turn off auto update and select main from the version list.

regions is just an extra optimisation step to improve the plan

naltsta, 2024-01-07:
Certainly looks to be working fine but no obvious changes to my plan

springfall2008, 2024-01-07:
So turning it on and off has no impact

naltsta, 2024-01-07:
Hadn't actually given it a chance to run! Sorry

mpartington, 2024-01-07:
I can't test unfortunately as not on agile, but if it's successful, would it be worth consolidating the different plan tweaks currently available?

For example, this could be renamed 'agile optimisation' (or similar, so it's really obvious what it does), then maybe remove the others.

gcoan, 2024-01-07:
calculate region optimisation = False:
[image]

calculate region optimisation = True
[image]

Predbat has coloured a lot of the import slots a different colour, and whilst the charge periods & target soc's look the same, the resultant plan is 9p cheaper y 8am tomorrow.

Just for comparison, here's 7.14.33 that I was on previously, which is half way between the two.  But some of the load figures are different that main & calculate region optimisation=true
[image]

And might as well, 7.14.34:
[image]

and back to main with calculate region optimisation=True, its cheaper again:
[image]

but as it ticked over to 15:30, the cost to 08:00 tomorrow morning has jumped up again by 20p:
[image]

@springfall2008 its hard to see whether this latest tweak option is having any benefit or not. The load figures seem to be slightly different each run and the cost changed when it goes over a half hour boundary.  Other that more is coloured red rather than green, the actual plan looks the same

gcoan, 2024-01-07:
Next run, the cost to tomorrow 8am has gone up again.  It can't necessarily be that we have got load on now and that's changing the figures, as the cost to 16:00 today is remaining constant at £4.30, but the projected cost into tomorrow is changing
[image]

Here's the logfile if its any use
[appdaemon (13).log](https://github.com/springfall2008/batpred/files/13853996/appdaemon.13.log)

naltsta, 2024-01-07:
At 04.30 on both plans I am 100% SOC with calculate regions 20p cheaper

springfall2008, 2024-01-07:
I found a bug that would cause the old colouring and suchlike and pushed a fix to main, so maybe update and try again?

gcoan, 2024-01-07:
latest main version. region optimisation=True
[image]

region optimisation=False
[image]

20p cheaper with region optimisation False :-(

as it ticked round 16:30, the price to 8:00 jumped up again by 14p:
(still region opt=false)
[image]

setting region optimisation to True again, its 25p cheaper:
[image]

Whilst I was looking at the different plans, trying to see what was different (not a lot except the second run with region optimisation=True didn't have a discharge slot), Predbat has made another run and the price to 8am has gone back up again:
[image]

So in summary, the latest main isn;t colouring things red as much.
Financially it looks like the region optimisation makes a small positive financial contribution but its hard to be absolutely clear on this as the price to a fixed point in time keeps changing on subsequent predbat runs anyway

naltsta, 2024-01-07:
Seems to be working and my plan is updating every 5 minutes but getting a lot of this

Changed to ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value

gcoan, 2024-01-07:
> Seems to be working and my plan is updating every 5 minutes but getting a lot of this
> 
> Changed to ERROR: Exception raised cannot access local variable 'charge_start_time' where it is not associated with a value

not getting that on mine.  Sounds like its something to do with EV car charging?

springfall2008, 2024-01-07:
I've released a fix for the charge_start_time issue, but you can also resolve it by setting that dropdown menu to a valid time

springfall2008, 2024-01-07:
The latest release has this feature enabled by default, I don't think it can do any harm, will see if it improves the plan.

BTW: Setting metric_battery_value_scaling to a lower number e.g 0.5 reduces the amount of charging you get

naltsta, 2024-01-07:
> I've released a fix for the charge_start_time issue, but you can also resolve it by setting that dropdown menu to a valid time

Honestly think I have bigger issues... GivTCP now seems very unhappy (but does seem to be working still, as does Predbat) - this could also be entirely unrelated

2024-01-07 19:50:15.103270 INFO pred_bat: Predbat /homeassistant/appdaemon/apps/batpred/predbat.py version v7.14.35 currently running, latest version is v7.14.35 latest beta v7.14.35
2024-01-07 19:50:15.109505 INFO pred_bat: Autoupdate: Currently up to date
2024-01-07 19:50:15.176756 INFO pred_bat: Inverter type GE forecast_hours 36 max_windows 72 num_cars 0 debug enable is False calculate_plan_every 5
2024-01-07 19:50:15.187340 INFO pred_bat: Predbat mode is set to Control charge & discharge
2024-01-07 19:50:17.534411 INFO pred_bat: Found 11521 load_today datapoints going back 8 days
2024-01-07 19:50:18.711786 INFO pred_bat: Car charging hold False threshold 6.0
2024-01-07 19:50:18.719282 INFO pred_bat: Current data so far today: load 11.52 kWh import 9.3 kWh export 4.6 kWh pv 6.9 kWh
2024-01-07 19:50:18.727530 INFO pred_bat: Fetching futurerate data from https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP
2024-01-07 19:50:18.736263 INFO pred_bat: Return cached futurerate data for https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP age 33.4 minutes
2024-01-07 19:50:18.744390 INFO pred_bat: Future rates - peak rate is 960 - 1140 minutes premium import 14 export 6
2024-01-07 19:50:18.904534 INFO pred_bat: Loaded 192 datapoints of futurerate analysis
2024-01-07 19:50:19.022924 INFO pred_bat: Predicted future rates: ['1190 => 20.46 / 8.41', '1250 => 19.58 / 8.05', '1310 => 18.36 / 7.55', '1490 => 17.61 / 7.24', '1550 => 16.99 / 6.99', '1610 => 16.4 / 6.74', '1670 => 16.18 / 6.65', '1730 => 16.87 / 6.94', '1790 => 19.25 / 7.92', '1850 => 20.35 / 8.37', '1910 => 22.71 / 9.34', '1970 => 23.75 / 9.77', '2030 => 22.72 / 9.34', '2090 => 20.92 / 8.6', '2150 => 20.23 / 8.32', '2210 => 19.7 / 8.1', '2270 => 20.43 / 8.4', '2330 => 21.79 / 8.96', '2390 => 23.59 / 9.7', '2450 => 40.67 / 16.68', '2510 => 40.98 / 16.81', '2570 => 39.17 / 16.07', '2630 => 22.13 / 9.1', '2690 => 19.89 / 8.18', '2750 => 19.89 / 8.18']
2024-01-07 19:50:19.153434 INFO pred_bat: Standing charge is set to 56.6265 p
2024-01-07 19:50:21.430040 INFO pred_bat: Rate min forward looking: now 17.094 at end of forecast 17.09
2024-01-07 19:50:21.442036 INFO pred_bat: Import rates min 17.09 max 40.53 average 21.65
2024-01-07 19:50:21.502328 INFO pred_bat: Export rates min 15.0 max 15.0 average 15.0
2024-01-07 19:50:21.516326 INFO pred_bat: Rate thresholds (for charge/discharge) are import 40.03p (0.0) export 17.09p (0.0)
2024-01-07 19:50:21.535076 INFO pred_bat: High export rate found rates in range 99 to -99
2024-01-07 19:50:21.584632 INFO pred_bat: Low Import rate found rates in range 17.09 to 37.63
2024-01-07 19:50:21.783991 INFO pred_bat: Todays energy import 9.3 kWh export 4.6 kWh cost 181.13 p import 250.13 p export -69.0 p
2024-01-07 19:50:21.809916 INFO pred_bat: PV Data for pv_forecast_today total 6.03 kWh
2024-01-07 19:50:21.836457 INFO pred_bat: PV Data for pv_forecast_tomorrow total 4.73 kWh
2024-01-07 19:50:21.865568 INFO pred_bat: PV Data for pv_forecast_d3 total 11.34 kWh
2024-01-07 19:50:21.889262 INFO pred_bat: PV Data for pv_forecast_d4 total 11.23 kWh
2024-01-07 19:50:22.026552 INFO pred_bat: Historical data totals for days [1, 2, 3, 4, 5, 6, 7] are [13.9, 6.79, 11.03, 10.41, 12.69, 11.31, 11.82] - min 6.79
2024-01-07 19:50:22.039820 INFO pred_bat: Model filter enabled - Discarding day 2 as it is the lowest of the 7 datapoints
2024-01-07 19:50:22.054272 INFO pred_bat: WARN: Historical day 1 has 10 minutes of gap in the data, filled from 13.9 kWh to make new average 14.0 kWh (percent 99%)
2024-01-07 19:50:22.072924 INFO pred_bat: WARN: Historical day 6 has 15 minutes of gap in the data, filled from 11.31 kWh to make new average 11.43 kWh (percent 99%)
2024-01-07 19:50:22.124362 INFO pred_bat: Historical days now [1, 3, 4, 5, 6, 7] weight [1, 1, 1, 1, 1, 4]
2024-01-07 19:50:22.298404 INFO pred_bat: Today's load divergence 132.48 % in-day adjustment 130.86 % damping 0.95x, Predicted so far 5.98 kWh with 0.0 kWh car excluded and 4.59 kWh import ignored and 0.0 forecast extra, Actual so far 9.32 kWh with 0.0 kWh car excluded and 2.24 kWh import ignored
2024-01-07 19:50:22.522955 INFO pred_bat: Inverter 0 using Rest API http://homeassistant.local:6345
2024-01-07 19:50:22.547578 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/readData - REST will be disabled
2024-01-07 19:50:22.598035 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-01-07 19:50:22.609776 INFO pred_bat: New Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 2600.0 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 3.6 kW export limit 3.6 kW reserve 4.0 % current_reserve 4.0 %
2024-01-07 19:50:22.630217 INFO pred_bat: WARN: Inverter 0 read bad REST data from http://homeassistant.local:6345/readData - REST will be disabled
2024-01-07 19:50:22.679059 INFO pred_bat: Inverter 0 SOC: 0.0 kW 0 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 0.0 w Current voltage 52.0
2024-01-07 19:50:22.689330 INFO pred_bat: WARN: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
[... cut ...]
2024-01-07 19:45:58,981 - Inv1 - write       -  [INFO    ] - Setting battery discharge limit 0 was a success
2024-01-07 19:46:02,161 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.238, 8899) failed: timed out
[2024-01-07 19:46:04 +0000] [100] [CRITICAL] WORKER TIMEOUT (pid:103)
[2024-01-07 19:46:05 +0000] [100] [ERROR] Worker (pid:103) was sent SIGKILL! Perhaps out of memory?
[2024-01-07 19:46:05 +0000] [237] [INFO] Booting worker with pid: 237
2024-01-07 19:46:05,611 - Inv1 - write       -  [INFO    ] - Setting battery discharge rate to: 0 (0)

gcoan, 2024-01-07:
I had exactly this happen last week. I restarted givtcp and it fixed the comms issue.

BTW there’s an automation in the predbat faq’s I wrote to detect givtcp issues

mpartington, 2024-01-08:
Hi Trefor, there are now 4 options that seem to have a strong overlap in function (I realise each will be slightly different ), however is there any scope.to combine / retire any of these? Feels like expert mode is getting busy again after the previous purge

[image]

springfall2008, 2024-01-09:
@mpartington good point, i'll have a think.

springfall2008, 2024-01-10:
This is now released

closed 2024-01-10 by springfall2008