springfall2008/batpred #595: Short notice Freeze Charge & Low Load

robyholmes, 2024-01-08:
**Describe the bug**
First raised on Facebook post: https://www.facebook.com/groups/615579009972782/permalink/896274541903226/
Predbat is reporting low load figures, and also choosing to freeze charge a lot, but not always planning it, it tends to choose to freeze charge just as the next slot starts.

**Expected behavior**
I would expect the load to be a little higher, our base load is around 350 to 400w. During the day it should be a little higher with PC running etc. However I do expect the load history will be a little lower due to New Year visiting family etc.

**Predbat version**
V7.14.25

**Environment details**
Dell Thin Client
HAOS
GivEnergy 5kW Gen 3 Hybrid Inverter with 2x 9.5kWh Batteries
[appdaemon.log](https://github.com/springfall2008/batpred/files/13858752/appdaemon.log)
[image]

robyholmes, 2024-01-08:
[apps.yaml.txt](https://github.com/springfall2008/batpred/files/13858774/apps.yaml.txt)
Forgot to attach this for you.

robyholmes, 2024-01-08:
We have been changing the day pervious, mainly has we didn't understand it at first, thinking that say 7 days meant it looked at the last 7 days (So all 7 days). We tried 1,2,3 last night to try and avoid new years which wasn't a 'normal' day of course.
I've also tried 1 & 7 today but although the values change, the behaviour doesn't seem to.

robyholmes, 2024-01-08:
I have changed metric_battery_cycle from 2 (Which it seemed to be at from new? Default is shown as 0 though) to 0. That seems to have improved things but early days, so I'll report back likely end of today with the result of this change.

robyholmes, 2024-01-08:
Update on what it's being doing this morning. Pretty much the same. Each half hour comes it freeze charges. It tends to do this a couple of minutes (I want to say 5 but might be wrong) into the new slot. As if it's re-run the plan and changed from having an idle this half hour to freeze charge.

robyholmes, 2024-01-08:
[image]
Log book of the status shows it well. I was off with my 5 minutes mind.

naltsta, 2024-01-08:
Are you actually on 7.14.25 - I'm on 7.14.35 but seeing very similar this evening

[image]

[image]

springfall2008, 2024-01-08:
I've had a go at a fix, please can you test it? https://github.com/springfall2008/batpred/releases/tag/v7.14.36

robyholmes, 2024-01-08:
Updated, I'll report back tomorrow with how it's looking.

robyholmes, 2024-01-10:
I can happily report Trevor that the Freeze Charge issue appears to be resolved. It's working far more like I'd expect now, using a lot more of the battery and picking the right slots.

Thank you very much for looking at this so quickly! Predbat is a very good tool.