springfall2008/batpred #598: WARNING AppDaemon: Queue size for thread thread-1 is 58, callback is 'update_time_loop() in pred_bat' - possible thread starvation

stefan-west1, 2024-01-09:
**Describe the bug**
I'm getting this error a lot. I've also noticed that when I try to change a setting in Home Assistant, it will initially switch back and then a couple minutes later will finally change to what I set.

**Predbat version**

7.14.36

**Environment details**

Running Appdaemon and Home Assistant in Docker Containers in Unraid
System is running 2x E5-2680 V4, with 128GB DDR4. 

Appdaemon.yaml settings:
- thread_duration_warning_threshold: 30
- threadpool: 50

**Log file**
2024-01-09 01:20:14.112038 WARNING AppDaemon: Queue size for thread thread-1 is 52, callback is 'update_time_loop() in pred_bat' called at 2024-01-09 01:19:39+00:00 - possible thread starvation
2024-01-09 01:20:14.112505 INFO AppDaemon: --------------------------------------------------
2024-01-09 01:20:14.112745 INFO AppDaemon: Threads
2024-01-09 01:20:14.112992 INFO AppDaemon: --------------------------------------------------
2024-01-09 01:20:14.113744 INFO AppDaemon: Currently busy threads: 1
2024-01-09 01:20:14.114067 INFO AppDaemon: Most used threads: 1 at 2024-01-09 01:06:52+00:00
2024-01-09 01:20:14.114391 INFO AppDaemon: Last activity: 2024-01-09T01:06:52+00:00
2024-01-09 01:20:14.114626 INFO AppDaemon: Total Q Entries: 52
2024-01-09 01:20:14.114884 INFO AppDaemon: --------------------------------------------------
2024-01-09 01:20:14.115284 INFO AppDaemon: thread-0 - qsize: 0 | current callback: idle | since never, | alive: True, | pinned apps: ['hello_world']

springfall2008, 2024-01-09:
It seems your machine is overloaded, an you look at the logfile (appdaemon.log or predbat.log) and see how long its taking to run.

Please have a look at: https://springfall2008.github.io/batpred/customisation/#performance-related and compare against your settings.

springfall2008, 2024-01-09:
Have a look at the logs and see how long an update cycle takes.

stefan-west1, 2024-01-09:
I think I’m looking at the right thing. It seems to vary between 30-180 seconds

springfall2008, 2024-01-20:
Do you still have the issue?

springfall2008, 2024-02-04:
I've done some optimisations and also improved the responsiveness of the controls, please re-test and file a new ticket if need be

closed 2024-02-04 by springfall2008