springfall2008/batpred #600: Cycling discharge/charge during charge window

7Van737, 2024-01-09:
**Describe the bug**
I’m on Eco 7 with a 00:30 to 07:30 cheap rate charge window.  I have the ‘Set Charge Low Power Mode’ enabled which is meant to set the charge to make full use of the available charge window.  Generally, predbat fully charges the battery (to 99%) at around 05:30 – and then for the remainder of the cheap rate window it starts cycling between discharging the battery to cover the house load and recharging.  

**Expected behavior**
During the cheap rate charge window all house load should be supplied from the grid and the battery not discharged

**Predbat version**

7.14.36

**Environment details**

AIO
Standard HAOS 

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]
[appdaemonLog.txt](https://github.com/springfall2008/batpred/files/13876586/appdaemonLog.txt)

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

7Van737, 2024-01-09:
Here are two screenshots of the predbat plan for this mornings charge

[image]

[image]

springfall2008, 2024-02-03:
This should be fixed in the latest release, it related to 99% charge levels on inverters where the reserve can't be 100%. Please re-test and file a new issue if you see it again.

closed 2024-02-03 by springfall2008