springfall2008/batpred #602: Load and scaling factors should be ignored when in holiday mode?

gcoan, 2024-01-09:
**Describe the bug**
Follows from a conversation on the GivEnergy forum 

Piansom is away on holiday and so his consumption is fairly flat.  Predbat set to holiday mode.
Every night excess battery charge is being exported prior to the overnight cheap rate period, but Predbat seems to be getting this wrong and is discharging too much, resulting in import occurring before the next overnight cheap period.

I've advised setting best_soc_keep and best_soc_min, but still not getting the right behaviour.

Looking through the settings and comparing to my own, and looking at the predbat defaults, I am wondering whether the behaviour of  metric_inday_adjust_damping and load_scaling should be changed when in holiday mode?

metric_inday_adjust_damping is set to 0.95
load_scaling to 1.15

Holiday mode is on so days_previous is ignored and just yesterday's consumption is being used.

But load_scaling is increasing that consumption by 15% and metric_inday_adjust_damping is reducing it by 5%, resulting in today's consumption not being assumed to be the same as yesterdays.

In holiday mode should these two factors be ignored (i.e. no scaling), or should there be advice given in the documentation to review and change these factors when setting holiday mode?

**Expected behavior**
In holiday mode, today's consumption should be modelled closely on yesterday's consumption.

**Predbat version**
7.14.36

PianSom, 2024-01-10:
One thing that is possibly relevant to note - following a suggestion from Geoffrey I have the following in my config:
```
  days_previous:
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8

  days_previous_weight:
    - 1
```

springfall2008, 2026-10-29:
Closing as stale. Holiday mode has been rewritten since this was raised, so the question about `load_scaling` and `metric_inday_adjust_damping` no longer applies in the same way. Predbat now matches holiday days in your load history and scales the predicted load by `input_number.predbat_holiday_load_scaling` (default 0.7). See [Holiday mode](https://springfall2008.github.io/batpred/customisation/) in the customisation docs.

If holiday-mode predictions still look wrong on a current release, please open a new issue with a `predbat_debug.yaml` attached.

closed 2026-10-29 by springfall2008