springfall2008/batpred #605: Very varied plans - changing every run

naltsta, 2024-01-10:
**Describe the bug**
Pretty drastic plan changes every 5 minutes, some of which are notably more expensive than others. Plan changing during slots so very spiky charge profile and over charged this morning compared to best plans I saw last night.

**Expected behavior**
Pick cheapest plan

**Predbat version**
7.14.37

**Environment details**
Gen 1 givenergy via GivTCP and 2x2.6kWh
- Standard HAOS installer
- Anything else?

**Screenshots**
[image]
**Log file**

springfall2008, 2024-01-10:
Can you provide examples of the different predbat plans and maybe a logged? Thanks

naltsta, 2024-01-10:
[image]

[image]

Charging plan for tonight - Not as drastic differences as last night. Plenty of PV forecast so charging less in cheaper

naltsta, 2024-01-10:
I'm still not really sure I totally understand the calibration chart but it doesn't normally look like this

[image]

springfall2008, 2024-01-11:
I'm not sure why the issue happens, but it's been exposed in your system as you have the plan re-calculation time set to 5 minutes rather than the default 10 minutes.

I will still have a look and see if I can find out the root cause, but if you set this time back to 10 it will probably go away

closed 2024-09-01 by springfall2008