springfall2008/batpred #608: Charge window for tomorrow affecting today

wilsondw, 2024-01-11:
**Describe the bug**
At approx 04:50, after charging to target SOC during cheap rate period, predbat switched to hold charging 82% and then to idle leaving the charge target at 82%. At the same time Predbat calculated and set a charge window (but did not set the charge target) for the next day extending to 8am (changed to 7am in a later run). As the charge slots are based on time only (not date), this inadvertently led to the inverter holding charge at 82% through to 7am when the plan was reporting idle. The future slot set by predbat appears as 'charge limit base' in the predbat plan with the 82% target from today's hold charging and at 100% target calculated by the plan as 'charge limit best'.

2024-01-11 04:45:11.968080 INFO pred_bat: Not setting charging SOC as we are not within the window (now 01-11 04:45:00 target set_soc_minutes 30 charge start time 01-11 04:00:00)
2024-01-11 04:45:11.972555 INFO pred_bat: Adjust reserve to target charge 82 % (set_reserve_enable is true)
2024-01-11 04:45:11.979126 INFO pred_bat: Inverter 0 Current reserve is 83.0 already at target
2024-01-11 04:45:12.025994 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-01-11 04:45:12.029619 INFO pred_bat: Completed run status Hold charging

2024-01-11 04:51:56.677774 INFO pred_bat: Charge window will be: 2024-01-12 02:00:00+00:00 - 2024-01-12 08:00:00+00:00 - current soc 82 target 100
2024-01-11 04:51:56.681767 INFO pred_bat: Configuring charge window now (now 01-11 04:50:00 target set_window_minutes 30 charge start time 01-12 02:00:00
2024-01-11 04:52:08.785443 INFO pred_bat: Inverter 0 set charge slot 1 {'start': '02:00', 'finish': '08:00'} via REST successful after retry 0
2024-01-11 04:52:08.856889 INFO pred_bat: Inverter 0 Updated start and end charge window to 02:00:00 - 08:00:00 (old 03:00:00 - 05:00:00)
2024-01-11 04:52:24.629364 INFO pred_bat: Set inverter 0 charge schedule True via REST successful on retry 0
2024-01-11 04:52:24.673358 INFO pred_bat: Inverter 0 Turning on scheduled charge
2024-01-11 04:52:24.688547 INFO pred_bat: Setting ECO mode as no discharge window planned
2024-01-11 04:52:24.700024 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to None - None
2024-01-11 04:52:24.709876 INFO pred_bat: Inverter 0 current discharge rate is 0 and new target is 3600
2024-01-11 04:52:34.138969 INFO pred_bat: Inverter 0 set discharge rate 3600 via REST successful on retry 0
2024-01-11 04:52:34.171466 INFO pred_bat: Not setting charging SOC as we are not within the window (now 01-11 04:50:00 target set_soc_minutes 30 charge start time 01-12 02:00:00)
2024-01-11 04:52:34.176292 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2024-01-11 04:52:34.181505 INFO pred_bat: Inverter 0 Current Reserve is 83.0 % and new target is 4.0 %
2024-01-11 04:52:50.194960 INFO pred_bat: Set inverter 0 reserve 4 via REST successful on retry 0
2024-01-11 04:52:50.273540 INFO pred_bat: Completed run status Idle

[image]

**Expected behavior**
Predat should probably have cleared the charge target when it switched from hold charging to idle and should probably not set future charge / discharge windows extending beyond 24hours in the future

**Predbat version**

7.14.36

**Environment details**

- Gen 2 with 9.5kWh battery
- Standard HAOS

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/13908815/appdaemon.log)

Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-01-12:
This does look like a real bug, I'll take a look

springfall2008, 2024-01-13:
Fixed in latest release

closed 2024-01-21 by springfall2008