springfall2008/batpred #611: Holding charge when limited variation in Agile rates

Boffinboy, 2024-01-13:
**Describe the bug**
Recently, I am seeing plans with long periods of hold or charge, even though using up charge would generate savings and allow recharge at lower rates (even with conversion losses). See attached images. I have combine charge disabled, combine discharge disabled. I have tried reducing the battery loss charge / discharge, and the inverter loss, but looking at the rates w/loss on the plan it should be allowing battery use. One thing that appears to force a change in behaviour is changing rate low threshold, moving it to 0.9 looks closer to what I’d expect and differentiates between slots. 

**Expected behavior**
Rather than hold charge, or remain charging to 100%, allow battery usage where the import rate is higher than the value with loss of previously stored energy. 

**Predbat version**

v7.14.39 but also last few version

**Environment details**

- AC3 + 9.5 battery
- HAOS in VM on synology 

**Screenshots**
[image]
[image]
[image]

springfall2008, 2024-01-13:
The numbers in brackets are the cost to charge at that rate while the other numbers would be the cost to use from the grid. The cost to discharge is basically the same as the cost to charge, so in this case using from the grid at 20.09 is cheaper than charging at 16.82 with the losses you have set. If you reduce your losses / metric battery cycle you will get different results

Boffinboy, 2024-01-13:
Thank you, for taking a look. I have tried reducing those values a bit and it didn’t seem to change. What confuses me is the costs, with loss, of a number of the slots are still below the cost from the grid of those circled slots. Eg the cost with loss of the 4:30 slot is 17.55, so any charge picked up at that point and earlier in the night which was even cheaper can be let go at 8:30 or 9:30 and be effectively lower cost than grid.

Perhaps I am misunderstanding how the losses are applied. Are the values with loss “one way” ie the loss on the way in, and then you have to apply further loss on the way out? Or are they “round trip” ie the cost after charging then releasing? They look to be quite a big reduction, so I had assumed it’s round trip. If my settings are battery charge loss 0.05, battery discharge loss 0.05, and inverter loss 0.03, I am assuming one way loss is roughly 0.08 and round trip 0.16 (maybe less if applied multiplicative, rather than additive) Is that how the calculation works?

Boffinboy, 2024-01-13:
Sorry, I’ve also done some tweaking and removed the battery cycle cost - looking at the slot costs with loss it seems to apply the full loss to every slot. I had assumed the value would be applied to a whole cycle. In my case it was set to 2, therefore I had assumed this says the full 9.5 battery cycle would costs 2p. However, looking at the two screenshots below - before with 2 cycle cost and after with 0 - it seems the full 2p is applied to the rate, not the cycle. That seems incorrect, if I am interpreting right, because a slot is only about 1.5kW of charge. Should the difference not be 1.5/9.5*2 = 0.31p difference in rate?

Boffinboy, 2024-01-13:
[image]
[image]

springfall2008, 2024-01-13:
Yes the loss is per kWh and in both directions and so is the battery cycle, so 2p battery cycle will cost an extra 4p per kWh to use the battery vs the grid (plus losses)

Boffinboy, 2024-01-13:
Thanks. See my comment above, probably need to update documentation to make it clear that the battery cycle cost is applied per kWh, and not per cycle. Alternatively, should the calculation not be 1/9.5*2 = 0.21p difference in rate per kWh (assuming 2p cycle cost, and cycle = 9.5. Or should a cycle even be 19kWh, so 1/19*2p = 0.1p extra per kWh

springfall2008, 2024-01-13:
Here's the text from the documentation: https://springfall2008.github.io/batpred/configuration-guide/

Please let me know if it's clear or not?
Thanks

The setting input_number.metric_battery_cycle (expert mode) can be used to put a 'virtual cost' in pence on using your battery for charging and discharging.
In theory if you think your battery will last say 6000 complete cycles and cost you £4000 and is 9.5kWh then each cycle is 19kWh and so the cost per cycle is £4000 / 19 / 6000 = 3.5p. If you configure this number higher then more expensive plans will be selected which avoids charging and discharging your battery as much. The default is 2p but can be set to 0 if you want to turn this feature off. Note that the cycle cost will not be included in the cost predictions that Predbat produces, its just taken into account in the planning stage.
NB: Setting this to a non-zero value will increase your daily cost, but will reduce your home battery usage.

gcoan, 2024-01-13:
> Hiya, I think my confusion has come because I assumed the cost was for the full cycle. The maths you lay out in documentation is clear but is potentially ambiguous because it talks about “cost per cycle” but actually the calculation is something more like “cost per kWh cycled”.
> 
> 
> 
> I read it too quickly and assumed the 2p cost is  the full cycle cost, ie 2 means the model assumes you incur a 2p cost to run 19kWh through the battery (without losses). However, upon second read it become clear the 2p cost is applied to every kWh. 

If @springfall2008 confirms the calculations I'll make it clearer in the next documentation release

springfall2008, 2024-02-03:
Metric battery cycle is the cost for each kWh of charge and each kWh of discharge so using the battery to cover load is 2 x metric_battery_cycle x kWh energy used

springfall2008, 2024-08-06:
Old ticket, closing for now. Feel free to open a new ticket on the latest version if still an issue

closed 2024-08-06 by springfall2008